wills106/homeassistant-solax-modbus #19: Support for Gen4

LuisPalacios, 2022-02-10:
Hi wills106,

Fantastic work, just discovered and implementing it. I have a Gen3, however a friend of mine just installed a Gen4 and haven't seen it as an option while setting up the integration.

Do you have plans to integrate it?.

Thanks
Luis

motclem, 2022-02-10:
> Hi @wills106,
> 
> Fantastic work, just discovered and implementing it. I have a Gen3, however a friend of mine just installed a Gen4 and haven't seen it as an option while setting up the integration.
> 
> Do you have plans to integrate it?.
> 
> Thanks Luis

@LuisPalacios  how do you know what generation you have?

maxrev17, 2022-02-10:
@motclem yes, good question - I'm unsure what gen I have...

Just to add here that as a professional dev, I absolutely LOVE projects like this that I can use with such ease - thanks so much to the contributors!

LuisPalacios, 2022-02-12:
> @LuisPalacios how do you know what generation you have?

HI @motclem, I bought my X1-HYBRID-G3 almost 2 years ago. My friend was offered a X1-HYBRID-G4 during the sales process. More clear answer, it's under 

solaxcloud.com > Inverter > Inverter Type > **X1-Hybrid-G3** 

Thanks, 
Luis

motclem, 2022-02-12:
> @motclem yes, good question - I'm unsure what gen I have...
> 
> Just to add here that as a professional dev, I absolutely LOVE projects like this that I can use with such ease - thanks so much to the contributors!

> > > Hi @wills106,
> > > Fantastic work, just discovered and implementing it. I have a Gen3, however a friend of mine just installed a Gen4 and haven't seen it as an option while setting up the integration.
> > > Do you have plans to integrate it?.
> > > Thanks Luis
> > 
> > 
> > @LuisPalacios how do you know what generation you have?
> 
> HI @motclem, I bought my X1-HYBRID-G3 almost 2 years ago, by then I was offered a G3. My friend bought it 3 weeks ago and was offered a X1-HYBRID-G4 during the sales process.
> 
> You can also find you Generation under:
> 
> solaxcloud.com > Inverter > Inverter Type > (in my case says: **X1-Hybrid-G3**)
> 
> Thanks, Luis spot on mine is g3 also

tom

motclem, 2022-02-12:
> HI @motclem, I bought my X1-HYBRID-G3 almost 2 years ago, by then I was offered a G3. My friend bought it 3 weeks ago and was offered a X1-HYBRID-G4 during the sales process.
>
> You can also find you [...]

> 
> Thanks, Luis

thanks Luis g3 for me too  when trying to install and cvonfigure the integration I get 👍 
[image]
[image]

LuisPalacios, 2022-02-12:
I documented it for myself today (sorry only in Spanish), but you'll get the idea with the screenshots... of the how to install it...

[image]

[image]

[image]

Hope it helps

Luis

LuisPalacios, 2022-02-12:
From README of this repository...

---

Installation

Copy the folder and contents of solax_modbus into to your home-assistant config/custom_components folder. After reboot of Home-Assistant, this integration can be configured through the integration setup UI

---

LuisPalacios, 2022-02-13:
Hi,

I've finished documenting in my blog.

[https://www.luispa.com/dom%C3%B3tica/2022/02/13/hass-solax.html](https://www.luispa.com/dom%C3%B3tica/2022/02/13/hass-solax.html)

Notice it's in Spanish, please use Chrome and use the Translate to option. I tried English and works pretty well.

Hope it helps

Luis

motclem, 2022-02-16:
> Hi,
> 
> I've finished documenting how I setup this Integration (modbus/tcp) in my blog.
> 
> https://www.luispa.com/dom%C3%B3tica/2022/02/13/hass-solax.html
> 
> Notice it's in Spanish, please use Chrome and use the Translate to option. I tried English and works pretty well.
> 
> Hope it helps
> 
> Luis

@LuisPalacios  amazing Thankyou so much got it all working now :) just meed to add in my electricity company to se battery charging times. @wills106  amazing work brilliant 👍

wills106, 2022-02-20:
Gen 4 support is now available.

It has only been tested on an X3 so report anything wrong with X1 and I'll try and correct it.

closed 2022-02-20 by wills106