wills106/homeassistant-solax-modbus #27: Support writing of Charger Times to the Inverter

wills106, 2022-02-21:
The problem though is that I have not worked out an easy way to program them within HA.
The first problem is that for built in Integrations (custom_components) there is no equivalent to input.date_time so I would either need a separate number entry for hours and mins for each time block. So a total of 8 number.solax_ 
Which would result in quite a messy user interface.

The other option is to look into how I monitor a user created input.date_time
Which would either need the user to follow an exact naming convention, or for the option to put in the various input.date_time into the initial Integration Configuration page. So it knows what to monitor.

Then to complicate things more SolaX decided that while you read hours as a single 16bit entry and minutes as another 16bit entry, you program both the hours and minutes for a timeslot as a single 16bit number. Using hi 8bit for mins and the low 8bit as the hours.

So if you want to program the Start Time of Slot 1 you need to do the following for 09:10 am

Hbyte = 10
Lbyte = 9
value = Hbyte << 8 | Lbyte;

Which gives you the number of 2569 which is what you program to the corresponding register.

wills106, 2022-02-24:
[Release 0.4.4](https://github.com/wills106/homsassistant-solax-modbus/releases/tag/0.4.4) Starts to partially address this issue, but in a crude hard encoded way.

wills106, 2022-02-24:
Just checked over the docs. Gen2 & Gen3 uses Lo for hours and Hi for mins. But Gen4 has swapped it to Lo for mins and Hi for hours...

Will need a separate "TIME_OPTIONS" for Gen4 ie "TIME_OPTIONS_Gen4"

infradom, 2022-02-24:
FYI:
I also intended to add time entry fields (see TIME_TYPES_G4), and was
intending so copy some of the work done ly Quitschibo
<https://community.home-assistant.io/u/Quitschibo> described in this thread
:
https://community.home-assistant.io/t/add-device-of-different-domain-in-custom-component/253189/19
But I ran out of time and postponed my effort.

Jan

wills106, 2022-02-24:
It's only meant as a quick stop gap, I'll do the same for Gen4 for the time being.
At least we have partial functionality even if it's in 15min chunks. Should cover most use cases?

I guess it depends if the HA mods would allow this method in an official Integration?
Hopefully date_time will end up as a proper [entity](https://developers.home-assistant.io/docs/core/entity) as some point in the near future. I know it's been asked a number of times in the Dev Chat on Discord.

wills106, 2022-02-24:
Do your times (discharger_start_time_2 etc) report back as 01:05 or 1:5?

I had to pad out the ones for Gen2 / Gen3 as the Inverter's return the values without the leading zero.

infradom, 2022-02-24:
Not sure I understand your question well, in the code I provided, I added a
function Gen4Timestrig that formats time with leading zero as 01:05

def Gen4Timestring(numb):
    h = numb % 256
    m = numb >> 8
    return f"{h:02d}:{m:02d}"

Hope this answers your question.

infradom, 2022-02-24:
So the Gen4 encodes hour and minutes in the same 16bit word, whereas
previous generations used 2 separate words

wills106, 2022-02-24:
It's annoying they keep switching things. Like the above mentioned flip of Hi and Lo for the hours and mins.

I have added in the Gen4 times & selects. Hopefully they are correct.
I have not added it to a release yet. Could you let me know if they look correct?

infradom, 2022-02-24:
I am not convinced that Gen4 is different from Gen2/3 in the byte sequence
My Gen4Timesting function might be wrong however .. but that is not my
feeling as the Solax GUI shows the same time as my sensor
Of course Gen2/Gen3 allocate dedicated 16 bit words for hour and minutes,
wheraas Gen4 encodes them in one single 16bit word

def Gen4Timestring(numb):
    h = numb % 256
    m = numb >> 8
    return f"{h:02d}:{m:02d}"

wills106, 2022-02-24:
So potentially TIME_OPTIONS_GEN4 isn't needed?

infradom, 2022-02-24:
Indeed, that is what I believe, I am still verifying,
It is unclear how you demultiplex the  values returned from the select into
two different 16 bit integers for Gen2/Gen3

infradom, 2022-02-24:
please forget my last comment, write operations are similar ... unlike read

I am still verifying

infradom, 2022-02-24:
I would not drop it yet,
I will test first and see how the values are shown in the solax GUI
I am not sure if the documentation is correct ....

Give me a half an hour to pull your version into my system ...

Jan

wills106, 2022-02-24:
`I am not sure if the documentation is correct ....`

Wouldn't be the first time

infradom, 2022-02-24:
I did a quick test with your current version (different TIMES_Gen4) and it
seems to behave coorectly:
If I load charger_end_time1 with valhue 01:15, it also displays this way in
the solax GUI
So your current interpretation seems to be right

Jan

wills106, 2022-02-24:
Thanks for validating that.

Once I have confirmation that Gen2 works the same as Gen3 I'll release it all as 0.4.5

AviadorLP, 2022-03-02:
Since you are changing times to full digits can you do that to the sensor.solax_rtc please ?

wills106, 2022-03-12:
Closing down due to being on the Todo List #37 for further investigation.

closed 2022-03-12 by wills106