wills106/homeassistant-solax-modbus #28: You need to manually add select and number entries to your Lovelace card

wills106, 2022-02-21:
You need to manually add select and number entries to your Lovelace card.
They are enabled by default but don't appear in the integration page.

Gen2 & 3

select.solax_charger_use_mode
select.solax_allow_grid_charge
number.solax_battery_charge_max_current
number.solax_battery_discharge_max_current
number.solax_battery_minimum_capacity
Gen3 Only

number.solax_forcetime_period_1_max_capacity
number.solax_forcetime_period_2_max_capacity

Gen4 ?

matthewjporter, 2022-02-21:
On the Gen2, The only one for me I have to add manually after creating my card based on entities is:
number.solax_battery_minimum_capacity even though I select it in my list of entities. Adding after the fact is fine

Other oddertiy I have is that number.x1master_battery_charge_max_current & number.x1master_battery_discharge_max_current show as a "number" input and not a slider

[image]

Yet number.x1master_battery_minimum_capacity gets automatically added as a slider

[image]

Trying to find a way to override this in lovelace

wills106, 2022-03-12:
Closing down due to being on the Todo List #37 for further investigation.

closed 2022-03-12 by wills106