wills106/homeassistant-solax-modbus #34: Solaredge Immersion Heater as a Sitch

balajibalasa, 2022-03-08:
Hi, I have an immersion controller V1 paired to the SolarEdge HD wave inverter , can you update this code to make that one available as a Switch , which can be used for automation purposes.

wills106, 2022-03-10:
Are you posting this request on the correct repository?
If you are using an SolarEdge Immersion Heater Controller and an SolarEdge Inverter don't you need one of the SolarEdge repositories?

wills106, 2022-03-12:
Closing down due to no response from original poster and Issue seems to be related to a different integration?
If this is related to SolaX please open a new issue with more details.

closed 2022-03-12 by wills106

balajibalasa, 2022-03-24:
Apologies for my mistake here , I have now posted this on the other integration , was an oversight.