wills106/homeassistant-solax-modbus #40: Document Gen4 Modes of Operation

wills106, 2022-03-23:
Yes, it's great with community efforts :)
Thanks again, I'll take a deep dive in all settings and do some trial and error first.

_Originally posted by @stringkillersa in https://github.com/wills106/homsassistant-solax-modbus/discussions/39#discussioncomment-2421205_

wills106, 2022-03-23:
@stringkillersa are you ok if I assign this task to you? I don't need a PR if you don't want. Just documenting it on this issue page is enough. I can then transfer it into the readme etc.

Anyone else with a Gen4 who want's to volunteer for documentation just let me know, I can assign you as well.

stringkillersa, 2022-03-23:
@wills106  Sure. Is there any documentation other than the Readme? If not, I might make a more thorough document while I'm at it.

wills106, 2022-03-23:
I only have the README.md and what's available in the SolaX User Manuals on their website.

I am expecting the descriptions of the various modes of operation to be quite short. But we can also look into the Github Wiki, if needed?

Once we see how technical / simple the modes are, we can then decide how to display the information.

wills106, 2022-03-29:
Closing issue as documentation has started in the [Wiki](https://github.com/wills106/homsassistant-solax-modbus/wiki)