wills106/homeassistant-solax-modbus #44: X3 Hybrid not responding anymore

haifishtime, 2022-04-01:
Hey,
first of all thanks for your work. It really helped me a lot and was working really well for me until yesterday evening.
Since then all the entities switched to unknown and never came back.

In the logs there was a message saying the request was canceld by the peer or something similar. I don't see it anymore. As a first troubleshooting step I upgraded your integration from 0.3.3 to 0.4.10 and reconfigured the integration. Now I don't get any log output but still all entities are unavailable.

The inverter is still connected to the network and I am able to ping it. I was doing some changes on my Homeassistant instance yesterday but everything worked fine when I stopped.

Do you have any idea what might be the issue here? Is it possible there was an update for the inverter which broke the interface?

Thanks for any help:)

wills106, 2022-04-01:
Have you tried to Power Cycle the Inverter?

haifishtime, 2022-04-02:
I feel a bit dumb saying that I had not power cycled it before but I did now and nothing changed sadly

wills106, 2022-04-02:
When you went to 0.4.10 to you install through HACS, download the release zip or clone the repository?

If you did a clone it could be possible you cloned while I was updating different files?

All I can recommend really is to remove the integration from the integration page.
Delete the custom_component from your HA reboot HA and try again.

If you keep the install name the same and use the same area location if set. It should all slot back into any automations etc.

haifishtime, 2022-04-02:
I downloaded the zip from the release page, deleted the directory in custom components ans replace them with the new ones. Did a restart of HA and because the integration showed errors I completely removed it from the integrations page and reconfigured it. The entities were picked up just like before as you said.

What you are suggesting now is completely removing everything, restarting and the reinstalling?

wills106, 2022-04-02:
Just the SolaX bits. Not sure what else could break it?

wills106, 2022-04-02:
The only time I had the inverter stop talking once was when I tried to connect to it from HA and at the same time trying to communicate with a Window Modbus software. The inverter didn't like it till I dropped all communications and rebooted everything including the Inverter.

Then I only communicated through HA to the Inverter.

wills106, 2022-04-12:
Did your Inverter start responding again to the Integration?

Have you tried any of the newer versions out like 0.4.13a or the pre-release 0.5.0b6?

haifishtime, 2022-04-12:
Hey, sorry for not responding for a couple of days. Illness and time issues...

Aynway the problem still persists and I just removed every trace of the old component and restarted HA.
After that I first tried 0.4.13a without luck.
With 0.5.0b6 I at least got an error message again:
```
2022-04-12 19:12:11 ERROR (MainThread) [custom_components.solax_modbus] cannot perform initial read for serial number
2022-04-12 19:12:11 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry SolaX for solax_modbus
```

Another weird behaviour which I observed is that I still get solar data in the energy dashboard and it seems correct. All the entities are unknown though so I don't know where this information comes from. The battery data is not displayed in the energy dashboard anymore though.

haifishtime, 2022-04-12:
I just went back to the full version 0.4.13a again as I saw some data in the history tab.
Looking a bit further into the history tab (I hope it is called that way in english) and I can actually find all the data I searched for.
And I can also see the data for the whole day and the last couple of days.
If I click on the device in the integration tab I don't see any data, nor do I see it in my lovelace dashboard. It just says that the data is not a numeric value.

The connection to the inverter seems to be working. No idea what is wrong then.

haifishtime, 2022-04-19:
Hey, the integration/connection magically started working again.
I have homeassistant running in a jail on my TrueNAS Server and I tried to install a Pcie so I restarted the whole server. After the restart it worked again.
Classic case of "Have you tried turning it off and on again?" I guess...
I hope it does not need that regularly because I don't really like to restart my NAS.

Thanks for you time anyway.

closed 2022-04-19 by haifishtime

wills106, 2022-04-19:
Glad you are sorted. Let us know if you get anymore strange behaviours though!

I have only used Home Assistant on Synology Docker and Unraid.