wills106/homeassistant-solax-modbus #48: X3 can not set use mode after update

mickemartinsson, 2022-04-04:
After the update - when changing use mode it changes back after 2-4s. Tried also change via Node-Red - same, the use mode cannot be changed any longer from HA.
It used to be very stable - have used Node-Red all winter to buy at the 7 cheapest hours, with a check for lower/higher prices after midnight. It has worked perfectly until the recent update. Now it can not be changed manual (from dropdown) or from Node-Red.
Tried to remove the integration and reinstalled it (after updating of the repository) - the same thing.

mickemartinsson, 2022-04-04:
No restart or lost cloud. Tried to press button.solax_unlock_inverter before reinstall and after - no difference.

Can not change times (start/stop) either - the same problem there - they return to the previous value after 2-4s.

wills106, 2022-04-04:
I am a bit lost with this fault. I am on 0.4.13 on my X1 Gen3 and I have full control.

Another X3 Gen3 issue #44 but then we have a new X3 Gen3 user that installed yesterday and no reported issues? https://community.home-assistant.io/t/solax-inverter-by-modbus-no-pocket-wifi-now-a-custom-component/140143/433?u=wills106

stringkillersa, 2022-04-05:
I noticed the same behavior yesterday. I reverted to 0.4.10, but no change.
Found the cause to be runmode=idle. Works again with 0.4.13 when it woke up. 
(Snowy weather yesterday and very little production, seems it went to idle early...)

infradom, 2022-04-05:
Glad to hear that the  0.4.13 change did non cause the problem.
This idle mode is indeed . hope we find a solution ...

ezfrag2021, 2022-05-23:
I think I am experiencing this problem too but on a new installation of the integration. I have tried deleting the integration from HA and reinstalling but the problem persists.

I have a Solax X3 Hybrid connected to my network via the LAN port on the inverter. The integration connects fine and I can read all sensor value fine but whenever I try to change a setting e.g. SolaX Charger Use Mode from Force Time Use to Self Use Mode I can see the change happen and then instantly change back in the HA Logbook. I have blocked the pocket wifi from accessing the network in case it was this over writing the command but the problem persists.

Do I need to do anything else to enable write commands from HA across the LAN port of the inverter? I found the inverter unlock button and pressed it but nothing happened and I still cannot send commands that stick. I love the integration, by the way, it is far superior to the way of pulling data from the cloud but it would be really nice to be able to push some commands to the inverter. Any help gratefully received!

wills106, 2022-05-27:
If this happens again it might be worth fully powering down your Inverter and power it back on.

I am guessing the Inverter got itself into some form of a pickle?

markjsmith, 2022-06-16:
Hi Will, 
I’ve noticed something similar to this too, when I try to change the mode manually with the select option I get the following error:

Failed to call service. Select/ select option must contain 1 of identity_id, device_id or area_id.

 I've not used selectors before, so I might not have set something properly, but I was thinking it had worked at one point so wondered if it is the same issue?

wills106, 2022-06-27:
@markjsmith 

Has your Inverter been connected to the cloud previously?

If it hasn’t been connected to the cloud previously or the Inverter has had a full power cycle since it might be in “Locked” mode. Try pressing the “Unlock Inverter” button. It’s full name will be something like `button.solax_unlock_inverter` depending on how you named it while setting up.

Also if you enable the disabled sensor `sensor.solax_lock_state` you can check the lock status.

wills106, 2022-08-11:
Is anyone still having any of these issues? Or can I close this issue down?

ezfrag2021, 2022-08-11:
Sorry I thought I had already given an update in this thread. I left everything for a few days and went on holiday and when I returned it was suddenly working.

I suspect the problem was this the inverter as you suggested and maybe it needed to cycle modes to start working properly again. All ok now. Thanks.

M0LTE, 2022-08-14:
@wills106 My X1 hybrid gets stuck in Idle mode and Battery Awaken, Unlock Inverter and Unlock Inverter - Advanced do nothing. The SolaX Cloud app permits me to get the inverter to Run Mode, and then changing inverter settings in HASS works. Would this be the same issue or would you like me to raise a different one?

markjsmith, 2022-08-15:
That’s the same. It happened last for me after a power cut.

Mark

wills106, 2022-08-15:
It's a known issue and I'm not sure what the Cloud does different to wake the Inverter up.
It's also very hard for me to trouble shoot as I have enough battery to last over 24hr If I do have two bad solar days I have a 3hr period during the night with cheap electric to charge from the grid.

Are you both trying to change Run Mode so you can charge from the Grid? or are you trying to do something else?

M0LTE, 2022-08-15:
That's exactly what I'm trying to do. My cheap energy starts at 00:30 local time and just before then I'd like to make a dynamic decision on what, if any, grid charge to take depending on the morning and full day solar forecast and anticipated local energy needs. I have 18kWh of storage and we utilise around 40kWh per day. On a good day we generate 95% of that from solar, on a bad day we generate 10% of that from solar. My cheap energy is more than 4x cheaper than my peak energy which is why this matters.

Is there anything I can do eg obtain logs to assist?

wills106, 2022-08-15:
I'm not sure what Inverter you are using but you mentioned 6S58 here #95 but that isn't a know start of a serial.

If you own aGen2 or Gen3 I would use "Force Time" mode see [wiki](https://github.com/wills106/homeassistant-solax-modbus/wiki/Gen2-&-Gen3-Modes-of-operation)

M0LTE, 2022-08-15:
H4502TH8... (truncating since I don't know if that's sensitive) if that's helps.

wills106, 2022-08-15:
You have a Gen4 then (Only needs first 4 digits), so you will need to look at the Gen4 Modes of Operation on the Wiki

M0LTE, 2022-08-15:
I'll look to see if that would work here. Thanks. 

Do you think there is any mileage in having the integration write a setting, then read it back from the inverter, and error (somewhere) or maybe retry if it doesn't stick?

M0LTE, 2022-08-15:
@wills106 would you say there is any mileage in pulling the Pocket WiFi dongle out? Would you expect things to work (other than the app) if I did this? I wonder if it is interacting badly with the modbus connection in some way.

wills106, 2022-08-15:
> I'll look to see if that would work or help here.

If you set the charge times before the Inverter goes to sleep it knows to wake up and start charging at say 00:30

> is any mileage in pulling the Pocket WiFi dongle out?

I don't know, I don't have a PocketWiFI or a PocketLan. I connect to the cloud using the built in ethernet.
All you can do is try?

M0LTE, 2022-08-15:
Right- reason for delaying it is to use the latest possible solar forecast for the next day.

Are you aware of any way to prevent the inverter going to Idle mode? It doesn't provide any benefit...

wills106, 2022-08-15:
If you are set to start charging at 00:30 but you don't need to charge have an automation at 00:31 to put the Inverter back into Self Use mode. Or modify the charge window.

I am assuming once the Inverter has woken up to start charging, it will be awake enough to receive normal commands.
Once it's wake at 00:30 and you put it back into Self Use mode at 00:31, I don't know how much longer it will stay awake for though.

I don't know if there is anything useful to you in #41

M0LTE, 2022-08-15:
Some good ideas, thanks, I'll try.

markjsmith, 2022-10-11:
Hi Will,

Yes the inverter is cloud enabled and unlocked.
Interestingly I can change the state by using an automation which is the main thing, but the selector doesn’t work.

Mark

wills106, 2022-11-29:
Is anyone still having issues changing modes? Especially when the Inverter is a sleep.

There is a comment here on waiting a short while after pressing battery_awaken https://github.com/wills106/homeassistant-solax-modbus/discussions/202#discussioncomment-4223942

I'll give it a couple of days and then close the issue down if no further feedback.

ezfrag2021, 2022-11-29:
I didn't realise there was a possible solution. I had never noticed the battery awaken button before. I will try it out the next time the inverter goes to idle.

I am still using the workaround of only changing the use mode a couple of minutes after the inverter wakes up to charge the battery in the night. This was a pain last week when we had a storm blow in and I couldn't set it to back-up mode because it was asleep. Had to create an automation to do it in the middle of the night.

stringkillersa, 2022-11-29:
Maybe related: I cannot set
number.solax_1_backup_discharge_min_soc

It reverts back to 30.

Inverter is unlocked and in "Back Up Mode" (which also prevents going to sleep)
Have restarted inverter.
I have not had time to investigate any deeper than that.

Regards
Stig

There is a comment here on waiting a short while after pressing battery_awaken #202 (comment)<https://github.com/wills106/homeassistant-solax-modbus/discussions/202#discussioncomment-4223942>

M0LTE, 2022-11-29:
I’ve just updated from 0.5 to 0.6. In 0.5 I have never been able to change
mode while the inverter is in idle, and I’ve never been able to get the
inverter out of idle using buttons from the add-on.

Is this changed in 0.6? 0.7 betas?

Cheers.

stringkillersa, 2022-11-29:
> @stringkillersa<https://github.com/stringkillersa> Do you know what version you are on? If you are on an old version a whole host of this have changed, so could well be fixed anyway?
>
> @M0LTE<https://github.com/M0LTE> I can't remember when battery_awaken button was introduced. If you read that comment it's more about giving the inverter a delay after pressing battery_awaken before doing something else, like changing mode. I think the Inverter needs time to wake up from being asleep.
>
> I personally haven't tried it as my Inverter never goes to sleep at night.

I have version 0.6.4

/Stig

wills106, 2022-11-29:
@stringkillersa that's the same as #83 isn't it?

ezfrag2021, 2022-11-29:
Ok I tested the suggested workaround this evening when my inverter went to idle mode.

[image]

Tried to change run mode from force use to back-up first and as usual when the inverter is idle it switched straight back to force use mode. The command was ignored.

Called the Solax battery awaken button press and the inverter switched to waiting and then checking and then just over a minute later it went to normal.

I tried to change to back-up mode and this time it successfully changed.

About a minute later the inverter went back to idle.

So the workaround does work but you need to put a "wait until state changes to normal" command in the automation before pushing the required command to the inverter.

wills106, 2022-11-29:
Thanks for further working on the provided idea.

Are you able to write up a simple automation with the basics. I can then add it to the wiki section.

stringkillersa, 2022-11-30:
Yeah, sounds like 83. I have The latest firmware available,  so I Will uppgrade and see what happens.
/Stig

Hämta Outlook for Android<https://aka.ms/AAb9ysg>

wills106, 2022-11-30:
There seems to be a bug in the firmware. 0.7.0bx branch wont fix backup_discharge_min_soc

M0LTE, 2022-12-01:
> This is an example script to put the inverter into Back Up Mode. It checks
> the state of the inverter first and if its not in Normal Mode it wakes it
> up and then selects Back Up Mode.
> `alias: Set Back Up Mode
> sequence:
>       - condition: state
>       entity_id: sensor.solax_run_mode
>       state: Normal Mode
>       option: Back Up Mode
>       target:
>       entity_id: select.solax_charger_use_mode
>       - service: button.press
>       entity_id: button.solax_battery_awaken
>       - wait_for_trigger:
>          - platform: state
>             - sensor.solax_run_mode
>             to: Normal Mode
>       mode: single`

Wouldn’t it be better if the add-in handled getting the inverter into a
state where it accepts commands transparently, rather than having to bake
this logic into every automation? (Or remembering to do it manually if
using the UI)

ezfrag2021, 2022-12-02:
I am eternally grateful to people like @wills106 who can do something that I cannot do. Without people like this some of us would be stuck with a dumb inverter that doesn't talk to HA and also stuck using the cloud service to manage it.

I will take whatever they give me and be thankful and try to give back with feedback and help where I can.

I'm sure if you want to make the suggestion about how to integrate this into the add-on it would be well received...

infradom, 2022-12-02:
> Wouldn’t it be better if the add-in handled getting the inverter into a state where it accepts commands transparently, rather than having to bake this logic into every automation? (Or remembering to do it manually if using the UI)

This is something we can consider, but it is not without any risk, carefull consideration and preparation will be needed. I suggest not to implement such things in a rush. The title of this thread is somewhat misleading, so maybe we should open a new 'feature request issue' thread where we can discuss the considerations to make.
The title of this issue could be: "Implement automatic wakeup when changing parameters when inverter is in sleep mode"

M0LTE, 2022-12-02:
Fully agree.

wills106, 2022-12-09:
I am going to close this issue down as the new Feature Request #226 covers the same topic.
Feel free to join into the conversation.

closed 2022-12-09 by wills106