wills106/homeassistant-solax-modbus #54: GEN2 X1 Missing Sensors / Readings

matthewjporter, 2022-04-09:
Hey Will

Missing the "Today's Yield" sensor on the Gen2 X1

If i setup as a Gen3 X1 I get them

Cheers

wills106, 2022-04-09:
Confirmed if I change my Integration to Gen2x1 "Today's Yield" drops to 0kWh

This is on 0.5.0b3

matthewjporter, 2022-04-09:
Some Additional that seem to have taken a bit of a holiday but work when using Gen3 integration

I know some may be reading the wrong register hence they are getting a reading but here they are

Grid Import 
Measured Power
Today's Yield (As above)
Total Yield

infradom, 2022-04-09:
I am trying to find the cause of the problem ... This is probably a side effect of the code restructuring but I have not found the error yet ...

wills106, 2022-04-09:
I have only looked into Today's Yield so far (today_yield) but I can't see anything obvious.

It's declared as `allowedtypes=ALLDEFAULT,` which matches all the other generic sensors between Generation's.
If that one didn't work you would expect most of the other sensors not to work either?

I'll look into the others tomorrow.

infradom, 2022-04-09:
Yes, the sensor declarations look ok, but I fear the __init__.py sequential reading of parameters might get out of sync for gen2.

infradom, 2022-04-10:
Found the error, but as I do not have a Gen2 description I do not know how to correct:
the error was not introduced  by the doce resturcturing, but by a prior change.
In read_input_registers0, there is an if statement searching for a GEN2 condition

The If clause consumes 5 variables, the ELSE clause consumes 6 variables.
I knot know where I should add a skip_bytes(4) statement

wills106, 2022-04-10:
That will be from when I added in input_energy_charge & output_energy_charge for the Gen2 as they sit in a different pattern of registers on the Gen2 vs Gen3 & Gen4

I'll double check the docs, maybe 2 registers got deleted by accident at some point? or the skip_bytes(4) did?

wills106, 2022-04-10:
Within `__init__.py` lines 768 to 772 shouldn't be inside the `else:` statement.

@matthewjporter can you change the following:
[image]

Could you confirm please and then I can push a new update out? Or would you prefer me to just push it out anyway?
I have switched to Gen2 mode and I now have Today's Yield showing.

to:
[image]

infradom, 2022-04-10:
OK, thanks, I will test in my development fork also ...

infradom, 2022-04-10:
Looks indeed OK this fix

wills106, 2022-04-10:
@matthewjporter

I have pushed it out as 0.5.0b4 as I am pretty confident this fixes that issue.

matthewjporter, 2022-04-10:
Just updated (via phone) and on the surface looks to be resolved as I am seeing states

Will check later when at my PC

Thanks Guys

closed 2022-04-12 by wills106

matthewjporter, 2022-04-14:
Hi

Need to reopen this

Looking closer at both the C & D and running the realtime.html direct to the wifi and in the iOS app direct to them it seems on the Gen2's the registers are a complete mess :(

They should be as follows:

"today_s_yield" is actually the Total PV generation for the day (for both C&D variants) and should be what is currently "today_s_solar_energy" (Disabled/Removed currently)

"total_solar_energy" is actually total PV + Batt Yield for the day so needs a renamed to "today_s_total_yield"

Hope that makes sense

This should bring back PV Yield and Total Yield to the Gen2's - I think