wills106/homeassistant-solax-modbus #55: "Total Solar Energy" Missing on earlier Gen2 SK-SU (U50EC)

wills106, 2022-04-10:
"Total Solar Energy" is missing on the Earlier Gen2 SK-SU with serial number starting U50EC it just returns 0kWh.
But on what is assumed to be the later U50ED it returns a value which is believed to be correct at the moment.

There is a comment in the docs that "PvEnergyTotal" which is "Total Solar Energy" in the integration. Got added but then 4 months later got moved? So possibly Inverter U50EC got sent out with wrong register address? As the register mention works for the U50ED Inverter.

@matthewjporter to manually scan for a uint32 that could provide the missing value.

wills106, 2022-04-10:
@GpMissi Have you tested any of the Pre-release's yet? As in 0.5.0bx

I know your SK-TL starts L50E but don't know the 5th character, so not sure if "Total Solar Energy" is available on your Inverter or not? It was added on 0.5.0b5.
Just so you know the default Gen2 charge / discharge is still 50A, not got round to implementing 100A based on your serial L50E yet.

matthewjporter, 2022-04-14:
Hi

Need to reopen this

Looking closer at both the C & D and running the realtime.html direct to the wifi and in the iOS app direct to them it seems on the Gen2's the registers are a complete mess :(

They should be as follows:

"today_s_yield" is actually the Total PV generation for the day (for both C&D variants) and should be what is currently "today_s_solar_energy" (Disabled/Removed currently)

"total_solar_energy" is actually total PV + Batt Yield for the day so needs a renamed to "today_s_total_yield"

Hope that makes sense

This should bring back PV Yield and Total Yield to the Gen2's - I think

matthewjporter, 2022-04-21:
The following is now fixed in 0.5.0b8a Pre-release

"today_s_yield" is actually the Total PV generation for the day (for both C&D variants) and should be what is currently "today_s_solar_energy" (Disabled/Removed currently)

"total_solar_energy" is actually total PV + Batt Yield for the day so needs a renamed to "today_s_total_yield"

Note: "total_solar_energy" is not available in the "C" Variant so that can be ticked off as completed
Many Thanks

wills106, 2022-04-25:
Closing down as I believe all sensors are corrected now?

Any further issues please open a new issue.

closed 2022-04-25 by wills106