wills106/homeassistant-solax-modbus #59: X1-AC - Unknown

Mouxy, 2022-04-17:
Hi Guys,

I've been following the discussion between Will and Mark (Cavester) - https://community.home-assistant.io/t/solax-inverter-by-modbus-no-pocket-wifi-now-a-custom-component/140143/482

I tried running the latest beta and also the X1-AC-5 beta and I believe I'm getting the same results as Mark, everything appears as Unknown.

I tried with using Waveshare and the USB adapter with the same results.

When checking the logs I'm getting this output:

Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 202, in async_refresh_modbus_data
   update_result = self.read_modbus_data()
  File "/config/custom_components/solax_modbus/__init__.py", line 244, in read_modbus_data
    return self.read_modbus_holding_registers_0() and self.read_modbus_holding_registers_1() and self.read_modbus_input_registers_0()
  File "/config/custom_components/solax_modbus/__init__.py", line 264, in read_modbus_holding_registers_0
    factoryname = decoder.decode_string(14).decode("ascii")
UnicodeDecodeError: 'ascii' codec can't decode byte 0x96 in position 8: ordinal not in range(128)

Happy to help in any way I can

infradom, 2022-04-17:
Thanks, this traceback will be very usefull. Apparently the serial number is either shorter than 14 characters or it contains  non-ascii characters.

Mouxy, 2022-04-17:
Hi,

This is the SN: XAC362H8210130

Any way to check the registers so I can share it with you?

infradom, 2022-04-17:
Sorry, It is not the serial number that causes the crash, but the next 14 character string: FactoryName

Mouxy, 2022-04-17:
Thanks for replying 😊
I have 0 experience with modbus so I might be doing something that is not useful at all but I got some readings: https://mouxy.cloud/AW7HEcN

I also asked SolaX if they have any docs for the registers for the X1AC

Thank you

infradom, 2022-04-18:
On first sight, these values correspond to the register documentation for the Solax Hybrid systems (as @wills106 predicted).
I believe the main problem (for now) is that the factoryname contains a non ascii character (38464 or 0x963e hex) 
To get further, we could simply modify the custom_components/solax_modbus/__init__.py file and replace lines

```
        factoryname = decoder.decode_string(14).decode("ascii")
        self.data["factoryname"] = str(factoryname)
```

By one line: (attention, identation is important, you need 8 spaces before following code)

`     decoder.skip_bytes(14)`

If you do not feel at ease modifying python code, I am sure @wills106 can prepare a special build for you

Mouxy, 2022-04-18:
It worked 😊 

And I can also confirm that I was able to change the Work Mode, confirmed it using SolaxCloud

Thanks for your help!

[image]

[image]

infradom, 2022-04-18:
Thanks for the feedback, 
We will adapt the main line in the coming days. It is important to realize that some parameters do not apply to the AC line. (PV related stuff). We will remove them in the coming days.
The time period settings are significantly different depending on the Generatio (Gen3 or Gen4). As we are guessing for thiis AC model, we need to be extremely careful not to put your inverter in a strange state. Be carefull with each untested modification and make sure to verify in the solax app after each change.
By the way, which version of the integration are you using ? The Beta/prerelease or the main line ? And have you declared as Gen3 or Gen4 or none ?

Thanks

Mouxy, 2022-04-18:
I'm running 0.5.0b6 and I didn't select any Gen. Let's see if SolaX will send the modbus docs for the AC.

Let me know if you need me to test anything else

Thank you

infradom, 2022-04-18:
As long as we do not have a modbus documentation, it would help if you send us also the registers starting at address 256 till 277
This area contains the read values for ForceTimeUseP1MaxCapacity and ForceTimeUseP2MaxCapacity

Mouxy, 2022-04-18:
[image]

P1 and P2 are set at 75% in the solax app

infradom, 2022-04-18:
Thanks, the '75' values do not appear at the offset we expected them to appear. So we are using the wrong read address for these 2 registers.
One question: when you tried to modify one of these values in the integration, did it appear as modified in the SolaxCloud app? In other words: do we use the correct write register address ?

Mouxy, 2022-04-18:
Changing the values in the integration works, they change in the app too. 

Strangely enough, I've never seen these values before in the app, they only appeared after setting up this integration.

wills106, 2022-04-18:
We might as well drop the whole "factoryname" & "modulename" part altogether.

infradom, 2022-04-18:
@wills106 Indeed, it brings no value.
So the register addresses seem to be 0x010C and 0x010D for this charger (instead of 0x010F and 0x0110) for this AC model

wills106, 2022-04-18:
Not sure what your PV1 and PV2 power are reporting as your Inverter doesn't have any PV connections.

Problem with the lack of documentation it's pretty much as guessing game trying to workout any missing Registers. Especially the write ones.

I had a Firmware update last year and have no documentation to know where these new Registers are.
It's ok doing read request. I was lucky to find "Charger Use Mode" without any documentation. But I wouldn't try it again.

infradom, 2022-04-18:
I agree, it is very dangerous doing writes without documentation
But aparently, @Mouxy confirmed that the write address did not change for these parameters

Mouxy, 2022-04-18:
The inverter as an option for CT2/Meter2 and I asked SolaX if this can be used. 

My guess is that CT2 can be used to monitor generation from another inverter like solar. They mention that in one of the many versions of the same user manual.

Mouxy, 2022-04-19:
Hi Guys,

I hope this helps: https://mouxy.cloud/A39VBbf

wills106, 2022-04-19:
That seems to be only half of a document. It's missing the write registers.

I am guessing they have the full version and a cut down read only document.
Normally the full documents are around 50 - 60 pages.

Could you ask if you can have a copy of the full document including the write register and write registers section?

Also if they have a newer Gen3 X1 & X3 Hybrid document than v3.25 That one was last updated 2020-8-27 but since then there has been a newer Firmware on at least the X1 Gen3 which has introduced a couple of new features that are not in the current docs.  It will be called something like "Hybrid X1_X3-G3 ModbusTCP_RTU V3.25- English.pdf" but hopefully with a newer Version number.

wills106, 2022-04-19:
Had a quick scan over, it is very similar to the Gen3 Hybrid like I expected with the PV & EPS parts removed.

They have shifted a few things about like ForceTimeUseP1MaxCapacity and ForceTimeUseP2MaxCapacity have been brought forward by 3 registers each.

@matthewjporter you know the other day you Lock Status was 2. Had you logged into your inverter with the higher level password? According to this document when it return 2 it means "Super Password Unlock"
Maybe that's why "Battery Awaken" doesn't work on the Gen3 as we need to unlock at a higher level?

mickemartinsson, 2022-04-19:
Yes!! That could be the problem.

You need to enter Advanced (not Super) to access Battery Awaken – see image:

[En bild som visar text  Automatiskt genererad beskrivning]

wills106, 2022-04-19:
@mickemartinsson

I have not quite finished with the update. But writing the super password does alter the unlock mode. So perhaps battery awaken will work on the Gen3?

[image]

mickemartinsson, 2022-04-19:
I’ll try the update and see if it works.

[image]<https://user-images.githubusercontent.com/18155231/164064262-36226670-7086-4010-8dfe-e94cbd59bc60.png>

Mouxy, 2022-04-21:
Thanks everyone for helping, and I can confirm the latest 2 betas resolved the issues without the need to manually change the files.

closed 2022-04-21 by Mouxy