wills106/homeassistant-solax-modbus #67: After upgrade to 5.0 sensors missing

Claudio1L, 2022-04-25:
Inverter X1-Hybrid-G3 
Had to rollback to the previous release because of missing the following sensors:

sensor.solax_battery_input_energy_today
sensor.solax_battery_output_energy_today
sensor.solax_today_s_export_energy
sensor.solax_today_s_import_energy

Only total sensors are available and missing daily ones for both Grid import/export and battery input/output

infradom, 2022-04-25:
Seems like your serial number prefix was not known to us yet.
Hopefully we can continue to derive the model type from the serial number prefix ...

Claudio1L, 2022-04-25:
discovered entities from 0.4.x to 0.5.0 reduced from 118 to 54

wills106, 2022-04-26:
Did you remove the integration from the Integration page before updating or did you just migrate directly to 0.5.0 as really the Integration should have failed with the following error in your log "unrecognized inverter type - serial number : {your_serial_number_here}"
As it's an unknown Inverter.

Could you have a look at the following discussion #26 and provide the rest of the info for me specifically [this post](https://github.com/wills106/homsassistant-solax-modbus/discussions/26#discussioncomment-2453543) they I can add in support for your Inverter.

pchaves1, 2022-04-26:
X1-Hybrid-G4
7.5kW
Single Phase
Dual MPPT
EPS External
Gen4
Serial: H4752
Had to rollback to the previous release because of missing sensors on 5.0 reported previously

wills106, 2022-04-26:
I have added in the X1-Hybrid-G3 serial XRE3 on 0.5.1 I have put the default Charge / Discharge to 20A for the time being.

Claudio1L, 2022-04-26:
Thanks for the effort, hope to find time to test 0.5.1 tomorrow.

Claudio1L, 2022-04-27:
I did the upgrade to 0.5.1, removing the previous 0.4 before updating. Now "today" entities are available. The funny thing is that it seems not recognize the inverter model 
[image]

Claudio1L, 2022-04-27:
Anyway, most of the entities are populated, thank you very much.  Just few in state "unknown" that I don't know if it's expected  or not.
In particular, they are:

sensor.solax_inverter_model_number
sensor.solax_battery_install_capacity
sensor.solax_battery_package_number
sensor.solax_battery_state_of_health
sensor.solax_discharge_cut_off_capacity_grid_mode

nothing vital, just to let you know.

Quick question: I'm not been able to understand which settings unlock the inverter with the unlock and advanced unlock button. Do you have any doc to share? Thanks for all the effort

wills106, 2022-04-27:
"Inverter Model Number" was meant to be disabled by default. Along with sensor.solax_inverter_model_number
& sensor.solax_battery_install_capacity these were added in a firmware update around early to mid 2021 so they may not be present on all Gen3 Inverters.

sensor.solax_battery_package_number & sensor.solax_battery_state_of_health are meant to be Gen2 only. So I will need to look at the Integration again and look at turning them off on the Gen3 & Gen4

disch_cut_off_capacity_grid_mode shouldn't be there on the Gen3 I renamed it to battery_minimum_capacity_gridtied so I need to remove the reference to disch_cut_off_capacity_grid_mode

If your Inverter has never been connected to the Cloud / possibly after a firmware update, you need to press the "Unlock Inverter" button. If you don't you can read values, but you are unable to adjust select / number.
"Unlock Inverter - Advance" button will unlock your Inverter using the higher level password. It may allow the battery awaken button to work on the Gen3, but this is untested.

Claudio1L, 2022-04-27:
Perfect. Thank you again for your effort.

wills106, 2022-04-27:
I have corrected the sensors on the Gen3 in 0.5.2

If there are any other issues let me know, if everything is ok I can close the issue down.

Claudio1L, 2022-04-27:
it seems perfect. Thank you very much again

wills106, 2022-04-30:
Closing issue as no further related issues reported.

closed 2022-04-30 by wills106