wills106/homeassistant-solax-modbus #69: Error setting up entrysolax for solax_modbus

motclem, 2022-04-26:
Having deleted the integration I don't upgrading the components in HACS when I try to re-add the integration I will get in the following error message:
This error originated from a custom integration.

Logger: homeassistant.config_entries
Source: custom_components/solax_modbus/__init__.py:254 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 20:03:43 (1 occurrences) 
Last logged: 20:03:43

Error setting up entry SolaX for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solax_modbus/__init__.py", line 89, in async_setup_entry
    inverter_data = hub.read_holding_registers(unit=hub._modbus_addr, address=0x0, count=7)
  File "/config/custom_components/solax_modbus/__init__.py", line 254, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 108, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.151:502)]

Read my mind

infradom, 2022-04-26:
You had a typo on the IP address: 192.168.151
It should have 4 ports

motclem, 2022-04-26:
Thank you sorry for wasting your time I'm an idiot!

infradom, 2022-04-26:
Dont worry, it happens to me also ..

motclem, 2022-04-26:
Thank you:-) on my overview dashboard i angle used to have drop downs to select the run mode sliders for other parameters why would they have disappeared?

wills106, 2022-04-26:
Number and select entities need to be manually added to a Dashboard.

For example on my Gen3 X1 I manually add a card in like:

```
entities:
  - entity: sensor.solax_battery_capacity
  - entity: sensor.solax_battery_temperature
  - entity: sensor.solax_battery_power_charge
  - entity: sensor.solax_battery_voltage_charge
  - entity: sensor.solax_battery_current_charge
  - entity: select.solax_charger_use_mode
  - entity: select.solax_allow_grid_charge
  - entity: number.solax_battery_charge_max_current
  - entity: number.solax_battery_discharge_max_current
  - entity: number.solax_battery_minimum_capacity
  - entity: number.solax_forcetime_period_1_max_capacity
  - entity: number.solax_forcetime_period_2_max_capacity
  - entity: select.solax_charger_start_time_1
  - entity: select.solax_charger_end_time_1
  - entity: select.solax_charger_start_time_2
  - entity: select.solax_charger_end_time_2
  - entity: button.solax_battery_awaken
  - entity: button.solax_unlock_inverter
  - entity: button.solax_unlock_inverter_advanced
show_header_toggle: false
title: SolaX Battery
type: entities
```

infradom, 2022-04-26:
Maybe your inverter is not recognized (we use the first digits of the serial number to determine which inverter you have). 
Is there an error message about unrecognized inverter in the logs?
See Discussion #26

motclem, 2022-04-26:
Ok Great  thanksin the previous version I just seem to appear.

wills106, 2022-04-27:
@motclem did you manage to get the Integration installed ok?

If you did, is it ok I close this Issue down?

wills106, 2022-04-30:
Closing issue as no further related issues reported.

closed 2022-04-30 by wills106