wills106/homeassistant-solax-modbus #72: missinentity x1 gen3

motclem, 2022-05-06:
if I look at the entities list for the solax modbus  integration there does not seem to be an entity for pv power as there was in devious versions.

motclem, 2022-05-06:
[image]

wills106, 2022-05-06:
@motclem I'm assuming from your post here https://github.com/wills106/homsassistant-solax-modbus/discussions/26#discussioncomment-2255645 you are using an Inverter that has a serial number starting H1E5?

wills106, 2022-05-06:
Looks like you have a different Gen3 X1 then.

What's your Charge / Discharge current - as indicated on the Inverter LCD or Cloud, not in the Integration?

I am just about to add another Inverter so I can add your's in as well.

motclem, 2022-05-06:
trying to confirm the s/n but have locked myself out of my solax cloud account! can I find it on the device itself do you know?

wills106, 2022-05-06:
Yea I think it's Advance - Battery - Charger on the LCD, off the top of my head.

It will probably only allow you to go upto 20 or maybe 25A

motclem, 2022-05-06:
20 for both I think

wills106, 2022-05-06:
I have just released 0.5.5 which should work for your Inverter.

motclem, 2022-05-06:
@wills106 you're a legend :)
[image]

<img width="865"

wills106, 2022-06-06:
Closing this issue down as I believe the Integration is working correctly for you.
If there are any further issues please raise a new Issue.

closed 2022-06-06 by wills106