wills106/homeassistant-solax-modbus #73: Gen 2 (SK-SU3000E) showing Battery Voltage x 10

blb4github, 2022-05-16:
The SolaX Battery Voltage Charge entity is showing the battery voltage multiplied by 10, currently it shows 529,8 V.

wills106, 2022-05-23:
This Integration has only been tested on Pylontech batteries I believe on the Gen2. I am guessing the LG Battery responds different, so I will need a different scale for it.

matthewjporter, 2022-05-23:
My Second Inverter has LG Chem ESS batteries and works the same as my Pylontech Batteries – Both on SK-SU5000E inverters

Reply to: wills106/homsassistant-solax-modbus ***@***.***>
Date: Monday, 23 May 2022 at 17:41
To: wills106/homsassistant-solax-modbus ***@***.***>
Cc: Subscribed ***@***.***>

blb4github, 2022-05-23:
It's also the Battery Current values which are x10. Power Values are looking fine, see attachement.

[image]

blb4github, 2022-05-23:
Thanks, very much appreciated!!

wills106, 2022-05-24:
What is the battery charge / discharge max current? Is it 50A like on the 5kW version?

wills106, 2022-05-24:
I have just added in the serial numbers for 3kW & 3.7kW in 0.5.6

After updating to 0.5.6 I would recommend you remove the Integration from the Integration page and readd it, so it detects your Inverter correctly.
Can you let me know if the total entities changes from 55?

blb4github, 2022-05-24:
I'm not sure which version I'm running, I have deleted the integration and added it again. In the error log I see:

2022-05-24 16:43:44 ERROR (MainThread) [custom_components.solax_modbus] unrecognized inverter type - serial number : U30EDxxxxxxxxxx (My serial number mentioned fully)
2022-05-24 16:44:04 ERROR (MainThread) [custom_components.solax_modbus] must be either HYBRID or AC

wills106, 2022-05-24:
Did you install through HACS?

You are now on a version of 0.5.x prior to 0.5.6

The first time round when you installed it and went to configure the Integration was there multiple tick boxes to select from?

blb4github, 2022-05-24:
I did indeed install trough HACS. In the mean time I have seen an upgrade to 0.5.6 is available and I have installed it. Now I have 92 entities available and Current and Battery Voltage is correct now!
[image]

They look like:
[image]
[image]
[image]

wills106, 2022-05-24:
I think when you first installed it and only got 55 entities this was because you had installed 0.3.x or 0.4.x

When installing on 0.5.x the correct behaviour for an unknown Inverter is the error you experienced just.

> 2022-05-24 16:43:44 ERROR (MainThread) [custom_components.solax_modbus] unrecognized inverter type - serial number : U30EDxxxxxxxxxx

Your Inverter only got added into 0.5.6

blb4github, 2022-05-24:
Thanks for this work! What do I have to do the be able to change setting via this integration? I don't have the idea something changes if I press one of the buttons.

wills106, 2022-05-25:
Have you added in the various number and selects to a Dashboard card?

```
entities:
  - entity: select.solax_charger_use_mode
  - entity: select.solax_allow_grid_charge
  - entity: number.solax_battery_charge_max_current
  - entity: number.solax_battery_discharge_max_current
  - entity: number.solax_battery_minimum_capacity
  - entity: number.solax_battery_minimum_capacity_grid_tied
  - entity: number.solax_forcetime_period_1_max_capacity
  - entity: number.solax_forcetime_period_2_max_capacity
  - entity: select.solax_charger_start_time_1
  - entity: select.solax_charger_end_time_1
  - entity: select.solax_charger_start_time_2
  - entity: select.solax_charger_end_time_2
  - entity: button.solax_battery_awaken
  - entity: button.solax_unlock_inverter
  - entity: button.solax_unlock_inverter_advanced
show_header_toggle: false
title: SolaX Battery
type: entities
```

You normally only have to press the unlock_inverter once if you haven't previously connected to the cloud of following a power cycle on the Inverter.
If you enable the disabled by default sensor lock_state what does that return?

wills106, 2022-06-06:
Is this Integration now working how you expected it to?

If so I will close down this issue.

blb4github, 2022-06-06:
Hi, I have just managed to have all connected again and I'm still having trouble to control the system. on https://community.home-assistant.io/t/solax-inverter-by-modbus-no-pocket-wifi-now-a-custom-component I read something about modbus address, could that be something I have to adapt?

infradom, 2022-06-07:
@blb4github @wills106 : As far as I know, modbus address 0 is the broadcast address, so I have a bad feeling about using this address. Maybe your inverter is configured for another non-default address, or something else is wrong ..

matthewjporter, 2022-06-07:
> Reply to: wills106/homeassistant-solax-modbus ***@***.***>
>
> @matthewjporter<https://github.com/matthewjporter> Do you know if it's possible to change the modbus address on the Gen2? I can't see anything in the manual, so I am guessing possibly not? Does the WiFi Portal / phone app display the address at all?
>
> As you connect the Gen2 and Gen3 via Ethernet over TCP I wouldn't really expect the need to change the address from the default to be honest?

Nothing on the App, I have not checked on the inverter itself if it can be done but do not remember anything

The SolaX portal for the Gen2 as far as I know cannot control the inverter

Maybe worth connecting to the inverter via the local App and perform a reset – I think can also be done on the inverter menu itself

blb4github, 2022-06-07:
I can't find a reset option in the app (local connection) and also not on the inverter itself. on the inverter itself I see "Reset Mgr EE"

matthewjporter, 2022-06-07:
Just checked and only seems to have Reset Energy & Reset Logs

Not sure if doing it via the WiFi interface will reset but worth a try. When I login on the WiFi IP I have a “Factory Reset” button but not sure what that resets

Date: Tuesday, 7 June 2022 at 11:56

blb4github, 2022-06-07:
Do you mean this button "Clear Config"?

[image]

matthewjporter, 2022-06-07:
I think that is just the backups in the app

If your inverter is connected to a WiFi network you should be able to connect to that IP address in your browser and get the following screen
***@***.***
It’s the only place I know of a place to “Reset” something

Not sure what that reset does, if it just reset the WiFi / Pocket WiFi interface or everything else as I have not tried it

Date: Tuesday, 7 June 2022 at 13:00

[IMG_1189]<https://user-images.githubusercontent.com/35904565/172373884-0f32120f-47a9-4386-8595-c51e32ba5235.jpg>

blb4github, 2022-06-07:
I have tried it, the only thing it resets is the WIFI SSID

matthewjporter, 2022-06-07:
Ok, Out of ideas then

Sorry

Date: Tuesday, 7 June 2022 at 14:02

blb4github, 2022-06-07:
thanks anyway. It is working for me now, but it is strange then the inverter works on address 0.

wills106, 2022-06-27:
Closing this issue down as I believe it's resolved.

If you have any further issues please raise a new issue.

closed 2022-06-27 by wills106