wills106/homeassistant-solax-modbus #78: SolaX Total Yield entity value

richsimpson, 2022-07-01:
Hi,

On my X1 G4 it is showing as MWh, I think it ought to be kWh? 

System up for less than a week, the Total Solar Energy value is 97.2 kWh but the Total Yield is showing 99.9MWh

infradom, 2022-07-01:
I can confirm this an error on Gen4 systems. I overlooked this in the past on my system.
@wills106 : not sure if the Gen3 or other models suffer from this scaling issue

wills106, 2022-07-01:
On the Gen2 / Gen3 it's scaled correctly. Although it doesn't always align correctly with the display in the Cloud.

Does the Gen4 need changing to be the same as Gen3 or a different rounding?

https://github.com/wills106/homeassistant-solax-modbus/blob/5d92a43c825a8840e8cc12246cf552a0cb4b5ccf/custom_components/solax_modbus/__init__.py#L880

infradom, 2022-07-01:
Unsure if the manual for Gen4 is correct
modbus is expressed as 2 words, and unit seems to be 0.001 kWh on Gen4

infradom, 2022-07-01:
In the current implementation, we only seem to read one of the registers. I suggest I make the correction first and test on my Gen4. The code makes already a distinction ...

richsimpson, 2022-07-01:
Not sure. I just assumed it needed to have the M change to a K for the unit of measurement.

infradom, 2022-07-01:
That would be simple, but we want some consistency with gen4 and gen2
So scaling is another solution. I would prefer kWh, but then we might become incompatible with the past for other Gen's

wills106, 2022-07-01:
Yea I would take the docs "With a Pinch of Salt", there is a number of other errors.

So the Gen4 needs to be scaled the same as the Gen3 then.

Both registers are read as a single 32bit_uint.

I had it as MWh as it matches the SolaX Cloud, also you would end up with a very long number. I am at 10.78MWh at the moment for my Total Yield. Older systems would potentially a lot higher value.

infradom, 2022-07-01:
@wills106 , agree, but a bit inconsitent with the Solar Energy Total counter, same magnitude but expressed in kWh.
Furthermore (not related) on Gen4, we need to make the calculation with both registers

infradom, 2022-07-01:
On gen3, the LSB register wont overflow as fast as on gen4, but we should use register 53 to avoid overrflows

wills106, 2022-07-01:
I can change the Solar Energy Total to MWh as well?

Both register 52 and 53 are being read for the total_yield, it's being read as a 32bit word instead of two single 16bit registers.

infradom, 2022-07-01:
Sorry, you are right, I overlooked the 32

I agree that it would be better to use MWh for both

richsimpson, 2022-07-01:
You pair know far far more about this than me, pleased to offer the issue up to be looked over, apologies I couldn't be more helpful in terms of the solution.

Great work
Rich

infradom, 2022-07-01:
@wills106 : I dont think the Gen4 needs to be scaled like the Gen3: there is a factor 0.1 for Gen3 and 0.0001 for Gen4 (manual says 0.001 but I believe this is wrong).
If you move to MWh, it would be good to have sufficient precision digits. If we change this to MWh, there is a chance that some user's calculation scripts need to be updated.

infradom, 2022-07-01:
@richsimpson Thanks for reporting this !

wills106, 2022-07-01:
If people need enough precision digit's it might be worth leaving it in kWh and putting Total Yield back into kWh?
I only put Total Yield as MWh to match the SolaX Cloud portal. I don't think Total Solar Yield is even on there? That's why I left it as the docs indicate.

I am not even sure if anyone uses either or not, as they are both disabled by default.
I could do a poll in the discussions section to see if anyone else has any thoughts?

wills106, 2022-08-10:
In 0.6.0 Beta 8 Total Yield is changed back to kWh.
Is everything else correctly now?

Just want to check before I close this down.

infradom, 2022-08-10:
Is ok now on my Gen4 hybrid, it shows the correct value in kWh

wills106, 2022-08-10:
Ok, closing issue down.

closed 2022-08-10 by wills106