wills106/homeassistant-solax-modbus #83: Gen4 X3 : modifying backup_discharge_min_soc modifies backup_nightcharge_upper_soc (firmware error ?)

infradom, 2022-07-02:
Seems like the inverter swaps both values when the backup_discharge_min_soc is modified though HA.
One would expect that modifying backup_nightcharge_upper_soc shows a similar problem, but that is not the case.
Reading both values is correct (compared to the solax app)

I believe this is not an error in our integration, but in the firmware of (some) Gen4 inverters.
The integraton seems to use the correct registers.

Dont know if we should keep this as an open issue since I believe the error is in the Solax firmware.

wills106, 2022-07-02:
I think it's worth keeping it open. Perhaps someone else who has it working correctly can share what Firmware version they are on. Then we can work out which are effected and perhaps ask for an updated Firmware from SolaX?

infradom, 2022-08-03:
I had my firmware upgraded, but the (minor) problem still remains:
Master 23
Mgr: 23
Modbus TCP 0 6
ARM: 2.03
DSP: 2.07

jan-bee, 2022-09-21:
Same in my case, when solax_backup_discharge_min_soc is updated it change solax_backup_nightcharge_upper_soc. It is not possible to change solax_backup_discharge_min_soc value from Ha integration (it works in inverter and cloud app). 

1. Integration version: 0.6.2
2. HA version: 2022.9.5
3. First 4 digits of Serial: H34T
4. Connection Method: USB RS485
5. Firmware of Solax Inverter: 1.23 1.24 (latest from Aug 2022)

johny-mnemonic, 2022-12-11:
Same here. 
Not sure whether it is something new, but what is reported in HA as `solax_backup_nightcharge_upper_soc` is called "Charge battery to %" in Solax Cloud app, which sounds more like the charging target independent on time, than just the night charge from the grid. But maybe just the Solax app is wong...

1. Integration Version: 2022.12.4
2. HA Version: 2022.12.1
3. Inverter Brand: Solax X3 Gen4 10kW
4. Plugin Used:
5. First 6 digits of Serial: H34A10
6. Connection Method: ModBus TCP using WaveShare RS485 TO ETH (B)

I am not sure how to find firmware version. Checking "About" menu on the inverter shows "DSP: 2.0.4" and "ARM: 2.0.3".

johny-mnemonic, 2023-01-16:
Any update on this @wills106? Is there a way to set  the backup discharge min SOC if the `sensor.solax_backup_discharge_min_soc` doesn't work on Gen4? Any sort of workaround?
I tried to use `solax_backup_nightcharge_upper_soc`, but that changes only itself, so it is not swapped after all...
Could I help somehow?

wills106, 2023-01-17:
They only thing wrong I can see in the Integration is that the allowed values for number.solax_backup_discharge_min_soc is 10-100%, but it should be 30-100%

Have you tried setting it to at least 30% ?
If you can't write to that register there isn't a great deal I can do about it sorry. There is probably an internal error in the mapping of registers in the Inverter.
Unfortunately this isn't the first time I have found registers to be missing on certain Inverters.

johny-mnemonic, 2023-01-17:
@wills106 you are correct about the range of 30-100%.
I can set the `solax_backup_discharge_min_soc` just fine from HA. Trouble is it updates `solax_backup_nightcharge_upper_soc` instead and `solax_backup_discharge_min_soc`stays untouched. So I think it is just about finding the correct register to set on Gen4 X3. Seems like another exception to deal with 🤷‍♂️

johny-mnemonic, 2023-01-29:
@wills106 Is there a way how I could help? Is there for example a way to list all writable registers from the inverter and try which one changes the backup discharge SOC? Or is it better to catch what the mobile/web app is doing when changing this value?

wills106, 2023-01-31:
You can scan read registers using a tool like Gmodbus.
I'm not aware of a means to scan for the writeable address though, the other problem is quite often, it reads from one location and writes to another. Unlike Sofar where it reads / writes to the same register.

The registers mentioned in the app / web portal don't match directly to Inverter registers.
Some might, but in the past I have observed they either appear random in relation or offset.

johny-mnemonic, 2023-02-06:
@wills106 thanks. I have played a bit with it, but ended with the same result as you predicted.
So I contacted Solax support and the reply looks like it is a known FW bug:
```
Please upgrade the inverter ARM firmware version to V1.27 and it will fix the problem.
```
Will let you know whether the upgrade helped so we could close this issue.

johny-mnemonic, 2023-02-08:
@wills106 I have updated the inverter and it now works. So the FW bug confirmed and you can close this issue. Thanks

wills106, 2023-02-08:
Glad you got this resolved.
@infradom @jan-bee might be worth contacting SolaX to see if you can get this version of Firmware update?

closed 2023-02-08 by wills106