wills106/homeassistant-solax-modbus #84: Latest Home Assistant 2022.7 - Just flagging

richsimpson, 2022-07-07:
Sure you will have seen all of this from the latest HASS update
[image]

At least I assume these are new, not seen them before.

infradom, 2022-07-07:
> Sure you will have seen all of this from the latest HASS update [image]
> 
> At least I assume these are new, not seen them before.

Thanks, I had not yet upgraded.
Have not looked into the details, but I believe it should be relatively easy to fix ...

tomcoleman, 2022-07-21:
also seeing this

wills106, 2022-07-25:
Release 0.5.9 fixes this issue.

closed 2022-07-25 by wills106

matthewjporter, 2022-07-25:
Getting errors with the latest 5.9 update

022-07-25 12:38:12 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 397, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 641, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 776, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 532, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 570, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 538, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 388, in state
    value = self.native_value
  File "/config/custom_components/solax_modbus/sensor.py", line 93, in native_value
    self._hub.data[self.entity_description.key]*self._attr_scale
TypeError: unsupported operand type(s) for *: 'datetime.datetime' and 'int'

matthewjporter, 2022-07-25:
Just rolling back the version to see what happens. Remove the custom_component folder and re-download, reboot

Currently back to 5.6 and same error messages, I have not updated for sometime as non of the changes are for my inverters so not sure where it may have crept in

matthewjporter, 2022-07-25:
> Just rolling back the version to see what happens. Remove the custom_component folder and re-download, reboot
> 
> Currently back to 5.6 and same error messages, I have not updated for sometime as non of the changes are for my inverters so not sure where it may have crept in

Back to 5.5 and same thing

Going to try complete removal off all folders, integrations and re-do

matthewjporter, 2022-07-25:
5.1, No HACS install, just straight copy to custom_component folder, No to move upwards

wills106, 2022-07-25:
@matthewjporter what version of HA are you on?

These changes came about because of 2022.7 I am just wondering if you are on say 2022.6.x these changes for 2022.7 have broken 2022.6?

matthewjporter, 2022-07-25:
> 
On 2022.7.6

Manually installed all the way to 5.8, When I do 5.9 get the reported errors

Have done a clean removal of all inverters/integrations and folders, rebooted, add in the folder for 5.9, rebooted and added integration for all inverters - no sensors

matthewjporter, 2022-07-26:
All fixed with 5.9a and now on 5.10

Thanks