wills106/homeassistant-solax-modbus #85: X3-Hybrid-10.0-D Gen 4 - "unrecognized inverter type - serial number : unknown"

jamespreedy, 2022-07-15:
Hi @wills106 - thanks for your work on this custom component, really excited to get it up and running.
Getting "unrecognized inverter type - serial number : unknown" - so posting my details here.
I posted this in the discussion group previously, but wasn't sure if this would count as an issue too.

X3-Hybrid-10.0-D
10kW
Three Phase
Dual MPPT
EPS - Yes External (I have EPS, how do I tell if it's External?)
Gen4
Serial: H34A

Thank you, please let me know if you need anything else! 🙏🏻

wills106, 2022-07-15:
How are you connecting to your Gen4? RS485 - USB or a device like the Waveshare?
As the H34 is already supported.

jamespreedy, 2022-07-15:
Ah interesting! I followed the instructions on here ...
https://github.com/wills106/homeassistant-solax-modbus/wiki/Gen4-Installation-Notes

I went with "Option 1: Waveshare RS485 to ETH (B): (DIN rail mounted model)"

I'll go back and see if perhaps I've missed something then - thank you @wills106 - really appreciate it. 😀

infradom, 2022-07-15:
The 'unknown' serial number indicates indeed that something must be wrong with the initial cummunication in which the software tries to fetch the serial number.
So make sure baudrate matches and watch the LED communication on the waveshare (see Wiki)

wills106, 2022-07-24:
@jamespreedy did you manage to get your Inverter working correctly?

jamespreedy, 2022-07-24:
@wills106 

Hiya! Yes ... thats perfect timing haha. I got it up and running last night, and have my data flowing into Home Assistant this morning. It was indeed user error - when I had re-flashed the webpage onto the Waveshare converter, I'd added an MQTT config in error. I eventually worked it out when I noticed that the command prompt over telnet wasn't right. Removing the special config sorted it. It's set to 19200bps, refreshing in HA every 5 seconds.

Thank you so much for your work on this, it's such a game changer to have up to data data and not rely on the cloud.

Just sent over a couple of coffees on Ko-Fi - hopefully that'll pay for a beer instead - thank you again. 🙏🏻🍺

closed 2022-07-24 by jamespreedy

wills106, 2022-07-24:
Glad you got it running in the end.