wills106/homeassistant-solax-modbus #88: X3-MIC/PRO

CristiXP, 2022-07-24:
Does this integration work with the X3-MIC/PRO as I have installed it, and I get unknown everywhere. Any ideea how to fix this?

Thank you,

Cristian

wills106, 2022-07-24:
There are other X3 MIC / Pro MIC users, but no one has confirmed if it works or not.

Does your Inverter Serial Number start MC10 or MU80?
If yours does start with one of them Serial Numbers then it confirms the serial Number is read at a different location to the other Inverters. So this Integration will require further work to support your Inverter.

wills106, 2022-07-24:
If your Inverter is a MC10 or MU80 can you try the built in Modbus component in HA and let me know if the below returns your Serial Number or not?

```
modbus:
  name: SolaX 
  type: serial # Comment out for Modbus over TCP
  method: rtu # Comment out for Modbus over TCP
  port: /dev/ttyUSB0 # Comment out for Modbus over TCP
  baudrate: 9600 # Comment out for Modbus over TCP
  stopbits: 1 # Comment out for Modbus over TCP
  bytesize: 8 # Comment out for Modbus over TCP
  parity: N # Comment out for Modbus over TCP

# Uncomment for Modbus over TCP
  # type: tcp
  # host: !secret inverter_ip
  # port: 502

  sensors:
  # Holding Registers
    - name: SolaX Serial Number
      address: 0x300
      count: 7
      data_type: string
      scan_interval: 5
```

CristiXP, 2022-07-24:
Hello, thank you for the fast response.
My inverter starts with this MP153TGAXXXXXX I have put this, but still shows unavailable :(
I am available to test. The serial connection is working, as I have used it to retrieve data from the SDM630 meter. Now I am on pin 4 and pin 5 and try to get data from inverter, but with no luck.
modbus:
  - name: SolaX
    type: serial
    baudrate: 9600
    bytesize: 8
    method: rtu
    parity: N
    port: /dev/ttyUSB0
    stopbits: 1
sensors:
- name: SolaX Serial Number
        address: 0x300
        count: 7
        data_type: string
        scan_interval: 5

wills106, 2022-07-24:
Can you try the following addresses as well?

address: 0

and

address: 768

Have you tried swapping pin 4 & 5 around?

CristiXP, 2022-07-24:
When I am using modpoll with wires like this, I am getting the Illegal data address exception message. When I am swapping them i receive an Checksum error message... so I think this is the wrong way on the pins.

Tried with 0 and 768 on the adress in pin 4 and 5 and swapped... still nothing :(

wills106, 2022-07-24:
Do you have a full example of what you are sending to the Inverter with Modpoll?

wills106, 2022-07-24:
> I get unknown everywhere. Any ideea how to fix this?

In your error log did you get:
unrecognized inverter type - serial number : MP153TGAXXXXXX

infradom, 2022-07-24:
If you can get this log message, the modbus communication looks fine. We may need to add your model ...

CristiXP, 2022-07-24:
This is what I am sending with modpoll /config/modpoll -m rtu -1 -a 1 -b 9600 -d 8 -p none -r 1 /dev/ttyUSB
I am only getting that the serial number is unknown.
[image]
0

infradom, 2022-07-24:
Are you sure the baudrate for your model is 9600?

CristiXP, 2022-07-24:
> Are you sure the baudrate for your model is 9600?

When trying other baud rate, I have a Reply timeout error. When going on 9600 I get an illegal adress exception message.

CristiXP, 2022-07-24:
In this thread someone stated that has fixed this for modbus, and that the X3 PRO is different than others... but no ideea what needs to be done to fix this. https://gathering.tweakers.net/forum/list_messages/2104298

wills106, 2022-07-24:
The Modbus docs mention Baud: 9600
I take it you are using a USB - RS485 adaptor for now and not a RS485 - Waveshare?

modpoll -b 9600 -p none -m rtu -a 1 -r 769 -c 7 /dev/ttyUSB

Does that return anything?

CristiXP, 2022-07-24:
> The Modbus docs mention Baud: 9600 I take it you are using a USB - RS485 adaptor for now and not a RS485 - Waveshare?
> 
> modpoll -b 9600 -p none -m rtu -a 1 -r 769 -c 7 /dev/ttyUSB
> 
> Does that return anything?
> 
> Edit: Or try
> 
> modpoll -b 9600 -p none -m rtu -a 1 -t 1 -r 1025 -c 7 /dev/ttyUSB
[image]

CristiXP, 2022-07-24:
1025 gives Illegal data address exception.

infradom, 2022-07-24:
> > The Modbus docs mention Baud: 9600 I take it you are using a USB - RS485 adaptor for now and not a RS485 - Waveshare?
> > modpoll -b 9600 -p none -m rtu -a 1 -r 769 -c 7 /dev/ttyUSB
> > Does that return anything?
> > Edit: Or try
> > modpoll -b 9600 -p none -m rtu -a 1 -t 1 -r 1025 -c 7 /dev/ttyUSB
> 
> [image]

This anwer seems to be the correct serial numbeer  MP15...

CristiXP, 2022-07-24:
I am trying address 0x301 right now... let me check

wills106, 2022-07-24:
Yea just need to work out why the second one doesn't work. Never used Modpoll before.

infradom, 2022-07-24:
The received answer for 769 was correct : I decoded it to MP15 ...

CristiXP, 2022-07-24:
Still unavailable for 0x301

- name: SolaX SN
        address: 0x301
        count: 7
        data_type: string
        scan_interval: 5

wills106, 2022-07-24:
Well I don't think getting modpoll -b 9600 -p none -m rtu -a 1 -t 1 -r 1025 -c 7 /dev/ttyUSB to work doesn't matter.

We know the Serial is at 0x300 and the registers should be at 0x400.

So just need to make the Integration try to look for a serial at 0x300 if 0x0 returns blank.

wills106, 2022-07-24:
If you are using the built in Modbus try:

```
  sensors:
  # Holding Registers
    - name: SolaX Serial Number
      address: 769
      count: 7
      data_type: string
      scan_interval: 5
```

infradom, 2022-07-24:
But in this case, the serial number seems to be at 0x301 (769 decimal)
Am I missing something ?

wills106, 2022-07-24:
Modpoll starts at 1
PyModbus starts at 0

infradom, 2022-07-24:
ok, thanks

wills106, 2022-07-24:
I have used a GUI based one on Windows and that starts at 1 which is annoying as you have to offset everything by +1

infradom, 2022-07-24:
> > I get unknown everywhere. Any ideea how to fix this?
> 
> In your error log did you get: unrecognized inverter type - serial number : MP153TGAXXXXXX
> 
> Or any other errors?

Do you see anything of this kind in the logs, as the serial number can be read now ...

CristiXP, 2022-07-24:
> If you are using the built in Modbus try:
> 
> 
> 
> ```
> 
>   sensors:
> 
>   # Holding Registers
> 
>     - name: SolaX Serial Number
> 
>       address: 769
> 
>       count: 7
> 
>       data_type: string
> 
>       scan_interval: 5
> 
> ```

Still unavailble it shows.

infradom, 2022-07-24:
strange ...

infradom, 2022-07-24:
Sorry I deleted your message by accident, wanted to delete mine

infradom, 2022-07-24:
Maybe the stringlength is not 7 for this model which makes the checksum fail. If we read it as numbers with modpoll, it does not complain ...

CristiXP, 2022-07-24:
[image]
[image]

wills106, 2022-07-24:
I have not used the built in Modbus so maybe group reading has changed

infradom, 2022-07-24:
What happens if you try with other stringlengths: 14, 6,7,8, 16, 18 ?

wills106, 2022-07-24:
Does the below return your PV1 and PV2 DC Voltages?

```
  sensors:
  # Holding Registers
    - name: SolaX PV1
      address: 0x400
      input_type: input
      count: 1
      data_type: uint16
      scale: 0.1
      scan_interval: 5
    - name: SolaX PV2
      address: 0x401
      input_type: input
      count: 1
      data_type: uint16
      scale: 0.1
      scan_interval: 5
```

wills106, 2022-07-24:
I also think HA Modbus starts at 0 so the following should return your Serial?

```
  sensors:
  # Holding Registers
    - name: SolaX Serial Number
      address: 768
      count: 7
      data_type: string
      scan_interval: 5
```

As long as it is 7 characters long?

CristiXP, 2022-07-24:
> Does the below return your PV1 and PV2 DC Voltages?
>
> ```
>   sensors:
>   # Holding Registers
>     - name: SolaX PV1
>       address: 0x400
>       input_type: input
>       count: 1
>       data_type: [...]

both unavailable. Also when modpoll for 1024 and 1025 get Illegal address exception.

infradom, 2022-07-24:
> I also think HA Modbus starts at 0 so the following should return your Serial?
> 
> ```
>   sensors:
>   # Holding Registers
>     - name: SolaX Serial Number
>       address: 768
>       count: 7
>       data_type: string
>       scan_interval: 5
> ```
> 
> As long as it is 7 characters long?

7 characters or 7 words? Our serial numbers are normally 14 characters ... I am not sure how the standard modbus integration treats string lengths

wills106, 2022-07-24:
> 7 characters or 7 words? Our serial numbers are normally 14 characters ... I am not sure how the standard modbus integration treats string lengths

I'm getting myself confused! It's 14 Chars, but over 7 Registers. We only need to read the first couple of registers anyway.

wills106, 2022-07-24:
> both unavailable. Also when modpoll for 1024 and 1025 get Illegal address exception.

Have you tried 1024 and 1025 with the Built in Modbus instead of the 0x400 and 0x401 they may have removed the ability to use Hex values?

CristiXP, 2022-07-24:
Yes, 1024 and 1025 and 0x400 and 0x401, same story. Only thing which is giving data, is when modpoll 769
[image]
[image]
.

wills106, 2022-07-24:
```
  sensors:
  # Holding Registers
    - name: SolaX Group i1
      address: 1024
      input_type: input
      count: 14
      data_type: custom
      structure: ">14H"
      scan_interval: 5
```

Does that return 14 sets of values comma separated?

infradom, 2022-07-24:
> > 7 characters or 7 words? Our serial numbers are normally 14 characters ... I am not sure how the standard modbus integration treats string lengths
> 
> I'm getting myself confused! It's 14 Chars, but over 7 Registers. We only need to read the first couple of registers anyway.

Indeed, but the modbus library we use might refuse the shorter substring because a CRC checksum is not correct anymore. Sorry I am following this discussion while doing other work, so it may not have my full attention ..

wills106, 2022-07-24:
For your SolaX Serial you need the count to be 7 not 14 assuming your serial is 14 chars long. If there is nothing after the 7 register like on the X1AC it will give an error.

CristiXP, 2022-07-24:
> ```
>   sensors:
>   # Input Registers
>     - name: SolaX Group i1
>       address: 1024
>       input_type: input
>       count: 14
>       data_type: custom
>       structure: ">14H"
>       scan_interval: 5
> ```
> 
> 
>      
>   
> 
> Does that return 14 sets of values comma separated?

it doesn't load the entity at all... Cannot find it. so something wrong in the syntax maybe.

wills106, 2022-07-24:
When in the Developer Tools Page, if you press check configuration does it give any errors?

CristiXP, 2022-07-24:
No errors. It was because I still had the PV1 on the same 1024 address
[image]

[image]

wills106, 2022-07-24:
Can your copy and paste your complete code for modbus. Don't forget to wrap the code with <>

Also do you still have an active session of Modpoll running at the same time as HA? It might be blocking the access.

CristiXP, 2022-07-24:
I am using only your suggestions. 
`modbus:
  - name: SolaX
    type: serial
    baudrate: 9600
    bytesize: 8
    method: rtu
    parity: N
    port: /dev/ttyUSB0
    stopbits: 1

    sensors:
        
    - name: SolaX Serial Number
      address: 768
      count: 14
      data_type: string
      scan_interval: 5
      
    - name: SolaX Group i1
      address: 1025
      input_type: input
      count: 14
      data_type: custom
      structure: ">14H"
      scan_interval: 5`

And this is an example of  entity for the SDM630 smart metter that worked before:

`- name: sdm630_phase_1_line_to_neutral_volts
        slave: 1
        address: 0
        input_type: input
        count: 2
        precision: 2
        data_type: float32
        unit_of_measurement: V
        device_class: voltage
        scan_interval: 60`

wills106, 2022-07-24:
I have not used the built in one in ages.
Does the following work?

```modbus:
  - name: SolaX
    type: serial
    baudrate: 9600
    bytesize: 8
    method: rtu
    parity: N
    port: /dev/ttyUSB0
    stopbits: 1

    sensors:
        
    - name: SolaX Serial Number
      slave: 1
      address: 768
      count: 7
      data_type: string
      scan_interval: 5
      
    - name: SolaX Group i1
      slave: 1
      address: 1024
      input_type: input
      count: 14
      data_type: custom
      structure: ">14H"
      scan_interval: 5
```

CristiXP, 2022-07-24:
> I have not used the built in one in ages. Does the following work?
> 
> ```
>   - name: SolaX
>     type: serial
>     baudrate: 9600
>     bytesize: 8
>     method: rtu
>     parity: N
>     port: /dev/ttyUSB0
>     stopbits: 1
> 
>     sensors:
>         
>     - name: SolaX Serial Number
>       slave: 1
>       address: 768
>       count: 7
>       data_type: string
>       scan_interval: 5
>       
>     - name: SolaX Group i1
>       slave: 1
>       address: 1024
>       input_type: input
>       count: 14
>       data_type: custom
>       structure: ">14H"
>       scan_interval: 5
> ```

No, but this started showing up some info

- name: SolaX Serial Number
      slave: 1
      address: 769
      count: 7
      data_type: string
      scan_interval: 5
      unit_of_measurement: V
      device_class: voltage
[image]

CristiXP, 2022-07-24:
> I have not used the built in one in ages. Does the following work?
>
> ```
>   - name: SolaX
>     type: serial
>     baudrate: 9600
>     bytesize: 8
>     method: rtu
>     parity: N
>     port: /dev/ttyUSB0 [...]
[image]

CristiXP, 2022-07-24:
For the serial number, it should read MP not PM... 15 not 51... it's backwards.

wills106, 2022-07-24:
Looks like the serial number is reversing each char set.
If you add the following template do the values match your Inverter?

```
sensor:
- platform: template
  sensors:
# SolaX Group I1
    solax_pv_voltage_1:
      friendly_name: "SolaX PV Voltage 1"
      unit_of_measurement: V
      value_template: "{{ states('sensor.solax_group_i1').split(',')[0]| float / 10 }}"
    solax_pv_voltage_2:
      friendly_name: "SolaX PV Voltage 2"
      unit_of_measurement: V
      value_template: "{{ states('sensor.solax_group_i1').split(',')[1]| float / 10 }}"
    solax_pv_current_1:
      friendly_name: "SolaX PV Current 1"
      unit_of_measurement: A
      value_template: "{{ states('sensor.solax_group_i1').split(',')[2]| float / 10 }}"
    solax_pv_current_2:
      friendly_name: "SolaX PV Current 2"
      unit_of_measurement: A
      value_template: "{{ states('sensor.solax_group_i1').split(',')[3]| float / 10 }}"
    solax_inverter_temperature:
      friendly_name: "SolaX Inverter Temperature"
      unit_of_measurement: °C
      value_template: "{{ states('sensor.solax_group_i1').split(',')[13] }}"
```

wills106, 2022-07-24:
Also does this correct your Serial?

```
  - name: SolaX
    type: serial
    baudrate: 9600
    bytesize: 8
    method: rtu
    parity: N
    port: /dev/ttyUSB0
    stopbits: 1

    sensors:
        
    - name: SolaX Serial Number
      slave: 1
      address: 768
      count: 7
      data_type: string
      swap: byte
      scan_interval: 5
      
    - name: SolaX Group i1
      slave: 1
      address: 1024
      input_type: input
      count: 14
      data_type: custom
      structure: ">14H"
      scan_interval: 5
```

CristiXP, 2022-07-24:
Any ideea where should I add them exactly? This is from my config yaml.

sensor:  
    - platform: rest
      resource: https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId=20210503024031531570543&sn=SWERBSDSBP
      value_template: '{{ ( value_json.result.yieldtotal | round(0) ) }}'
      name: "Solax total"
      unit_of_measurement: 'kWh'
      scan_interval: 3600
    
    - platform: rest
      resource: https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId=20210503024031531570543&sn=SWERBSDSBP
      value_template: '{{ ( value_json.result.acpower | round(0) ) }}'
      name: "Solax now"
      unit_of_measurement: "W"
      scan_interval: 65
    - platform: rest
      resource: https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId=20210503024031531570543&sn=SWERBSDSBP
      value_template: '{{ ( value_json.result.powerdc1 | round(0) ) }}'
      name: "Terasa"
      unit_of_measurement: "W"
      scan_interval: 300
    
    - platform: rest
      resource: https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId=20210503024031531570543&sn=SWERBSDSBP
      value_template: '{{ ( value_json.result.powerdc2 | round(0) ) }}'
      name: "Acoperis"
      unit_of_measurement: "W"
      scan_interval: 300
      
    - platform: rest
      resource: https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId=20210503024031531570543&sn=SWERBSDSBP
      value_template: '{{ value_json.result.feedinpower }}'
      name: "Feed In Power"
      unit_of_measurement: "W"
      scan_interval: 65

    - platform: rest
      resource: https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId=20210503024031531570543&sn=SWERBSDSBP
      value_template: '{{ value_json.result.feedinenergy }}'
      name: "Feed In Energy"
      unit_of_measurement: "KWh"
      scan_interval: 3600
    
    - platform: rest
      resource: https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId=20210503024031531570543&sn=SWERBSDSBP
      value_template: '{{ value_json.result.consumeenergy }}'
      name: "Consumed Energy"
      unit_of_measurement: "KWh"
      scan_interval: 3600
    
    - platform: rest
      resource: https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId=20210503024031531570543&sn=SWERBSDSBP
      value_template: '{{ value_json.result.yieldtoday }}'
      name: "Solax today"
      unit_of_measurement: "KWh"
      scan_interval: 3600

    
    - platform: rest
      resource: https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId=20210503024031531570543&sn=SWERBSDSBP
      value_template: "{{ value_json.result.uploadTime }}"
      name: "Solax upload time"
      scan_interval: 300
    
    - platform: rest
      resource: https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId=20210503024031531570543&sn=SWERBSDSBP
      value_template: >
          {% if value_json.result.inverterStatus == '100' %}Wait
          {% elif value_json.result.inverterStatus == '101' %}Check
[... 90 lines cut ...]
    baudrate: 9600
    bytesize: 8
    method: rtu
    parity: N
    port: /dev/ttyUSB0
    stopbits: 1

    sensors:
        
    - name: SolaX Serial Number
      slave: 1
      address: 768
      count: 7
      data_type: string
      swap: byte
      scan_interval: 5
      
    - name: SolaX Group i1
      slave: 1
      address: 1024
      input_type: input
      count: 14
      data_type: custom
      structure: ">14H"
      scan_interval: 5

CristiXP, 2022-07-24:
> Also does this correct your Serial?
>
> ```
>   - name: SolaX
>     type: serial
>     baudrate: 9600
>     bytesize: 8
>     method: rtu
>     parity: N
>     port: /dev/ttyUSB0
>     stopbits: 1 [...]

Serial corrected.

wills106, 2022-07-24:
You shouldn't need to declare each one as a sensor, they should inherit the main sensor
```
template:
  - sensor:
    - name: energy_solax_today
      state: "{{ states('sensor.solax_today') }}" 
      unit_of_measurement: 'kWh'
      state_class: total_increasing
      device_class: energy

    - name: energy_total
      state: "{{ states('sensor.solax_total') }}" 
      unit_of_measurement: 'kWh'
      state_class: total_increasing
      device_class: energy

    - name: energy_consumed_energy
      state: "{{ states('sensor.consumed_energy') }}" 
      unit_of_measurement: 'kWh'
      state_class: total_increasing
      device_class: energy

    - name: energy_feed_in_energy
      state: "{{ states('sensor.feed_in_energy') }}" 
      unit_of_measurement: 'kWh'
      state_class: total_increasing
      device_class: energy
      
    - name: energy_car
      state: "{{ states('sensor.tesla_model_3_energy_added_sensor')}}" 
      unit_of_measurement: 'kWh'
      state_class: total_increasing
      device_class: energy

    - name: solax_pv_voltage_1:
      friendly_name: "SolaX PV Voltage 1"
      unit_of_measurement: V
      value_template: "{{ states('sensor.solax_group_i1').split(',')[0]| float / 10 }}"

    - name: solax_pv_voltage_2:
      friendly_name: "SolaX PV Voltage 2"
      unit_of_measurement: V
      value_template: "{{ states('sensor.solax_group_i1').split(',')[1]| float / 10 }}"

    - name: solax_pv_current_1:
      friendly_name: "SolaX PV Current 1"
      unit_of_measurement: A
      value_template: "{{ states('sensor.solax_group_i1').split(',')[2]| float / 10 }}"

    - name: solax_pv_current_2:
      friendly_name: "SolaX PV Current 2"
      unit_of_measurement: A
      value_template: "{{ states('sensor.solax_group_i1').split(',')[3]| float / 10 }}"

    - name: solax_inverter_temperature:
      friendly_name: "SolaX Inverter Temperature"
      unit_of_measurement: °C
      value_template: "{{ states('sensor.solax_group_i1').split(',')[13] }}"

```

CristiXP, 2022-07-24:
With the part of my template, it is working. But when I am adding your part, I am receiving an error:

```
template:
  - sensor:
    - name: energy_solax_today
      state: "{{ states('sensor.solax_today') }}" 
      unit_of_measurement: 'kWh'
      state_class: total_increasing
      device_class: energy

    - name: energy_total
      state: "{{ states('sensor.solax_total') }}" 
      unit_of_measurement: 'kWh'
      state_class: total_increasing
      device_class: energy

    - name: energy_consumed_energy
      state: "{{ states('sensor.consumed_energy') }}" 
      unit_of_measurement: 'kWh'
      state_class: total_increasing
      device_class: energy

    - name: energy_feed_in_energy
      state: "{{ states('sensor.feed_in_energy') }}" 
      unit_of_measurement: 'kWh'
      state_class: total_increasing
      device_class: energy
      
    - name: energy_car
      state: "{{ states('sensor.tesla_model_3_energy_added_sensor')}}" 
      unit_of_measurement: 'kWh'
      state_class: total_increasing
      device_class: energy
```

When I add this it brakes it

```
- name: solax_pv_voltage_1:
      friendly_name: "SolaX PV Voltage 1"
      unit_of_measurement: V
      value_template: "{{ states('sensor.solax_group_i1').split(',')[0]| float / 10 }}"
```
[image]

wills106, 2022-07-24:
Sorry remove the colon ( : ) at the end of each name.

- name: solax_pv_voltage_1: 
should be
- name: solax_pv_voltage_1

- name: solax_pv_voltage_2:
- should be
- name: solax_pv_voltage_2

etc

[161 more replies]

closed 2022-08-22 by wills106