wills106/homeassistant-solax-modbus #89: X1-Hybrid-G3 e-charge today wrong value

Claudio1L, 2022-07-24:
hi

the sensor "sensor.solax_e_charge_today" seems providing wrong value.

It shows about 80 kwh per day, when should be no more than 7-8 kwh

[image]

Claudio1L, 2022-07-24:
comparing between the E_charge today and Battery input_energy today, it seems a scaling (10x) issue.

￼  ￼

wills106, 2022-07-25:
I can't even remember what E Charge is meant to be for?

Just Turned mine back on as it's off by default. Does yours only increase when you grid charge? As mine is just sat at 0, but I haven't charged from the grid in months.

Claudio1L, 2022-07-25:
E Charge should be the kWh provided to the battery inverter to charge the battery (AC side). Battery input Energy should be the same in DC, not including the lost efficiency of the inverter. Same thing for the Solax today Yield (AC side) vs Battery output energy (DC side)- That's my understanding and seems confirmed by the similar value and trends, with a small difference given by the inverter loss.

Claudio1L, 2022-07-25:
It is also confirmed by the difference between the "e Charge total" vs the "e charge today". About 8 kwh variation in the total vs the 80 kwh in the today graph

wills106, 2022-07-26:
Should be fixed in [0.5.10](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/0.5.10)

wills106, 2022-07-26:
My E Charge only increases when I force charge from the Grid. It doesn't increase from any PV diverted power going into the battery.

So for the Energy dashboard I don't know how useful it is? Unless you are using it as a separate consumption device only.

Claudio1L, 2022-07-26:
I see charge increase perhaps because I have different inverters for PV and Battery, so my charge will come always from "outside" regardless if PV or Grid

Claudio1L, 2022-07-26:
seems ok with 0.5.11-Beta

wills106, 2022-07-26:
Closing as release [0.5.11](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/0.5.11) fixes issue.

closed 2022-07-26 by wills106