wills106/homeassistant-solax-modbus #91: TypeError: unsupported operand type(s) for: 'datetime.datetime' and 'int'

wills106, 2022-07-26:
Seems like HA 2022.7 has introduced a change for Time sensors.
If you turn "RTC" on you get the following error.

TypeError: unsupported operand type(s) for: 'datetime.datetime' and 'int'

infradom, 2022-07-26:
I will have a look at this ... I still believe this is unrelated to HA2022.7
I am still on previous version, but will test with both

wills106, 2022-07-26:
This is the full error:

```
2022-07-26 09:07:02 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/init.py", line 228, in async_refresh_modbus_data
    update_callback()
  File "/config/custom_components/solax_modbus/sensor.py", line 73, in _modbus_data_updated
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 532, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 570, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 538, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/init.py", line 388, in state
    value = self.native_value
  File "/config/custom_components/solax_modbus/sensor.py", line 93, in native_value
    self._hub.data[self.entity_description.key]self._attr_scale
TypeError: unsupported operand type(s) for: 'datetime.datetime' and 'int'
```

I have noticed the following change to the built in sensor within HA.
There is now an addition of ` | Decimal:` at the end of the native_value

https://github.com/home-assistant/core/commit/981249d330b9605b878bdf87195b8cb31ca8421e

It might be unrelated though?

wills106, 2022-07-26:
Just noticed there is a red X against that comit. So that might not be the cause.

Is it because it's returning an integer?

I don't know how long rtc has not been working, as personally I don't use it. But it was reported to me.

infradom, 2022-07-26:
> Seems like HA 2022.7 has introduced a change for Time sensors. If you turn "RTC" on you get the following error.
> 
> TypeError: unsupported operand type(s) for: 'datetime.datetime' and 'int'

Trying to reproduce this, but not sure I understand what you mean by "if you turn RTC on" ...
How do I do this (or what does this mean) ?

wills106, 2022-07-26:
Sorry I meant sensor.solax_rtc it's normally disabled by default.

infradom, 2022-07-26:
I believe the sensor code should be enhanced (not yet tested):
```
    @property
    def native_value(self):
        """Return the state of the sensor."""
        try: # works only for float (and integer) sensorss
            scaled = self._hub.data[self.entity_description.key]*self._attr_scale
            if self.entity_description.key in self._hub.data
            else None
        except: # strings and other data types cannot be scaled
            scaled = self._hub.data[self.entity_description.key]
            if self.entity_description.key in self._hub.data
            else None
        return  scaled
```

wills106, 2022-07-26:
I'll try it

infradom, 2022-07-26:
or maybe simpler:
```

    @property
    def native_value(self):
        """Return the state of the sensor."""
        if self._attr_scale != 1:
            return self._hub.data[self.entity_description.key]*self._attr_scale
            if self.entity_description.key in self._hub.data
            else None
        else: # strings and other data types cannot be scaled
            return self._hub.data[self.entity_description.key]
            if self.entity_description.key in self._hub.data
            else None

```

wills106, 2022-07-26:
I have managed to get the following to work:

```
    @property
    def native_value(self):
        """Return the state of the sensor."""
        if self._attr_scale != 1:
            return self._hub.data[self.entity_description.key]*self._attr_scale
        else: # strings and other data types cannot be scaled
            return self._hub.data[self.entity_description.key]
```

On restarting HA I get the following error:

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 397, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 641, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 776, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 532, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 570, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 538, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 388, in state
    value = self.native_value
  File "/config/custom_components/solax_modbus/sensor.py", line 98, in native_value
    return self._hub.data[self.entity_description.key]
KeyError: 'battery_capacity_charge'
```

But it doesn't seem to effect anything though? The battery sensor functions as normal.
sensor.solax_rtc functions normally.

infradom, 2022-07-26:
I believe the if  self.entity_description.key in self._hub.data condition remains necessary,  but I probably forgot the parentehesis to group things together: So it should be:
```
    @property
    def native_value(self):
        """Return the state of the sensor."""
        if self._attr_scale != 1:
            return (
                self._hub.data[self.entity_description.key]*self._attr_scale
                if self.entity_description.key in self._hub.data
                else None
            )
        else:
            return (
                self._hub.data[self.entity_description.key]
                if self.entity_description.key in self._hub.data
                else None
            )
```

infradom, 2022-07-26:
yes, this is equivalent and is better for readability. I however still dont understand why battery_charge_capacity would not be in _hub.data . Strange that it works ...

infradom, 2022-07-26:
Installed 0.5.11a but it still contains the faulty code (manifest.json says 0.5.11)
sensor.py still contains the faulty code on github
Can you verify once again ?

wills106, 2022-07-26:
I'll have a look shortly and use the proper Desktop App when I get home.

wills106, 2022-07-26:
Hopefully I have sorted the mess out now 😔

closed 2022-07-27 by wills106