wills106/homeassistant-solax-modbus #92: 0.5.11 error installing

Claudio1L, 2022-07-26:
I'm back :-)

tried upgrade from 0.5.11 beta and gived me error, tried clean install removing everything from HA integration and HACS, during reinstalling given me the following:

[image]

Redownloading the beta works fine. The log is reporting the following:

Logger: homeassistant.setup
Source: setup.py:162
First occurred: 15:36:05 (1 occurrences)
Last logged: 15:36:05

Setup failed for custom integration solax_modbus: Unable to import component: cannot import name 'ATTR_MANUFACTURER' from partially initialized module 'custom_components.solax_modbus.const' (most likely due to a circular import) (/config/custom_components/solax_modbus/const.py)

wills106, 2022-07-26:
0.5.11a Fixes this, I don't have access to GitHub Desktop at the moment and the bottom of const.py got chopped of some how.

wills106, 2022-07-26:
Just had another look at const.py for 0.5.11, it's actually missing the middle of the file.
Must of had a clipboard error / corruption while manually copying the code over.

Claudio1L, 2022-07-26:
it works now :-) thank you for the prompt solution.

wills106, 2022-07-26:
I'll leave this open for a couple of hours in case anyone else installed 0.5.11 before I released 0.5.11a
Then I'll close the issue down.

wills106, 2022-07-26:
Anyone reading this, goto 0.5.12, 0.5.11a didn't work...

closed 2022-07-27 by wills106