wills106/homeassistant-solax-modbus #93: integration completely broken after upgrade

gelse, 2022-08-01:
i upgraded the integration via HACS today a few minor versions (to: 0.5.12 from: 0.5.something_sorry_dont_remember) which completely broke it.

what i did (i try to remember):
- upgrade to 0.5.12
- integration was broken (did not load on startup)
    - __init__() got an unexpected keyword argument 'native_min_value' in the log files
- tried to remove the integration, which failed with the same error
- tried to remove the configuration via GUI, which failed with the same error
- manually removed the integration via manipulating the .storage folder according to some other post here
- removed the solax folder in custom_components
- removed the configuration via GUI
- re-installed the integration via HACS (worked)
- tried to add new configuration via GUI which again failed with the same error as above
all of the above with a few restarts/reloads of home assistant in between.

now i am out of ideas...

system information:
home assistant running in docker
Solax Inverter X3-Hybrid-8.0-D with Modbus-over-TCP

is there anything else of information that i can provide ?

gelse, 2022-08-01:
from what i see it as something to do with NumberEntityDescription from homeassistand itself ... wtf...
i am using core 2022.5.5

gelse, 2022-08-01:
switching to the linuxserver.io image fixed the issue (but opened new, different problems with other components).

guys, if i am not completely mistaken and somehow missed the stable tagged image - this is not cool. and yes, i know, but i did NOT check the "show beta versions" flag when installing the solax_modbus integration.

gelse, 2022-08-01:
relevant issue: https://github.com/wills106/homeassistant-solax-modbus/issues/84

gelse, 2022-08-01:
additional information:
the image i used was
ghcr.io/home-assistant/home-assistant:stable
which - from whatever reason - does not seem to exist any more, despite being the one that gets promoted in the home-assistant installation documentation.

man, they fucked that up ...

and you are completely innocent and i will now close this issue.

closed 2022-08-01 by gelse

wills106, 2022-08-01:
You are correct Release [0.5.9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/0.5.9) Introduced the change to comply with the refactoring that was needed to comply with HA 2022.7 (With the old formatting being dropped by HA 2022.10)

Release [0.5.11](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/0.5.11) was needed to address another issue relating to sensors with scaling, I think this is another issue related to HA 2022.7

Release [0.5.12](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/0.5.12) was needed to clean-up the codebase from 0.5.11 & 0.5.11a due to a number of issues relating to how the code got updated to GitHub.

wills106, 2022-08-01:
I have also added a section to the README.md under Known Issues.