wills106/homeassistant-solax-modbus #94: GEN 1 doesn’t work with version after 0.5.7

blb4github, 2022-08-02:
I updated to the latest version (0.5.12) and I didn’t get any data from my GEN 1 inverter anymore. After this I tried earlier versions, but only version 0.5.7 did work again. In the release notes I see now only GEN 2,3 and 4 mentioned, is GEN 1 not supported in the newer versions?

wills106, 2022-08-03:
Can you share some error logs?

Also what version of HA are you on? 2022.7.x and newer or pre 2022.7?

wills106, 2022-08-03:
Could you also provide the first 4 digits of your serial, so I know exactly which inverter you have.
As for some reason, some Gen3 X3 report as “X3-Hybrid-G1/G2” in the SolaX Cloud.

blb4github, 2022-08-03:
Thanks! I'm running 2022.5.4, should I update? first 4 characters of the serial number are U30E. I'll search for the logs

wills106, 2022-08-04:
To use a version newer than 0.5.8 you need to be on HA 2022.7

0.5.9 Introduced the change to comply with the refactoring that was needed to comply with HA 2022.7 (With the old formatting being dropped by HA 2022.10)

closed 2022-08-04 by wills106

blb4github, 2022-08-04:
I have updated HA to 2022.08 and the integration version 0.5.22 works now, thanks!

Closed #94 as completed.