wills106/homeassistant-solax-modbus #95: [BUG]

M0LTE, 2022-08-04:
**Describe the bug**
Upgraded to 0.5.12

Now get "Unexpected exception importing component custom_components.solax_modbus" and integration is down

**Mandatory details**
1. Integration version: 0.5.12
2. HA version: 2022.3.8
3. First 4 digits of Serial: 6S58
4. Connection Method: RS485 via Ethernet adapter

**Detailed Error Log**
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 553, in get_component
    cache[self.domain] = importlib.import_module(self.pkg_path)
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 19, in <module>
    from .const import GEN2, GEN3, GEN4, X1, X3, HYBRID, AC, EPS, PV
  File "/config/custom_components/solax_modbus/const.py", line 225, in <module>
    SolaxModbusNumberEntityDescription( name = "Battery Minimum Capacity",
TypeError: __init__() got an unexpected keyword argument 'native_min_value'
```

wills106, 2022-08-04:
[Known Issues](https://github.com/wills106/homeassistant-solax-modbus#known-issues)

#84
Versions of the Integration from 0.5.9 onwards don't work on HA versions prior to 2022.7

closed 2022-08-04 by wills106

M0LTE, 2022-08-04:
Just found that. Thanks, upgrading.