wills106/homeassistant-solax-modbus #106: [BUG] Polling stops over night and sometimes every 4-6h

tremmer99, 2022-08-13:
**Describe the bug**
Every 4-6h hours and always overnight polling stops and I have to restart home-assistant to be working again.

**Mandatory details**
1. Integration version: 0.6.0b4
2. HA version: 2022.8.1
3. First 4 digits of Serial: H34T
4. Connection Method: Modbus via Waveshare TCP/Modbus

**Detailed Error Log**
no error i guess

infradom, 2022-08-13:
> no error i guess
Could you please check the full detailed logs (not only the errors, but also warning and info messages related to Solax).
I almost have exactly the same setup (with Waveshare abnd H34T) and did not encounter the problem. 
What is your polling interval? Did you use the default 15 seconds or a shorter interval? What is the configured baudrate on Solax and Waveshare?

tremmer99, 2022-08-13:
Concerning home-assistant.log there is no message other than this module is has not been tested by Home Assistant. Is there another log location to look for?
It makes no difference if polling interval is 15 or 2 seconds (right now I have 2s). Configured baudrate is 19200.

infradom, 2022-08-13:
Thanks for the information.
There must be more messages in the full log.  We write several informational messages during startup. Dependent on your language settings (, you can find the full logs under (my language is not set to English, so you may see something slightly different)

Settings -> System  -> Logs  > at bottom of page press "load full logs"

Now the full logs are loaded, and if you scroll down, you will see them.
Once the full logs are shown, you can use the search function in your browser to search for "solax" related entries

wills106, 2022-08-13:
Prior to installing 0.6.0b4 were you on an older version such as 0.5.x or is 0.6.0b4 the first version you have installed?

If you were on an older version such as 0.5.x did you delete the Integration from the Integration page before updating to 0.6.0b4 and then re adding the Integration on the Integration page?

tremmer99, 2022-08-13:
I was on an older version but recently did a fresh virtualbox install of ha and installed 0.6.0b4 directly via HACS. Was one of my "let's try this" :-)

wills106, 2022-08-13:
Have you tried a newer version yet as we are on 0.6.0b10 at the moment. Before updating I recommend you remove the Integration from the Integration page before you update, then restart HA and reconfigure the Integration.

wills106, 2022-08-13:
Are the any communication logs on the Waveshare itself? Just wondering if it's the Waveshare that's having the issue?

tremmer99, 2022-08-13:
I haven't found logs on the Waveshare so far - I'll have to check with the vircom tool later. I upgraded to 0.6.0b10 as you described - let's see tomorrow morning ;-)

tremmer99, 2022-08-14:
Oh yes - now there are errors all the time:

Logger: homeassistant
Source: custom_components/solax_modbus/__init__.py:368
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 13. August 2022 um 22:42:58 (1391 occurrences)
Last logged: 08:11:01

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 230, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solax_modbus/__init__.py", line 317, in read_modbus_data
    return self.read_modbus_holding_registers_0() and self.read_modbus_holding_registers_1() and self.read_modbus_holding_registers_2() and self.read_modbus_input_registers_0() and self.read_modbus_input_registers_1() and self.read_modbus_input_registers_2()
  File "/config/custom_components/solax_modbus/__init__.py", line 368, in read_modbus_holding_registers_1
    bootloader_version = decoder.decode_16bit_uint()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 402, in decode_16bit_uint
    return unpack(fstring, handle)[0]
struct.error: unpack requires a buffer of 2 bytes

wills106, 2022-08-14:
Do you have any batteries attached to your Inverter?

Reason being is your errors started at 22:42, so I was wondering if your Inverter is going into sleep mode if you don't have any batteries attached.

tremmer99, 2022-08-14:
Yes, batteries attached

infradom, 2022-08-14:
Maybe batteries have reached their min discharge level by 22:42, in that case, the inverter goes to sleep also. 
@tremmer99 : Is it possible that the battery is already at configured minimum leverl by 22:42 ?
@wills106 :  I dont understand the crash at line 369, we could read the previous registers, why is this one failing

tremmer99, 2022-08-14:
@infradom battery SoC is at 68 % at 22:42 according to solax cloud - and I switched to 15s polling intervall after updating to the latest version yesterday.

wills106, 2022-08-14:
I have had similar before where an Inverter was missing either 1 or 2 registers on the XRE #90 That's why I made series number global so I could [skip it](https://github.com/wills106/homeassistant-solax-modbus/blob/eb2358d44636f157c29e4716a58230ccd00a7757/custom_components/solax_modbus/__init__.py#L593) but if this H34T was missing `bootloader_version` you would expect it to error even during the day.

@tremmer99 if you enable `bootloader_version` does it return a value? it will take about 30s to update after you enable it.

infradom, 2022-08-14:
> @infradom battery SoC is at 68 % at 22:42 according to solax cloud - and I switched to 15s polling intervall after updating to the latest version yesterday.

@tremmer99 Thanks for this valuable information. What is your battery minimum discharge  level (I gues you are in selfuse mode, or are you in feedin_priority  or manual mode) ?

tremmer99, 2022-08-14:
@infradom I am using feedin priority (SoC 20min, charge battery to 80 %) mode as grid power is cheaper than what I get feed_in for at the moment. I also have set a charge battery at 00:00 to 02:00 and battery discharge at 02:15 to 07:00 for some tests.

@wills106 I have enabled bootloader_version - in the logbook it says "SolaX Bootloader Version wechselte zu 8" so should be "8"?

wills106, 2022-08-14:
> should be "8"?

It relates to what version Firmware you have. I just wanted to make sure it was actually returning a value, at least we know it's readable during the day. It doesn't explain `First occurred: 13. August 2022 um 22:42:58` happening though?

infradom, 2022-08-14:
The feedin_priority explains why the inverter goes to sleep mode when there is no more sunlight.
My H34T reports bootloader version 7 (I should try to force it to sleep mode to see what happens then)
@tremmer99 The firmware versions may be relevant, can you enable these entities:
[image]

tremmer99, 2022-08-14:
[image]

wills106, 2022-08-14:
Just released 0.6.0b11 @infradom has added improved exception handling for modbus errors.

@tremmer99 if you update to 0.6.0b11 can you let us know what the new logs report when you next loose communication, hopefully we can workout what it's failing on your Inverter.

tremmer99, 2022-08-14:
@wills106 thank you very much - i've removed/updated/restarted/installed/restartet - let's see!

tremmer99, 2022-08-14:
@wills106 polling is down since 18:06

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:372
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 18:06:32 (269 occurrences)
Last logged: 19:13:32

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 319, in read_modbus_data
    return self.read_modbus_holding_registers_0() and self.read_modbus_holding_registers_1() and self.read_modbus_holding_registers_2() and self.read_modbus_input_registers_0() and self.read_modbus_input_registers_1() and self.read_modbus_input_registers_2()
  File "/config/custom_components/solax_modbus/__init__.py", line 372, in read_modbus_holding_registers_1
    bootloader_version = decoder.decode_16bit_uint()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 402, in decode_16bit_uint
    return unpack(fstring, handle)[0]
struct.error: unpack requires a buffer of 2 bytes

wills106, 2022-08-14:
It really doesn't like that bootloader register!
Is that the only error for SolaX?

Out of curiosity is your H34T one of the following:

H34T08H
H34T10H
H34T12H
H34T15H

Or do you have something different after H34T?

tremmer99, 2022-08-14:
Yes - it's the only error. I've got a H34T15H

wills106, 2022-08-14:
> I've got a H34T15H

Ahh ok, I was kind of hoping you had a different Inverter and we could try and skip that register like we do on the XRE version of the Gen3 and U50EC on the Gen2 as they are both missing certain registers.

I am assuming your Waveshare is setup exactly the same as @infradom screenshots on the Wiki?

wills106, 2022-08-14:
Does reloading the Integration bring the sensors back?

[image]

Or do you have to restart Home Assistant?

infradom, 2022-08-14:
I could try to let my H34T08 inverter go to sleep mode so see if I can reproduce the problem, but I cannot do this today and probably not in the coming days (because I participate in a dynamic price trial and due to the high energy prices)

infradom, 2022-08-14:
@tremmer99 When you reproduced the problem, I  assume that the inverter was already in sleep mode. How much time difference was there between entering sleep mode and the error messages ?

wills106, 2022-08-14:
> I participate in a dynamic price trial and due to the high energy prices

I hope @tremmer99 understands but I wish you to fully participate in your Dynamic Trials over this Integration and troubleshooting.

I strongly believe in dynamic grid consumption over the coming years, while we search for better ways to power our houses.

tremmer99, 2022-08-14:
@wills106 I think an integration reload does not restart polling, but I will try next time!
@infradom no worries :-) But I don't think the inverter went into sleep mode - we had still sun and solax cloud reporting pv production at 6pm (18:0X).

tremmer99, 2022-08-15:
@wills106 i never used the waveshare for something else - as i have another waveshare I can try the other one too. Reloading the integration started polling again. Same error: Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:673
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 01:49:37 (634 occurrences)
Last logged: 07:06:09

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 319, in read_modbus_data
    return self.read_modbus_holding_registers_0() and self.read_modbus_holding_registers_1() and self.read_modbus_holding_registers_2() and self.read_modbus_input_registers_0() and self.read_modbus_input_registers_1() and self.read_modbus_input_registers_2()
  File "/config/custom_components/solax_modbus/__init__.py", line 673, in read_modbus_holding_registers_2
    meter_1_id = decoder.decode_16bit_uint()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 402, in decode_16bit_uint
    return unpack(fstring, handle)[0]
struct.error: unpack requires a buffer of 2 bytes

wills106, 2022-08-15:
> I was on an older version but recently did a fresh virtualbox install of ha...

Do you know what version you used?
Also were you using Virtualbox or were you hosting HA in a different method?
Were you also using the Waveshare then?

tremmer99, 2022-08-17:
No I am not sure which version but I had those problems too. Reinstall was due to other plugins i tried - fresh install is without other plugins.

wills106, 2022-08-17:
If you had these issues prior to 0.6.0bx I don't think it's related to the changes made in the 0.6.0bx branch.

Instead of running HA in a VM, is there any chance you can directly run HA on the host machine? Docker is always a good option for Core.

tremmer99, 2022-08-17:
Yes of course - i can try docker but I have to find out how to active HACS etc when using docker on Windows OS

wills106, 2022-08-17:
I used to run on Synology. I just want to try and rule out what the actual issue is.
You could always manually download the latest 0.6.0b release and manually install instead of HACS

tremmer99, 2022-08-18:
Docker up&running without the entities you mentioned (bootload, firmware_versions) - let's see.

wills106, 2022-08-18:
> (bootload, firmware_versions)

They will still be there, they are just disabled by default. 
Which version did you install? 0.5.x or 0.6.0bx
Just wondering if the Docker version is any more stable compared to the VM

tremmer99, 2022-08-18:
I've installed 0.6.0b12 via HACS

wills106, 2022-08-19:
Has using the docker version made and difference?
Did polling carry on? Or is it still stopping every 4-6h?

tremmer99, 2022-08-19:
No - same thing ;-) Stopped polling 12h ago :/

wills106, 2022-08-19:
Is it failing on the same register again in the logs?

tremmer99, 2022-08-19:
yes!

tremmer99, 2022-08-19:
I can try to use the other Waveshare I've got but will take 1 week to try - but I don't think this will change a thing - do you?

wills106, 2022-08-19:
Just noticed you are having issues with different registers:

bootloader_version = decoder.decode_16bit_uint()
meter_1_id = decoder.decode_16bit_uint()

So it's not actually the same one each time.

wills106, 2022-08-19:
In last nights logs do you have any other decoder.decode_16bit_uint() errors?

tremmer99, 2022-08-19:
Oh - the one from last night:

fired 2k+ times

Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 319, in read_modbus_data
    return self.read_modbus_holding_registers_0() and self.read_modbus_holding_registers_1() and self.read_modbus_holding_registers_2() and self.read_modbus_input_registers_0() and self.read_modbus_input_registers_1() and self.read_modbus_input_registers_2()
  File "/config/custom_components/solax_modbus/__init__.py", line 372, in read_modbus_holding_registers_1
    bootloader_version = decoder.decode_16bit_uint()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 402, in decode_16bit_uint
    return unpack(fstring, handle)[0]
struct.error: unpack requires a buffer of 2 bytes

wills106, 2022-08-20:
Did Skip-missing-bytes survive the night? Or did it just start erroring else where?

tremmer99, 2022-08-20:
I am currently on vacation - I‘ll install it this evening - thank you!

wills106, 2022-08-22:
Have you managed to try it out? Did the issue go away? Or did it just error on another register instead?

tremmer99, 2022-08-22:
Just updated today - keep you updated - thank you!

tremmer99, 2022-08-24:
Stalled again - 4 errors:

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:363 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 23. August 2022, 18:43:43 (2 occurrences) 
Last logged: 05:41:16

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 319, in read_modbus_data
    return self.read_modbus_holding_registers_0() and self.read_modbus_holding_registers_1() and self.read_modbus_holding_registers_2() and self.read_modbus_input_registers_0() and self.read_modbus_input_registers_1() and self.read_modbus_input_registers_2()
  File "/config/custom_components/solax_modbus/__init__.py", line 363, in read_modbus_holding_registers_1
    firmwareversion_modbustcp_major = decoder.decode_16bit_uint()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 402, in decode_16bit_uint
    return unpack(fstring, handle)[0]
struct.error: unpack requires a buffer of 2 bytes

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:374 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 23. August 2022, 16:30:27 (1432 occurrences) 
Last logged: 08:26:01

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 319, in read_modbus_data
    return self.read_modbus_holding_registers_0() and self.read_modbus_holding_registers_1() and self.read_modbus_holding_registers_2() and self.read_modbus_input_registers_0() and self.read_modbus_input_registers_1() and self.read_modbus_input_registers_2()
  File "/config/custom_components/solax_modbus/__init__.py", line 374, in read_modbus_holding_registers_1
    rtc_seconds = str(decoder.decode_16bit_uint())
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 402, in decode_16bit_uint
    return unpack(fstring, handle)[0]
struct.error: unpack requires a buffer of 2 bytes

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:383 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 23. August 2022, 18:44:13 (1195 occurrences) 
Last logged: 08:26:16

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 319, in read_modbus_data
    return self.read_modbus_holding_registers_0() and self.read_modbus_holding_registers_1() and self.read_modbus_holding_registers_2() and self.read_modbus_input_registers_0() and self.read_modbus_input_registers_1() and self.read_modbus_input_registers_2()
  File "/config/custom_components/solax_modbus/__init__.py", line 383, in read_modbus_holding_registers_1
    self.data["rtc"] = datetime.strptime(date_string, '%d/%m/%y %H:%M:%S')
  File "/usr/local/lib/python3.10/_strptime.py", line 568, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
  File "/usr/local/lib/python3.10/_strptime.py", line 349, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data '00/00/00 00:00:02' does not match format '%d/%m/%y %H:%M:%S'

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:718 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 23. August 2022, 18:43:58 (1196 occurrences) 
Last logged: 08:26:31

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 319, in read_modbus_data
    return self.read_modbus_holding_registers_0() and self.read_modbus_holding_registers_1() and self.read_modbus_holding_registers_2() and self.read_modbus_input_registers_0() and self.read_modbus_input_registers_1() and self.read_modbus_input_registers_2()
  File "/config/custom_components/solax_modbus/__init__.py", line 718, in read_modbus_holding_registers_2
    disch_cut_off_point_different_s = decoder.decode_16bit_uint()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 402, in decode_16bit_uint
    return unpack(fstring, handle)[0]
struct.error: unpack requires a buffer of 2 bytes

wills106, 2022-08-24:
I think you are having some communications error.
It's just failing on the previous / next set of registers after the ones I have blocked out.

tremmer99, 2022-08-28:
Is it possible to integrate a wait for 2-5 seconds and retry or something like this if it fails to pull a former working register?

infradom, 2022-08-28:
I am more and more convinced that you have a wiring/communication problem. I have a H34T running for almost a year without any communication problems of this kind. The length of the serial cable between solax and waveshare is in my case around 12 meter.
The waveshare settings may also play a role (see the Wiki page for my waveshare settings).

wills106, 2022-10-26:
Closing due to no further reports of this issue / feedback.

closed 2022-10-26 by wills106