wills106/homeassistant-solax-modbus #113: Entity button.solax_battery_awaken cannot be added a second time to an entity platform

matthewjporter, 2022-08-25:
Not sure when this crept in but as I was going over my HA log, I noticed I get the following error in my log, it comes up twice for each instance of my integration and seems to be on HA startup/reboot etc

I have added just a single instance (I have two Gen2 inverters) and only get the error twice for that integration

2022-08-25 12:00:40.591 ERROR (MainThread) [homeassistant.components.button] Error adding entities for domain button with platform solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 428, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 468, in _async_add_entity
    entity.add_to_platform_start(
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 745, in add_to_platform_start
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity button.solax1_battery_awaken cannot be added a second time to an entity platform
2022-08-25 12:00:40.637 ERROR (MainThread) [homeassistant.components.button] Error while setting up solax_modbus platform for button
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 289, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 428, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 468, in _async_add_entity
    entity.add_to_platform_start(
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 745, in add_to_platform_start
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity button.solax1_battery_awaken cannot be added a second time to an entity platform

It is not affecting the running of my HA but is an error being logged

Regards

wills106, 2022-08-25:
I have tried removing the Integration from my Instance and adding it again with a number ie solax5
This does not replicate your error.

Only way I can replicate the error is by adding a second instance of SolaX Modbus, it fails by not finding a serial but I get the same comment about `battery_awaken cannot be added a second time to an entity platform`

Only difference I can find between `battery_awaken` and the other button's is the missing `entity_category = EntityCategory.CONFIG,` but that shouldn't make a difference.
Locally I have added then line into the `battery_awaken` description and I can still replicate the error on adding a second instance.

infradom, 2022-08-25:
This might be related to the identity of each instance (recent config_flow.py changes) or use of global variables. I am busy now and may not have time in de very near future, but I certainly help investigate this later in the week or next week.

matthewjporter, 2022-08-25:
Sorry Guys, Seems I hit the submit button too early - Thanks @wills106 for adding in a bit of info ;)

Extra Info:
Home Assistant 2022.8.6
Supervisor 2022.08.3
Operating System 8.5
Frontend 20220802.0 - latest

A bit of investigating further shows when I add in one instance "SolaX1" I get the error in the log immediately and then on reboot

When I add my second instance "SolaX2" I get the error again in the log for that instance only and not both

On reboot, I get for both instances

@infradom As I said, not affecting the running of it as the button is actually created and I don't need to use the awaken button so no real rush

My assumption is that something is looping around and trying to re-create it 2 or 3 times. 1st time it creates it, the second time it fails as its there so logs the error, then tries again so logs again, and then moves on

matthewjporter, 2022-08-25:
0.6.1 fixed the issue

Updated, restarted and no error, rebooted and no error

closed 2022-08-25 by matthewjporter