wills106/homeassistant-solax-modbus #140: Options is available as a standard property of SelectEntityDescription

epenet, 2022-10-10:
As of Home Assistant Core 2022.11, options is available as a standard property of SelectEntityDescription.
This may cause issues in custom components if a custom options property was previously implemented.

Please adjust the custom component by either dropping or renaming the custom options property.

See https://github.com/home-assistant/core/pull/78882

epenet, 2022-10-10:
Please be aware that this will only be available from 2022.11 release.
If you wish to be compatible with both 2022.10 and 2022.11 you may need to rename the options property.

wills106, 2022-11-10:
Closing down as updated to Home Assistant 2022.11.2 and still got fully functioning selects.

closed 2022-11-10 by wills106