wills106/homeassistant-solax-modbus #154: Unable to import component: No module named 'pymodbus.client.sync' after recent updates (2022.10.3 and/or OS 9.2)

Claudio1L, 2022-10-14:
HI

I started receiving the attached log. They appear after reboot and usually after one or two reboots it works fine. Today I'm not able to make it works. It seems a module missing

2022-10-14 08:22:57.748 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration solax_modbus: Unable to import component: No module named 'pymodbus.client.sync'

Claudio1L, 2022-10-15:
Don't understand why but it was a conflict with the Bluetooth usb key I recently added. My apologise for the confusion

infradom, 2022-10-15:
I did not have time to look into it into detail, but the problem did not occur on my system after the upgrade.
We might still need to update our code to support the new pymodbus version

Claudio1L, 2022-10-20:
after update ABB integration, to 2.1.0, supporting pymodbus 3.0 issue appeared again on bot integration.  I had to rollback to the previous snapshot. Before rolling back, I tried to remove everything from integration page and redownloading from hacs, ABB integration (2.1.0) was working fine, Solax integration refused to install for missing pymodbus.client.sync.

infradom, 2022-10-20:
We recently adapted our source code ot support both the old pymodbus and the new pymodbus 3.0.x
This should solve your problem with operating abb and solax_modbus together
I am not sure if this code is already in the 0.7.0 prerelease 10 beta version.
Maybe @wills106 can confirm or issue a new prerelease

wills106, 2022-10-20:
Just released 0.7.0b11 hopefully it will fix the issue.

Claudio1L, 2022-10-21:
Ok, removing everything and reinstalling seems fine. A first try failed because I forgot to set the inverter address to 254, so just 4 generic entities were discovered. Reinstalling with 254 address works. Thank you for the support

wills106, 2022-10-26:
Closing due to no further reports of this issue.

closed 2022-10-26 by wills106