wills106/homeassistant-solax-modbus #161: [BUG]

LuisPalacios, 2022-10-17:
Hi, after installing the 0.7.0b10 the integration doesn't start. I had 0.7.0b9 working well.

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solax_modbus/__init__.py", line 105, in async_setup_entry
    _LOGGER.info(f"Ready to load plugin {config[CONF_PLUGIN]}")
KeyError: 'plugin'
```

Tried to reload, no success

```
2022-10-17 14:53:04.887 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-10-17 14:53:11.940 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry SolaXM for solax_modbus
File "/config/custom_components/solax_modbus/__init__.py", line 105, in async_setup_entry
2022-10-17 15:05:23.833 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry SolaXM for solax_modbus
File "/config/custom_components/solax_modbus/__init__.py", line 105, in async_setup_entry
```

Home Assistant 2022.10.4
Supervisor 2022.10.0
Operating System 9.2
Frontend 20221010.0 - latest

wills106, 2022-10-17:
Have you tried removing the Integration from the Integration page. Restarting Home Assistant and adding the Integration to the Integration page?

LuisPalacios, 2022-10-17:
Just did it with success and was coming here back to report it. 

Thank you and confirmed.

- I deleted the integration. 
- Rebooted HA
- Re-downloaded Beta 10.
- Name it exactly the same so nothing changes in my configuration.

Up and running !!

Thanks
Luis

wills106, 2022-10-17:
I thought that would solve the issue.

When an Integration undergoes a major code change you sometimes need to remove Integration from the Integration page. Restart Home Assistant and add the Integration to the Integration page again. So Home Assistant can fully reload the integration as it seems to cache parts of the code.

I will leave this issue open for now while we are Beta Testing.
I'll add a comment to the release notes to perform the above when moving to 0.7.0

wills106, 2022-10-26:
Closing due to no further reports of this issue.

closed 2022-10-26 by wills106