wills106/homeassistant-solax-modbus #163: Question - Load Not Matching Calculation

wranglatang, 2022-10-19:
**Describe the bug**
I have been looking at the MANY sensors created for my X1-Hybrid Gen4 inverter and trying to create a dashboard to show them. I have been using the below sensors:
[image]

But things don't seem to add up... If i do the calculations manually i get the below:
PV Total Power - Battery Power Charge = House Load
555 - 404 = 151
But the "House Load" Sensor is 245 (Which i think it more accurate - based on my typical use house use)

Can you help me understand what is happening here and why these numbers don't add up - have i misunderstood the sensors?

wranglatang, 2022-10-19:
@wills106 - Do you have a discord or something we can chat on? I have a couple of questions about the interface and also what different entities are for / how they have been named etc that I'd like to share with you!

wills106, 2022-10-20:
The house load etc is calculated in this section:

https://github.com/wills106/homeassistant-solax-modbus/blob/c9ea07bfdece6764320ae84c52f315391ce3b21f/custom_components/solax_modbus/plugin_solax.py#L177-L194

You won't be to directly convert PV or Battery directly into house load due to the losses from the DC - AC conversion etc.

There a few answers or what the entities are here:
https://community.home-assistant.io/t/solax-inverter-by-modbus-no-pocket-wifi-now-a-custom-component/140143/689

https://community.home-assistant.io/t/solax-inverter-by-modbus-no-pocket-wifi-now-a-custom-component/140143/727

For real-time I use the "power-distribution-card" Card:
[image]

wranglatang, 2022-10-21:
Morning Willis! Your comment below
> You won't be to directly convert PV or Battery directly into house load due to the losses from the DC - AC conversion etc.

this makes complete sense, however there is a difference of ~100W's . Im taking a look at the code snipped below - i cant see feedin_power as a sensor in home assistant?
``` python
def value_function_house_load(initval, descr, datadict): 
     return datadict['inverter_load'] - datadict['feedin_power'] 
```

wills106, 2022-10-21:
It's called Measured Power in the integration.

Calling it feedin_power is a bit misleading as it's a bidirectional sensor.

wills106, 2022-10-26:
Closing due to no further communications within the past 5 days and not actually an issue with the Integration.

closed 2022-10-26 by wills106