wills106/homeassistant-solax-modbus #169: HA log flooded

martas99, 2022-10-24:
Hi,
I've found out that home-assistant.log is being flooded by Solax Modbus integration records. I have HAOS running on a x86 machine, connected to the inverter by USB-RS485 dongle, inverter type X3 hybrid G4. Any suggestions?

`022-10-24 14:13:03.505 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 296: return self._client.read_input_registers(address, count, **kwargs)
2022-10-24 14:13:03.586 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 296: return self._client.read_input_registers(address, count, **kwargs)
2022-10-24 14:13:08.226 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 290: return self._client.read_holding_registers(address, count, **kwargs)
2022-10-24 14:13:08.315 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 290: return self._client.read_holding_registers(address, count, **kwargs)
2022-10-24 14:13:08.387 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 296: return self._client.read_input_registers(address, count, **kwargs)
2022-10-24 14:13:08.499 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 296: return self._client.read_input_registers(address, count, **kwargs)
2022-10-24 14:13:08.580 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 296: return self._client.read_input_registers(address, count, **kwargs)
2022-10-24 14:13:13.225 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 290: return self._client.read_holding_registers(address, count, **kwargs)
2022-10-24 14:13:13.314 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 290: return self._client.read_holding_registers(address, count, **kwargs)
2022-10-24 14:13:13.386 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 296: return self._client.read_input_registers(address, count, **kwargs)
2022-10-24 14:13:13.498 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 296: return self._client.read_input_registers(address, count, **kwargs)
2022-10-24 14:13:13.579 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 296: return self._client.read_input_registers(address, count, **kwargs)
2022-10-24 14:13:18.227 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 290: return self._client.read_holding_registers(address, count, **kwargs)
2022-10-24 14:13:18.315 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 290: return self._client.read_holding_registers(address, count, **kwargs)
2022-10-24 14:13:18.388 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 296: return self._client.read_input_registers(address, count, **kwargs)
2022-10-24 14:13:18.500 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 296: return self._client.read_input_registers(address, count, **kwargs)
2022-10-24 14:13:18.581 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 296: return self._client.read_input_registers(address, count, **kwargs)
2022-10-24 14:13:23.228 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 290: return self._client.read_holding_registers(address, count, **kwargs)
2022-10-24 14:13:23.317 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 290: return self._client.read_holding_registers(address, count, **kwargs)
2022-10-24 14:13:23.390 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 296: return self._client.read_input_registers(address, count, **kwargs)`

`

wills106, 2022-10-24:
It's a [known issue](https://github.com/wills106/homeassistant-solax-modbus#known-issues) when you use a RS485 - USB adaptor, that's one of the reasons it's recommended to use a device like the Waveshare.

It potentially needs a rewrite of the Serial section.

For the time being you could change your logger setup for this custom_component

martas99, 2022-10-24:
@wills106 thx

closed 2022-10-24 by martas99