wills106/homeassistant-solax-modbus #181: [BUG] Flooding "SolaX Selfuse Backup SOC" events

alibasta, 2022-11-03:
**Describe the bug**
I'm using this excellent integration since a week and wondering why I see every few seconds events called "SolaX Selfuse Backup SOC" with random int values in my logbook.I'm afraid that these are values that are written every few seconds into the inverter EEPROM. Is this a known bug? 

**Mandatory details**
1. Integration version: 0.6.4
2. HA version: 2022.11.1
3. First 4 digits of Serial: H34T
4. Connection Method: Waveshare RS485 to RJ45 Ethernet

infradom, 2022-11-04:
We really need to see some logfile lines to understand what type of message you receive. Preferrably a log extract with all the solax related messages

alibasta, 2022-11-04:
Would love to do so but honestly I can't find any location the integration is logging to. Could you tell me where the homeassistant-solax-modbus logs are located. The only messages I see are the continuous updates in the logbook:
[image]

wills106, 2022-11-04:
If there are no log's showing for SolaX try the following, you can find the full logs under:

Settings → System → Logs > at bottom of page press “LOAD FULL LOGS”

In top left you can search for SolaX, show it only shows the related errors.
If you still don't get any SolaX errors try adding the following into your configuration.yaml

```
logger:
  default: info
```

alibasta, 2022-11-04:
Thanks. I added it to the configuration.yaml 

`2022-11-04 15:02:08.881 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-11-04 15:02:19.729 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 2: {'met', 'script', 'homematicip_local', 'search', 'timer', 'my', 'mobile_app', 'nuki', 'persistent_notification', 'tag', 'ibeacon', 'cast', 'solax_modbus', 'repairs', 'default_config', 'device_automation', 'scene', 'map', 'logbook', 'input_number', 'growatt_server', 'forecast_solar', 'input_select', 'input_datetime', 'diagnostics', 'blueprint', 'application_credentials', 'input_text', 'tts', 'counter', 'sun', 'radio_browser', 'schedule', 'automation', 'system_health', 'lovelace', 'ffmpeg', 'person', 'raspberry_pi', 'tapo', 'shelly', 'hardware', 'homekit', 'hacs', 'input_boolean', 'onboarding', 'analytics', 'config', 'ipp', 'file_upload', 'media_source', 'history', 'zone', 'rpi_power', 'homeassistant_alerts', 'image', 'energy', 'trace', 'auth', 'input_button', 'threshold'}
2022-11-04 15:02:20.500 INFO (MainThread) [homeassistant.setup] Setting up solax_modbus
2022-11-04 15:02:20.545 INFO (MainThread) [homeassistant.setup] Setup of domain solax_modbus took 0.0 seconds
2022-11-04 15:02:20.629 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting infradom configflow solax_modbus name
2022-11-04 15:02:20.630 INFO (MainThread) [custom_components.solax_modbus] solax setup entries - data: {}, options: {'name': 'SolaX', 'interface': 'tcp', 'read_modbus_addr': 1, 'scan_interval': 5, 'read_eps': False, 'read_dcb': False, 'host': '192.168.2.138', 'port': 502}
2022-11-04 15:02:20.630 INFO (MainThread) [custom_components.solax_modbus] solax serial port /dev/ttyUSB0 interface tcp
2022-11-04 15:02:20.630 INFO (MainThread) [custom_components.solax_modbus] solax modbushub creation with interface tcp baudrate (only for serial): 19200
2022-11-04 15:02:20.631 INFO (MainThread) [custom_components.solax_modbus] solax modbushub done {'_hass': <homeassistant.core.HomeAssistant object at 0x7fac3f6560>, '_client': <ModbusTcpClient at 0x7f9a937d90 socket=None, ipaddr=192.168.2.138, port=502, timeout=5>, '_lock': <unlocked _thread.lock object at 0x7f9a92b700>, '_name': 'SolaX', '_modbus_addr': 1, '_invertertype': 0, '_seriesnumber': 'still unknown', 'interface': 'tcp', 'read_serial_port': '/dev/ttyUSB0', '_baudrate': 19200, '_scan_interval': datetime.timedelta(seconds=5), '_unsub_interval_method': None, '_sensors': [], 'data': {}}
2022-11-04 15:02:20.653 INFO (MainThread) [custom_components.solax_modbus] Read Solax serial number: H34XXX, swapped: False
2022-11-04 15:02:20.708 INFO (MainThread) [homeassistant.components.button] Setting up button.solax_modbus
2022-11-04 15:02:20.710 INFO (MainThread) [homeassistant.components.number] Setting up number.solax_modbus
2022-11-04 15:02:20.713 INFO (MainThread) [homeassistant.components.select] Setting up select.solax_modbus
2022-11-04 15:02:20.719 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus`

I can't see any abnormalities but while copying the logs above there are every second the following methods flooding the logbook:
[image]

Is this something that is writing into the EEPROM and probably damaging it?

wills106, 2022-11-04:
Unless you have an automation there isn't anything in the Integration that writes to a register by it's self.
The range should only be 10 - 100. So I am not sure how it could be 126 or 225 etc.

I'm not sure if this is a Firmware error like #83

alibasta, 2022-11-04:
Do you know how the menu item for "Backup SOC" is named / in which section it is on the cloud portal?

wills106, 2022-11-04:
I don't own a Gen4 so I am not sure how it is laid out as there are quite a few new sensors and control options on the Gen4 which were not present in the Gen3.

infradom, 2022-11-04:
I have a Gen4, but it is currently disconnected from the cloud, I will reconnect it later to have a look how it is called there.
You are showing state logs in the screenshot,  the debug/info/error/warning logs do not seem to contain entries?
There is indeed something wrong with one of the read registers, but if we were writing to eeprom, in recent versions, we output a line in the info/error/warning/debug logs.  So I am not yet worried.
Version 7 Beta of the integration is getting relatively stable and if @wills106 agrees, it is probably worth trying this version as the code is quite different, so we might be chasing an error that already disappeared

wills106, 2022-11-04:
Yes, I agree 0.7.0b branch is stable now and it is worth trying out.

alibasta, 2022-11-04:
Just installed 0.7.0b20, configured and restarted everything and I see the same result: 

[image]

wills106, 2022-11-04:
When you went to 0.7.0b20 did you remove the Integration from the Integration Page. Upgraded to 0.7.0b20. Restarted HA and then added the Integration to the Integration Page?

If you don't remove the old and restart, HA caches parts of the old code still.

alibasta, 2022-11-04:
No I haven't done it exactly like this, but now I did :-) 

Still the same loggings though. Could I somehow provide some raw modbus export for further debugging?

wills106, 2022-11-04:
Like @infradom mentions if the Integration was writing to the Inverter you would get messages in the log.
What you are seeing is the Inverter informing the Integration that the value has changed.

Does Backup SOC match any other entities?

alibasta, 2022-11-04:
Ok I'm now calmed regarding the writing to the inverter. 

@wills106  What do you mean with "Does Backup SOC match any other entities?"?

wills106, 2022-11-04:
You can see it going from 235 to 113 then 144
Can you match them numbers up to another entity at the same time?
For example does Measured Power and Backup SOC have the same values? Or a different entity.
I am wondering if there is a bug in the firmware, where it's reading a different sensor instead of Backup SOC internally to the Inverter.

infradom, 2022-11-04:
In the solax app , it is under

settings->User settings->backup Mode

On the solaxcloud website: you have to enter devicemanagement -> remote settings first

infradom, 2022-11-04:
correction, I was showing a different parameter; this parameter does not seem to exist in the cloud app.
Let us verify the docs to see if this is a genuine GEN4 setting ...
Sorry ..

infradom, 2022-11-04:
I have no clue what this parameter does ... It is in a relatively recent protocol doc, but not in the older documents.
I do not have the most recent specs, maybe @wills106 can look into the most recent gen4 doc for holding register 0x11F
and write register 0xC5

wills106, 2022-11-04:
It's just called 0xC5 - SelfUse_BackupSoc and 0x11F - bSelfUse_BackupSoc
Turned up in ARM Version 1.16 (Single Phase) and somewhere between 1.10 - 1.19 on Three Phase.

infradom, 2022-11-04:
maybe we should disable this by default

wills106, 2022-11-04:
@alibasta Do you know what version Arm is on your Inverter? Just wondering if you are on an early Firmware still?

I have set it to disabled by default.

alibasta, 2022-11-04:
@wills106 thanks! Where is this version number visible?

wills106, 2022-11-04:
Have a look at Page 98 - 99 https://www.solaxpower.com/wp-content/uploads/2022/03/X3-Hybrid-G4.1-User-Manual-SOLAX.pdf

infradom, 2022-11-04:
FYI my ARM version is 2.03
my DSP version is 2.07

alibasta, 2022-11-05:
Sorry for responding late. I was away from the inverter. 
My versions:
ARM: 2.03
DSP: 2.07

wills106, 2022-11-05:
Perhaps SolaX haven't fully implemented Backup SOC yet?
I have disabled it by default anyway in 0.7.0b21

alibasta, 2022-11-07:
I'll retest the latest beta today. If the issue still exist I'll reopen it. Thanks @wills106

wills106, 2022-11-07:
I'll leave it open till you have tested.

alibasta, 2022-11-08:
Retested and it doesn't appear anymore in the log book! Thanks!

closed 2022-11-08 by alibasta