wills106/homeassistant-solax-modbus #186: [BUG]

richardcavspain, 2022-11-05:
**Describe the bug**
After adding the integration using HACS no inverter information is displayed in HA (0W is all that is displayed)

**Mandatory details**
1. Integration version: 0.6.4
2. HA version: 2022.11.1
3. First 4 digits of Serial: XB33 (this is the inverter serial, I assume thats what you mean?)
4. Connection Method: TCP/IP

**Detailed Error Log**
There aren't really any "errors", just no data?  I do see"unknown status" when looking at "Solax Inverter status" but most of the items simply show 0W.

If there are no log's showing for SolaX try the following depending on your language settings, you can find the full logs under:

Ok, looking through this it might be related to a missing serial number?  I see a few entries that relate to this...
Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:92
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 1:05:13 PM (1 occurrences)
Last logged: 1:05:13 PM

reading serial number from address 0 failed; other address may succeed

and

This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:95
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 1:05:18 PM (1 occurrences)
Last logged: 1:05:18 PM

cannot find serial number, even not for MIC

**Additional context**
I'm sorry if I have done anything stupid, this is my first use of HA, HACS etc and I'm struggling a bit :-(

wills106, 2022-11-05:
I believe the X1-Boost Inverters don't have built in Ethernet.
I see you have put "Connection Method: TCP/IP" are you trying to connect with a Waveshare or the PocketLan / Pocket WiFi?
As the PocketLan / Pocket WiFi don't provide a Modbus connection.

richardcavspain, 2022-11-05:
Hi, I am trying to connect to the Pocket Wifi, a user on the Facebook group "Solax Installation" stated that he had no problems using this integration in HA in that configuration and recommended it to me, he has the same setup with Pocket Wifi V3.

wills106, 2022-11-05:
They might be using the Built in SolaX Integration https://www.home-assistant.io/integrations/solax/
If they are using the Pocket WiFi and this custom_component they are using the Pocket WiFi in Access Point mode, which is a massive security risk.

To use this custom_componet you will need to either use an RS485 - USB adaptor, or the prefered method is a RS485 - Ethernet adaptor, such as the Waveshare mentioned in the Wiki

richardcavspain, 2022-11-05:
Ok thanks, but I am clear that its this integration as he posted a link to it in the post - so maybe thats how he is doing it - access point mode.

wills106, 2022-11-05:
You might want to have a read from this conversation on the 20th September https://community.home-assistant.io/t/universal-solar-inverter-over-modbus-rs485-tcp-custom-component-formally-solax-no-pocket-wifi/140143/716?u=wills106

If you want to use it like that just be aware of the security risk and I won't be able to offer any support in that mode.

richardcavspain, 2022-11-05:
ok, thanks for your help :-)

wills106, 2022-11-07:
I'm going to close this issue down. If you want help in the future with a device such as the Waveshare then feel free to raise another issue if there isn't enough detail in the Wiki for getting connected via RS485.

closed 2022-11-07 by wills106