wills106/homeassistant-solax-modbus #190: [BUG]

t0nyb, 2022-11-07:
**Describe the bug**
Defined by error report

**Mandatory details**
1. Integration version:beta 7.24
2. HA version: 22.11
3. First 4 digits of Serial: 6031
4. Connection Method: serial

**Detailed Error Log**

This error originated from a custom integration.

Logger: homeassistant.config_entries
Source: custom_components/solax_modbus/plugin_solis.py:273 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 11:25:37 (1 occurrences) 
Last logged: 11:25:37

Error setting up entry solax for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solax_modbus/__init__.py", line 92, in async_setup_entry
    plugin = importlib.import_module(f".plugin_{plugin_name}", 'custom_components.solax_modbus')
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 273, in <module>
    scale = value_function_rtc,
NameError: name 'value_function_rtc' is not defined

Detailed Error Log, if you press on the individual error further information is displayed.
This is the content we need, the description alone doesn't provide enough context.

If there are no log's showing for SolaX try the following depending on your language settings, you can find the full logs under:

Settings → System → Logs > at bottom of page press “LOAD FULL LOGS”

Now the full logs are loaded. If you scroll down, you will see them.
Once the full logs are shown, you can either use the search function in your browser to search for “solax” related entries or use the search entry field on top of the page.

Search for

solax

and report us the logs. Make sure to replace sensitive information by xxxx (if any)

**Additional context**
Add any other context about the problem here.
Sorry if this is the wrong place to report - I'm a user not a developer

wills106, 2022-11-07:
0.7.0b25 Should fix this, noticed it while adding in the Solis RHI-3P10K-HVES-5G

wills106, 2022-11-07:
Could you also try 0.7.0b27 to see if it detects your Inverter serial number I am not sure which version of the Modbus docs these Inverters are using.

When upgrading make sure you remove the Integration from the Integration page. Update & Restart HA and then add the Integration again on the Integration page.
HA holds the old version of code till you remove and re-add.

t0nyb, 2022-11-07:
This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:112 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 15:43:40 (1 occurrences) 
Last logged: 15:43:40

unrecognized asolis inverter type - serial number : 60312202232200

t0nyb, 2022-11-07:
asolis because it doesn't like duplicate names and hopefully the entities will be top of the tree

wills106, 2022-11-07:
Which beta is this? 0.7.0b27?

Is this with plugin_solis_old.py?

Also what model Inverter are you using and is it single or 3 Phase?

It's a different serial to the RHI-6K-48ES-5G which starts 603105. This being a 6kW mode 3105
603122 I am assuming is 6kW and whichever model 3122 refers to.

t0nyb, 2022-11-07:
Hi

Thanks for your help
The log was from solis new

Single phase 3.6Kwh RHI-3.6K-48ES-5G

This lot from Solis old:

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solis_old
Source: custom_components/solax_modbus/plugin_solis_old.py:107 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 16:36:48 (1 occurrences) 
Last logged: 16:36:48

unrecognized solis_old inverter type - serial number : unknown

wills106, 2022-11-07:
Ok, I'll update with your serial. I was expecting it to be a 6kW Inverter not, 3.6kW. But I don't understand how the serial numbers are structured yet.

wills106, 2022-11-07:
I have just release 0.7.0b28 with your Serial number in solis_old

I will eventually swap the naming between solis and solis_old. As your Inverter is the newer style. (I will let you know when the names are swapped)

wills106, 2022-11-07:
If you serial was read correct in plugin_solis.py I have just released 0.7.0b29 with your Serial in plugin_solis.py

t0nyb, 2022-11-07:
From version 29

Using soils:

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:112 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 17:22:24 (1 occurrences) 
Last logged: 17:22:24

unrecognized aaasolis inverter type - serial number : 60312202232200

t0nyb, 2022-11-07:
Version 29

This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/plugin_solis.py:166 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 18:53:30 (1 occurrences) 
Last logged: 18:53:30

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 292, in read_modbus_data
    res = self.read_modbus_registers_all()
  File "/config/custom_components/solax_modbus/__init__.py", line 362, in read_modbus_registers_all
    res = res and self.read_modbus_block(block, 'input')
  File "/config/custom_components/solax_modbus/__init__.py", line 351, in read_modbus_block
    self.treat_address(decoder, descr)
  File "/config/custom_components/solax_modbus/__init__.py", line 322, in treat_address
    self.data[descr.key] = descr.scale(val, descr, self.data)
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 166, in value_function_rtc
    return datetime.strptime(val, '%d/%m/%y %H:%M:%S')
  File "/usr/local/lib/python3.10/_strptime.py", line 568, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
  File "/usr/local/lib/python3.10/_strptime.py", line 349, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data '53/31/00 18:07:11' does not match format '%d/%m/%y %H:%M:%S'

wills106, 2022-11-07:
Do any of the sensors show readings? Or does RTC block it all? I didn't really look into the RTC bit much.

wills106, 2022-11-07:
0.7.0b30 removes RTC. I can revisit that another day. Do you get any working sensors?

t0nyb, 2022-11-07:
no none. 

will load 30 and report back

t0nyb, 2022-11-07:
Excellent!!

Most sensors working, grid import and export not.

Logs:

This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:277 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 19:28:59 (4 occurrences) 
Last logged: 19:34:05

'H' format requires 0 <= number <= 65535: also cannot read input registers at device 1 position 0x50990
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 331, in read_modbus_block
    if typ == 'input': realtime_data = self.read_input_registers(unit=self._modbus_addr, address=block.start, count=block.end - block.start)
  File "/config/custom_components/solax_modbus/__init__.py", line 277, in read_input_registers
    return self._client.read_input_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/common.py", line 125, in read_input_registers
    return self.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 109, in execute
    return self.transaction.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 174, in execute
    response, last_exception = self._transact(
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 271, in _transact
    packet = self.client.framer.buildPacket(packet)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/rtu_framer.py", line 248, in buildPacket
    data = message.encode()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/register_read_message.py", line 33, in encode
    return struct.pack('>HH', self.address, self.count)
struct.error: 'H' format requires 0 <= number <= 65535

Logger: homeassistant.util.async_
Source: util/async_.py:180 
First occurred: 19:28:59 (4 occurrences) 
Last logged: 19:34:05

Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 277: return self._client.read_input_registers(address, count, **kwargs)

wills106, 2022-11-07:
Could you give a screenshot of all the sensors?

infradom, 2022-11-07:
0x50990 hex is 330128 decimal
I believe meter_voltage declaration is wrong

wills106, 2022-11-07:
I'll go back over these registers in the next day or so. These Solis docs are an utter mess.

There are docs with newer dates in them that are for older Inverters I believe.

t0nyb, 2022-11-07:
[image]

t0nyb, 2022-11-07:
[image]

t0nyb, 2022-11-07:
[image]

t0nyb, 2022-11-07:
[image]

wills106, 2022-11-07:
That's my SolaX Gen3, do might be different sensors and different order.

t0nyb, 2022-11-07:
**Do you have any sensors reporting values though?**
Yes most of them

wills106, 2022-11-07:
That's what I want to see. Then I know which are reporting and which need further work. I can also see if any need the scaling or rounding altering.

t0nyb, 2022-11-07:
OK I misunderstood

Can you think of an efficient way to do that short of screen shotting every one?

In my myenergi integration for instance I can add all the entities to a dashboard. No option to do that here

t0nyb, 2022-11-07:
My bad as usual Wait 5

wills106, 2022-11-07:
Don't you see this?

[image]

Or

[image]

wills106, 2022-11-07:
I don't need all of them. But when you are in the Integrations page, if you find any with funny values screenshot around them.

t0nyb, 2022-11-07:
[image]

wills106, 2022-11-07:
I think GitHub is playing up, looks like that Image started to load and gave up?

t0nyb, 2022-11-07:
Yes I saw that eventually sorry.

will go through them all tomorrow and highlight what I see as anomalous

wills106, 2022-11-07:
Ok, no rush.

I am very pleased with the progress we have made so far!
I doesn't help when there are 3 or even 4 versions of the documents floating about on the Internet and none of the time stamps make sense.

Thanks for the patience with testing this.

infradom, 2022-11-07:
> I'll go back over these registers in the next day or so. These Solis docs are an utter mess.
> 
> There are docs with newer dates in them that are for older Inverters I believe.

I believe it is just a typo in the meter_voltage address

wills106, 2022-11-07:
Updated 0.7.0b31

wills106, 2022-11-08:
Just released 0.7.0b32 with more sensors and hopefully working Number entities.

t0nyb, 2022-11-08:
[image]
[image]
[image]
[image]

wills106, 2022-11-08:
Is that on 0.7.0b30?

t0nyb, 2022-11-08:
yes haven't installed 31 yet will do soon

wills106, 2022-11-08:
You might as well go to 32

t0nyb, 2022-11-08:
V32 Log:

This error originated from a custom integration.

Logger: homeassistant.config_entries
Source: custom_components/solax_modbus/plugin_solis.py:182 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 09:26:40 (1 occurrences) 
Last logged: 09:26:40

Error setting up entry Inverter for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solax_modbus/__init__.py", line 92, in async_setup_entry
    plugin = importlib.import_module(f".plugin_{plugin_name}", 'custom_components.solax_modbus')
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 182, in <module>
    SolisModbusSensorEntityDescription(
TypeError: SolisModbusSensorEntityDescription.__init__() got an unexpected keyword argument 'fmt'

wills106, 2022-11-08:
I'll correct it. Will push a new release out in a few minutes.

wills106, 2022-11-08:
Just released 0.7.0b33

t0nyb, 2022-11-08:
V33 log

And no problem I'm happy to keep testing

This error originated from a custom integration.

Logger: homeassistant.config_entries
Source: custom_components/solax_modbus/plugin_solis.py:672 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 10:30:53 (1 occurrences) 
Last logged: 10:30:53

Error setting up entry inverter for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solax_modbus/__init__.py", line 92, in async_setup_entry
    plugin = importlib.import_module(f".plugin_{plugin_name}", 'custom_components.solax_modbus')
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 672, in <module>
    native_unit_of_measurement = POWER_VOLT_AMPERE_REACTIVE,
NameError: name 'POWER_VOLT_AMPERE_REACTIVE' is not defined

wills106, 2022-11-08:
Opps, forgot to update const.py as it's the first time I have used POWER_VOLT_AMPERE_REACTIVE

wills106, 2022-11-08:
0.7.0b34 corrects this.

t0nyb, 2022-11-08:
34 Log:

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:520 
Integration: Sensor (documentation, issues) 
First occurred: 12:03:57 (2 occurrences) 
Last logged: 12:04:27

Platform solax_modbus does not generate unique IDs. ID solvent_meter_activepower already exists - ignoring sensor.solvent_meter_activepower

Logger: homeassistant.util.async_
Source: util/async_.py:180 
First occurred: 12:04:12 (8 occurrences) 
Last logged: 12:04:43

Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 277: return self._client.read_input_registers(address, count, **kwargs)

t0nyb, 2022-11-08:
but sensors working wait for photos

t0nyb, 2022-11-08:
No problem

Waiting for my Apple photos to update from my ancient phone

t0nyb, 2022-11-08:
[image]
[image]
[image]
[image]
[image]

t0nyb, 2022-11-08:
[image]

wills106, 2022-11-08:
I think the ActivePower, House Load Today / Yesterday, Meter Current seem very high? Or do you have electric heating or/and an EV?
The rest seem to be ok. Can you verify any of them from your Inverter LCD?

t0nyb, 2022-11-08:
I will look. No EV or electric bar immersion heater sometimes

wills106, 2022-11-08:
I think I need to add scaling to your Meter Current and Meter Voltage. Your Voltage is 2451v I think it should be 245.1v

wills106, 2022-11-08:
I don't know much about ApparentPower, ActivePower, ReactivePower etc so not sure what they should even look like?

t0nyb, 2022-11-09:
Morning

Suggest that house load today, house load total, house load yesterday should be /10.

I have no idea what "meter" and its associated figures mean but they look suspect.

Thanks for your continuing efforts.

My son has the 5KWh version of the same inverter as me. I will forward his serial number when he is home today.

wills106, 2022-11-09:
Just released 0.7.0b36 with the mentioned changes.

wills106, 2022-11-10:
Did you try this on your Sons Inverter?

closed 2022-11-10 by wills106

t0nyb, 2022-11-11:
No but will do hopefully over the weekend. Will report.

t0nyb, 2022-11-11:
V40 log:

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:97 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 09:22:26 (1 occurrences) 
Last logged: 09:22:26

unrecognized solvent inverter type - serial number : 60312202232200

[12 more replies]