wills106/homeassistant-solax-modbus #198: House load and Solax grid import/export no longer available

markjsmith, 2022-11-09:
Since upgrading to the 0.7.0b series I
These entities are no longer available.
I switched to the beta at 0.7.0b26 and all versions since are the same.
I uninstalled and reinstalled the integration prior to upgrade as suggested.
Running 2022.11.2 with supervisor 20221108.0
No log errors 
Inverter is an XAC362……..
I’ve noticed also that the “Solax grid export limit” seems to be dynamic, showing positive and negative figures,similar to the reverse of the “inverter power” output, while I would have expected a limit to be static?
I’m away at the moment but will try and respond

wills106, 2022-11-09:
There have been a few name changes, I'll have to go back over the changelogs and check.

The grid exports are a little confusing.
number.solax_grid_export_limit is dynamic while at the same time you can also "manipulate it" to cause export.

It might make more sense to look at [Grid Export for Gen3](https://github.com/wills106/homeassistant-solax-modbus/wiki/SolaX-Gen2-&-Gen3-Modes-of-operation#grid-export-gen-3-only) which your Inverter is.

As this project started back in 2019 from manually probing the Modbus registers on my Inverter there is a combination of my Interpreted sensor definitions and SolaX naming convention's and I am in the tricky situation of do I clarify the sensor namings or while it break everyone's automations etc from name changes.

wills106, 2022-11-09:
So you have lost sensor.solax_grid_export & sensor.solax_grid_import ?

markjsmith, 2022-11-09:
Hi Will,
Yes and sensor.solax_house_load

wills106, 2022-11-10:
Just released 0.7.0b38

markjsmith, 2022-11-10:
Hi Will,
Just updated and that has got the sensors back, many thanks. I only spotted the other query when I was looking to see if another sensor was giving the readings, i don't use it myself 

Thanks 

Mark

wills106, 2022-11-10:
If everything is back to normal working I'll close down this issue then.

closed 2022-11-10 by wills106