wills106/homeassistant-solax-modbus #199: 0.7.0b20 SolaX Total Yield

CristiXP, 2022-11-10:
Since 0.7.0b20 SolaX Total Yield the value is showing 22338800 instead of 22338.8 for ex.

[image]

wills106, 2022-11-10:
When you upgraded to 0.7.0b20 did you remove the Integration from the Integration page and add it again after a restart?

We are now on 0.7.0b39 so we have had many code changed since then.

CristiXP, 2022-11-10:
Installed each of one after. Eachone  brakes the entity and shows x1000, but the 20. I am restoring from backup everytime :)

wills106, 2022-11-10:
Could you please add all the missing information from the Issue template that has been deleted.
[Link to issue template](https://github.com/wills106/homeassistant-solax-modbus/blob/main/.github/ISSUE_TEMPLATE/bug_report.md)

CristiXP, 2022-11-11:
---
name: Bug report
about: Create a report to help us improve
title: "[BUG]"
labels: ''
assignees: ''

---

**Describe the bug**
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version:0.7.0b39 
2. HA Version:Home Assistant 2022.11.2
3. Inverter Brand:Solax
4. Plugin Used:SolaX Inverter Modbus
5. First 6 digits of Serial:MP153T
6. Connection Method:modbus

**Detailed Error Log**
Detailed Error Log, if you press on the individual error further information is displayed.
This is the content we need, the description alone doesn't provide enough context.

If there are no log's showing for SolaX try the following depending on your language settings, you can find the full logs under:

Settings → System → Logs > at bottom of page press “LOAD FULL LOGS”

Now the full logs are loaded. If you scroll down, you will see them.
Once the full logs are shown, you can either use the search function in your browser to search for “solax” related entries or use the search entry field on top of the page.

Search for

solax

and report us the logs. Make sure to replace sensitive information by xxxx (if any)

**Additional context**
Add any other context about the problem here.

CristiXP, 2022-11-11:
This is what is shows on .39 for that entity.

[image]

wills106, 2022-11-11:
The X3 MIC / MIC Pro Gen1 has "Total Yield" as 0.001 scaling
The X3 MIC / MIC Pro Gen2 has "Total Yield" as 0.1 scaling.

I need to try and find a way to detect if the Inverter is Gen1 or Gen2 and adapt the scaling to suit.

wills106, 2022-11-11:
I think I have spotted the difference in Serial Numbers between the Gen1 and Gen2.
Could you give 0.7.0b40 a try please.

CristiXP, 2022-11-11:
Confirming that the issue has been fixed. Thank you.

wills106, 2022-11-11:
I'll close the issue down then thanks.

closed 2022-11-11 by wills106