wills106/homeassistant-solax-modbus #201: Unable to get any data from SoFar HYD 20KTL-3PH

reklame33, 2022-11-13:
Hello.
Just installed and powered up my brand new Sofar HYD 20KTL and connected wires to pin 1 and 4 on the com port on the inverter.
Installed the Solax Inverter Modbus plugin in Home Assistant through HACS.

Tried to connect both with USB to RS485 and ethernet to RS485 modules, using sofar.py protocol, but does not get any data into home assistant. No errors in log.

Tried to change to sofar_old.py protocol. Then it created a device in home assistant and 2 entities; run mode and serial. No data on either of them, and error in log also.

Now I have tried from a PC using ModScan32, it can connect and get data om input register 1-64 on both USB-to-RS485 and ethernet-to-RS485. The data does not seem to change though, but at least it has successful reading.

What might cause that the integration does not seem to be able to communicate with the inverter on any RS485 device?

wills106, 2022-11-13:
Does your Inverter Serial Number start SP1?

If you have working coms, but a different Serial Number you should at least get the following error in your log.

"unrecognized {hub.name} inverter type - serial number : {seriesnumber}" seriesnumber = your serial number.

reklame33, 2022-11-13:
Yes, serial SP1ES120Nxxxxx

Tried to delete integration and added back in. Using TCP, restarted HA, no errors in log.

wills106, 2022-11-13:
What version of the Integration are you trying to use?

reklame33, 2022-11-13:
Downloaded on HACS today, cannot find where to check for actual version.

reklame33, 2022-11-14:
Found version 0.7.0.4 in manifest file.

reklame33, 2022-11-14:
Here is a debug logging when HA restarts, serial masked with xxxx by me:

So reading the serial correct I assume the communication is ok, but integration will not pull any more data.

2022-11-14 21:27:21.167 DEBUG (MainThread) [custom_components.solax_modbus] using pymodbus library 2.x
2022-11-14 21:27:21.173 INFO (MainThread) [homeassistant.setup] Setting up solax_modbus
2022-11-14 21:27:21.195 INFO (MainThread) [homeassistant.setup] Setup of domain solax_modbus took 0.0 seconds
2022-11-14 21:27:21.198 INFO (MainThread) [homeassistant.setup] Setup of domain hardware took 0.2 seconds
2022-11-14 21:27:21.202 INFO (MainThread) [custom_components.hacs] <HacsData restore> Restore started
2022-11-14 21:27:21.237 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2022-11-14 21:27:21.237 DEBUG (MainThread) [custom_components.solax_modbus] setup entries - data: {}, options: {'name': 'sofar', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'custom_components/solax_modbus/plugin_sofar.py', 'scan_interval': 15, 'read_eps': True, 'read_dcb': False, 'read_pm': False, 'host': '192.168.10.37', 'port': 502}
2022-11-14 21:27:21.237 DEBUG (MainThread) [custom_components.solax_modbus] Ready to load plugin custom_components/solax_modbus/plugin_sofar.py
2022-11-14 21:27:21.243 DEBUG (MainThread) [custom_components.solax_modbus] Setup solax_modbus.sofar
2022-11-14 21:27:21.243 DEBUG (MainThread) [custom_components.solax_modbus] solax serial port /dev/ttyUSB0 interface tcp
2022-11-14 21:27:21.243 DEBUG (MainThread) [custom_components.solax_modbus] solax modbushub creation with interface tcp baudrate (only for serial): 19200
2022-11-14 21:27:21.243 DEBUG (MainThread) [custom_components.solax_modbus] solax modbushub done {'_hass': <homeassistant.core.HomeAssistant object at 0x7f7639e33d00>, '_client': <ModbusTcpClient at 0x7f762662ada0 socket=None, ipaddr=192.168.10.37, port=502, timeout=5>, '_lock': <unlocked _thread.lock object at 0x7f761e0d7240>, '_name': 'sofar', '_modbus_addr': 1, '_invertertype': 0, '_seriesnumber': 'still unknown', 'interface': 'tcp', 'read_serial_port': '/dev/ttyUSB0', '_baudrate': 19200, '_scan_interval': datetime.timedelta(seconds=15), '_unsub_interval_method': None, '_sensors': [], 'data': {}, 'cyclecount': 0, 'slowdown': 1, 'inputBlocks': {}, 'holdingBlocks': {}, 'computedRegs': {}, 'plugin_name': 'sofar'}
2022-11-14 21:27:21.243 DEBUG (MainThread) [custom_components.solax_modbus] sofar: ready to call plugin to determine inverter type
2022-11-14 21:27:21.243 INFO (MainThread) [custom_components.solax_modbus.plugin_sofar] sofar: trying to determine inverter type
2022-11-14 21:27:21.387 INFO (MainThread) [custom_components.solax_modbus.plugin_sofar] Read sofar 0x445 serial number: SP1ES120N3xxxx, swapped: False
2022-11-14 21:27:21.438 INFO (MainThread) [homeassistant.components.button] Setting up button.solax_modbus
2022-11-14 21:27:21.438 INFO (MainThread) [homeassistant.components.number] Setting up number.solax_modbus
2022-11-14 21:27:21.439 INFO (MainThread) [homeassistant.components.select] Setting up select.solax_modbus
2022-11-14 21:27:21.441 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
2022-11-14 21:27:21.449 DEBUG (MainThread) [custom_components.solax_modbus.sensor] holdingBlocks: []
2022-11-14 21:27:21.449 DEBUG (MainThread) [custom_components.solax_modbus.sensor] inputBlocks: []
2022-11-14 21:27:21.449 INFO (MainThread) [custom_components.solax_modbus.sensor] computedRegs: {}

wills106, 2022-11-16:
This Inverter doesn't like you reading large blocks of data.
There seems to be black holes in the register maps. That's why I have 'newblock = true' every so often.

Are you using a Waveshare? If so did you see the comment in the Installation section of the Wiki? "Modbus Gateway Type might need setting to "Auto query storage type" "

reklame33, 2022-11-16:
Not using Waveshare. I have testet different generic RS485-to-USB and RS485-to-ethernet adapters.
They all are able to read inverter serial number, but only intermittent random blocks of data from inverter to HA.

Maybe I can I add this newblock = true more places in the code to see if this helps. Maybe my inverters firmware are different from the N6 serial inverters regarding available registers and black holes.

wills106, 2022-11-16:
When that user had "Simple Modbus to TCP/IP mode" set on the Waveshare they we having issues like you are having, but when they changed to "Auto query storage type" the issues went away. So there might be a similar setting on your adaptor?

I would have assumed they will behave similar regardless of the firmware versions.
The Inverter doesn't like you reading over the "reserved" registers.

Try adding newblock between 251 and 252.

https://github.com/wills106/homeassistant-solax-modbus/blob/cdd47fe4fe0fa059449983e13f7243c69e0c62e0/custom_components/solax_modbus/plugin_sofar.py#L248-L255

wills106, 2022-11-17:
@reklame33 Did that newblock make any difference?

reklame33, 2022-11-18:
Yes, it's better, thanks for the tip. 
I get more data now, but uses a few minutes after restart of HA before they populate. Not sure I get them all as many of them show 0.
Have not tried to write back to inverter yet. Will update when done.

reklame33, 2022-11-19:
I do miss a entity for real time production power, same that show on inverter lcd. 

Do you know if this is available in modbus register? And maybe poll this i.e every 10 second?

reklame33, 2022-11-21:
I have done some extensive editing on the plugin_sofar.py to make units to be correct (as I understand the value). 
Also changed some typing errors.

Attached is the sofar and const file if it is any interesting for you. Not finished editing yet, but maybe you want to implement the new SensorDeviceClass enum on your future releases.
[plugin_sofar_new.zip](https://github.com/wills106/homeassistant-solax-modbus/files/10060913/plugin_sofar_new.zip)

wills106, 2022-11-24:
Where did you find out that device_class is changing? Also noticed state_class is changing as well. But that's only because I look at senssor.py on the main Home Assistant GitHub

reklame33, 2022-11-25:
Hello

found the info in home Assistant const file. 
https://github.com/home-assistant/core/blob/dev/homeassistant/const.py

wills106, 2022-11-26:
Ok thanks, I thought there might have been a proper communication that I had missed.
Home Assistant really are poor at communicating / documenting changes.

wills106, 2022-12-06:
@reklame33 I have just released 0.7.0b66 could you see if I matched all your Suggested changes? I have also corrected a few typos I have made. I didn't add in any of the extra ```newblock = True,``` as I am not sure if there were needed or not.

wills106, 2022-12-08:
@reklame33 Have you had time to check if the newer Beta's match up with your suggested corrections?

reklame33, 2022-12-08:
Hello. Just upgraded now. So far it seems to work and units seems to be correct. 
It still throws some errors in log, last 10 minutes:
This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: /config/custom_components/solax_modbus/__init__.py:365
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 16:47:34 (4 occurrences)
Last logged: 16:55:18

sofar error reading holding registers at device 1 position 0x1104
sofar error reading holding registers at device 1 position 0x58f
sofar error reading holding registers at device 1 position 0x484
NoneType: None

The reason for all the newblock was to avoid the register holes and be sure that there is no timeout. 

I am still unable to get it poll all settings every 15 seconds, did not work in my version either. 

Overall I am quite happy with this pluging to get data in real time, Solarman does not give me this option.

wills106, 2022-12-08:
I'm not sure how to get around the random not reading of registers.
The problem with adding additional '''newblock = True,''' is that you are breaking the reading down into smaller and smaller blocks. So you are increasing the time it takes to read everything.

As the Sofar Inverter's are only running at 9600. Maybe they are not designed to respond so quickly?

wills106, 2022-12-08:
If you have tried the really new release 0.7.0b71 do the selects provide the details of them instead of being blank?

They should no longer look like https://github.com/wills106/homeassistant-solax-modbus/issues/228#issue-1474870013

Also "Charger Use Mode" should actually work if you try and change the mode (Other selects / numbers won't be fully working yet.

reklame33, 2022-12-08:
Oh sorry, I tried the released version om hacs. 
installed b71 now. Seem to be more responsive and updating entities now is approx 30 second. 

Will try to change mode later. 

Any entity that will show which mode it is currently in?

wills106, 2022-12-09:
I am going to close this Issue down now as the basics are working. I'll leave #228 open as that centres around the fact the selects and numbers aren't fully functional.

Feel free to join into that conversation and then the topic is more focused to the one Issue.

closed 2022-12-09 by wills106