wills106/homeassistant-solax-modbus #205: Error in LOG - receiving error in log in Home Assistant

johnnyhero83, 2022-11-15:
A clear and concise description of what the bug is.

Not sure exactly, but receiev this error:

Logger: homeassistant.util.async_
Source: util/async_.py:180
First occurred: 20:39:14 (3817 occurrences)
Last logged: 21:04:46

Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 291: return self._client.read_holding_registers(address, count, **kwargs)
Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 297: return self._client.read_input_registers(address, count, **kwargs)

**Mandatory details**
1. Integration Version: 0.6.4
2. HA Version:  2022.11.2
3. Inverter Brand: Solax X3 hybrid 10D G4
4. Plugin Used: I dont understand what plugin
5. First 6 digits of Serial: H34A10I
6. Connection Method: USB to RS485 reduction in USB port on raspberrry pi4 connected via UTP cabel to COM port on Solax

**Detailed Error Log**

please see above the full description

**Additional context**
Add any other context about the problem here.

reklame33, 2022-11-15:
Not a fix, but I got the same error in my log. I have DEBUG log level so have some more info as well:

2022-11-15 21:57:34.600 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 271: return self._client.read_holding_registers(address, count, **kwargs)
2022-11-15 21:57:34.631 ERROR (MainThread) [custom_components.solax_modbus] sofarHYD error reading holding registers at device 1 position 0x418
NoneType: None
2022-11-15 21:57:34.632 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10

wills106, 2022-11-16:
The blocking calls is a known issue when using a RS485 - USB adaptor.
https://github.com/wills106/homeassistant-solax-modbus#known-issues

You could filter out the errors by the looks.
https://www.home-assistant.io/integrations/logger/#log-filters

johnnyhero83, 2022-11-16:
Thanks. So does it mean that it has no impact on anything. It is just a message that I can ignore? No need to worry about HA or about Solax as such? Thank you!

wills106, 2022-11-16:
We are not aware of any harm other than filling your log up. The serial routine needs looking at, but it's difficult to do as my Inverter has built in ethernet so I don't use an RS485 adaptor.

Most people are using a Waveshare which don't give the warnings as HA is communicating Modbus over TCP which is a different routine to Modbus RTU

wills106, 2022-11-29:
Closing down as not an issue as such as it's on the "Known Issues" list.

closed 2022-11-29 by wills106