wills106/homeassistant-solax-modbus #209: Support for Sofar HYD 10ktl-3ph Serial number : SP2ES110N6E056

juraurban, 2022-11-17:
Hi everyone, 
thank you for first promising integration for my solar inverter.

I have Waveshare RS485 -> WIFI - DINRAIL modul connected to inverter.
But as I have expected this integration does not support my model - would you be so kind and add the support.

Log says:
"unrecognized Sofar inverter type - serial number : SP2ES110N6E056"

Thanks a lot 
Jiri

wills106, 2022-11-17:
Just added the Serial for SP2 Inverters in 0.7.0b42

juraurban, 2022-11-17:
Thanks, It getting some data but most of sensor value is unknown except for:
[sensor.sofar_heatsink_temperature_1] = 38
[sensor.sofar_ambient_temperature_1] = 47
[sensor.sofar_module_temperature_1] = 39
[sensor.sofar_serial_number]= SP2ES110N6E056

Log:
2022-11-17 15:32:41.982 ERROR (MainThread) [pymodbus.factory] unpack requires a buffer of 2 bytes
2022-11-17 15:32:41.983 ERROR (MainThread) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 208, in execute
self.client.framer.processIncomingPacket(response,
File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
self._process(callback)
File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2022-11-17 15:32:41.986 ERROR (MainThread) [custom_components.solax_modbus] Sofar error reading holding registers at device 1 position 0x418
NoneType: None

wills106, 2022-11-17:
You can try 0.7.0b43 now.

juraurban, 2022-11-17:
Logger: custom_components.solax_modbus
Source: /config/custom_components/solax_modbus/__init__.py:337
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues)) 
First occurred: 3:48:54 PM (3 occurrences) 
Last logged: 3:53:39 PM

Sofar error reading holding registers at device 1 position 0x484
NoneType: None

Logger: pymodbus.factory
Source: custom_components/solax_modbus/__init__.py:271
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 3:48:54 PM (3 occurrences) 
Last logged: 3:53:39 PM

unpack requires a buffer of 2 bytes

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request

wills106, 2022-11-17:
Do you have a Modbus power meter on your Inverter or just CT Clamps?

wills106, 2022-11-17:
Would you be able to help trouble shoot by editing your local plugin_sofar.py file?
It will save me having to release so many releases.

I can explain what to do.

juraurban, 2022-11-17:
no problem I am available for 1 more hour to assist

juraurban, 2022-11-17:
what should I change in the file?

juraurban, 2022-11-17:
will give it a try, and let you know

juraurban, 2022-11-17:
I have been editing that line  - right now I am at - should I go on with that? - It seems none of that registers Is working:
SofarModbusSensorEntityDescription( 
     name = "Grid Frequency", 
     key = "grid_frequency", 
     native_unit_of_measurement = FREQUENCY_HERTZ, 
     device_class = DEVICE_CLASS_FREQUENCY, 
     register = 0x484, 
     newblock = True, 
     scale = 0.01, 
     rounding = 2, 
     allowedtypes = HYBRID | GEN2,
    ), 
    SofarModbusSensorEntityDescription(
        name = "Active Power Output Total",
        key = "active_power_output_total",
        native_unit_of_measurement = ENERGY_KILO_WATT_HOUR,
        device_class = DEVICE_CLASS_ENERGY,
        register = 0x485,
        unit = REGISTER_S16,
        scale = 0.01,
        rounding = 2,
        allowedtypes = HYBRID | GEN2,
    ),
    SofarModbusSensorEntityDescription(
        name = "Reactive Power Output Total",
        key = "reactive Power_output_total",
        native_unit_of_measurement = ENERGY_KILO_WATT_HOUR,
        device_class = DEVICE_CLASS_ENERGY,
        register = 0x486,
        unit = REGISTER_S16,
        scale = 0.01,
        rounding = 2,
        entity_registry_enabled_default=False,
        allowedtypes = HYBRID | GEN2,
    ),
    SofarModbusSensorEntityDescription(
        name = "Apparent Power Output Total",
        key = "apparent_power_output_total",
        native_unit_of_measurement = ENERGY_KILO_WATT_HOUR,
        device_class = DEVICE_CLASS_ENERGY,
        register = 0x487,
        unit = REGISTER_S16,
        scale = 0.01,
        rounding = 2,
        allowedtypes = HYBRID | GEN2,
    ),
    SofarModbusSensorEntityDescription(
        name = "Active Power PCC Total",
        key = "active_power_pcc_total",
        native_unit_of_measurement = ENERGY_KILO_WATT_HOUR,
        device_class = DEVICE_CLASS_ENERGY,
        register = 0x488,
        unit = REGISTER_S16,
        scale = 0.01,
        rounding = 2,
        allowedtypes = HYBRID | GEN2,
    ),
    SofarModbusSensorEntityDescription(
        name = "Reactive Power PCC Total",
        key = "reactive Power_pcc_total",
        native_unit_of_measurement = ENERGY_KILO_WATT_HOUR,
        device_class = DEVICE_CLASS_ENERGY,
        register = 0x489,
        unit = REGISTER_S16,
        scale = 0.01,
        rounding = 2,
        entity_registry_enabled_default=False,
        allowedtypes = HYBRID | GEN2,

juraurban, 2022-11-17:
I stopped at - added GEN2, but error appears "Sofar error reading holding registers at device 1 position 0x584
NoneType: None"

      SofarModbusSensorEntityDescription(
        name = "PV Voltage 1",
        key = "pv_oltage_1",
        native_unit_of_measurement = ELECTRIC_POTENTIAL_VOLT,
        device_class = DEVICE_CLASS_VOLTAGE,
        register = 0x584,
        newblock = True,
        scale = 0.1,
        rounding = 1,
        allowedtypes = HYBRID | PV | GEN | GEN2,

wills106, 2022-11-17:
I think you might be having some comms error then?

What's the exact model of WiFi Waveshare?

I don't have any proper confirmations of WiFi Waveshares working. Someone was trying to use one on a SolaX X3 Mic and I was having to tell the Integration to only read 7 registers at a time. Where Modbus is meant to be able to read over 100 at a time.

Do you have access to a RS485 - USB adaptor?

juraurban, 2022-11-17:
[I have ](https://www.amazon.com/Rail-Mount-Converter-MultiTransparent-Transmission-Configuration/dp/B0B54TGG1R)

juraurban, 2022-11-17:
https://www.amazon.com/Rail-Mount-Converter-MultiTransparent-Transmission-Configuration/dp/B0B54TGG1R

juraurban, 2022-11-17:
unfortunately dont have RS485 - USB adaptor, any advice how to test it anyway?

juraurban, 2022-11-17:
I saw that note in wiki, but I could not find any option like this. This is what I can see:

[image]
[image]

wills106, 2022-11-17:
Looks like you have the same Waveshare as the X3 MIC that's having issues.

Is there any chance you can get Ethernet to your Inverter?

juraurban, 2022-11-17:
Ok ... I will get ethernet one and test again

wills106, 2022-11-17:
I don't want people to spend more money on their hobbies. But I have no way of confirming if these WiFi ones work correctly.
So I don't know if we are wasting time trying to trouble shoot a problem that doesn't exist, if the WiFi ones truly aren't compatible.

juraurban, 2022-11-17:
Did you mean the functionality which is described in ethernet only Waveshare manual in chapter 7.2?
https://www.waveshare.com/w/upload/6/66/RS485-to-eth-b-user-manual-EN-v1.3.pdf

juraurban, 2022-11-17:
that version apparently has some kind of cache which send the requests in batches if I get it right

wills106, 2022-11-17:
I don't connect via RS485. So I don't know the differences nor have the ability to test anything sorry.

juraurban, 2022-11-17:
ok. I will order ETH version and let you know, if it works. Thanks a lot for support

juraurban, 2022-11-23:
ETH version at home, had to set "autostorage parameter" and up and running smoothly ;) 
Have used updated sofar files from [reklame33] which helped to fix some entity values.

Thank a lot for support

wills106, 2022-11-24:
I haven't looked into the changes yet, I have been too busy.

I am glad the Ethernet Version is working. I'll add a note to the Wiki saying the WiFi version isn't supported.

wills106, 2022-11-26:
When you turned on "Auto query storage type" Did "Enable RS485 Multi-Host" & "Enable RS485 bus conflist detection" become enabled as well?

I am trying to work out what #204 is required to do in order to get their Sofar Inverter working with the Waveshare.

wills106, 2022-11-26:
Are the connections on your KTL inver the same as what's on @superino1 ES Inverter?

Also did you use the default address of 1 or was your Inverter set different?

[image]

superino1, 2022-11-26:
Where do I find that value relating to my inverter?  I tried 1 and 4

juraurban, 2022-11-26:
1) Did "Enable RS485 Multi-Host" & "Enable RS485 bus conflist detection" become enabled as well? >> yes both, without that wasn’t able to save the settings 
2) address is default = 1

juraurban, 2022-11-26:
Reading is fine, but I would also like to tell inverter to charge at specific time - when spot price is cheaper. 
How can I make this happen?

superino1, 2022-11-27:
Ma non è che la mia porta sia disabilitata? La spia blu su ACT sta sempre accesa fissa.. Confermo che l'indirizzo è 1

superino1, 2022-11-27:
I'm about to throw it all in the trash😔😔😃

wills106, 2022-11-29:
I am going to close down as the Integration now supports 'SP2' Inverters.

Any further issues, please raise a new issue.

closed 2022-11-29 by wills106