wills106/homeassistant-solax-modbus #210: [BUG] Unrecognized Solis inverter type RHI-3P8K-HVES-5G

dewafre, 2022-11-18:
Hello guys,
I dont get any entities in HA and the log says "unrecognized SolisSolax inverter type"

**Mandatory details**
1. Integration Version: 
2. HA Version:
Home Assistant 2022.11.3
Supervisor 2022.10.2
Operating System 9.3
Frontend-versie: 20221108.0 - latest
3. Inverter Brand: Solis RHI-3P8K-HVES-5G
4. Plugin Used: Solis
5. First 6 digits of Serial: 110C92*********
6. Connection Method: LAN

**Detailed Error Log**
Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:97
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 16:33:58 (1 occurrences)
Last logged: 16:33:58

unrecognized SolisSolax inverter type - serial number : 110C9222******

**Additional context**
Add any other context about the problem here.
I used the IP of the logger + port 8899

wills106, 2022-11-18:
Just released 0.7.0b44 which should work with your Inverter now.

dewafre, 2022-11-19:
Nice, it works perfect!!!
Thx alit

dewafre, 2022-11-20:
Hello after reading the wiki and trying to install the energy dashboard I have one little problem.
Can I refer to the https://github.com/wills106/homeassistant-solax-modbus/discussions/211#discussion-4583863

Thx alot !

dewafre, 2022-11-21:
> Hello after reading the wiki and trying to install the energy dashboard I have one little problem. Can I refer to the [#211 (comment)](https://github.com/wills106/homeassistant-solax-modbus/discussions/211#discussion-4583863)
> 
> Thx alot !

Hello @wills106 

I think i solved the problem by adding **scale = 0.1,** in plugin_solis.py
Is this correct?

wills106, 2022-11-24:
It's built into the dashboard to reset production values if it drops to zero. (STATE_CLASS_TOTAL_INCREASING)

Does your grid_import_total drop to zero as well? If not you could use that in your dashboard instead?

dewafre, 2022-11-25:
> It's built into the dashboard to reset production values if it drops to zero. (STATE_CLASS_TOTAL_INCREASING)
> 
> Does your grid_import_total drop to zero as well? If not you could use that in your dashboard instead?

_sensor.solis_grid_import_today => solved scale = 0.1
sensor.solis_meter_grid_import_total => solved scale = 0.01
sensor.solis_grid_export_today => solved scale = 0.1
sensor.solis_meter_grid_export_total => solved scale = 0.01
sensor.solis_meter_total_activepower => solved scale = 0.01_

Ok i will try sensor.solis_meter_grid_import_total & sensor.solis_meter_grid_export_total instead in the energy dashboard and let you know if works OK

wills106, 2022-11-25:
It should work fine, we have to do a similar thing on the Gen2 SolaX.

On the Gen3 & Gen4 you have "Battery Input Energy Today" which we use for the Dashboard.

The Gen2 doesn't have that register, so you have to use "Battery Input Energy Total"

wills106, 2022-11-26:
@kai-seipp do the above sensors read ok on your Inverter or do they need scaling how @dewafre has suggested?

kai-seipp, 2022-11-26:
Hey Guys, 

I don't trust my inverters import / exports readings at all. 
I use a shelly 3EM to monitor these values. So I have not yet really checked them. 

But yes they do need scaling from what I can see. 
Here is what I could find: 

Grid Import Yesterday Shelly: 7,66 kWh
Grid Import Yesterday Solis South Inverter: 84 kWh
Grid Import Yesterday Solis North Inverter: 60 kWh

Grid Export Yesterday Shelly: 3,09 kWh
Grid Export Yesterday Solis South Inverter: 12 kWh
Grid Export Yesterday Solis North Inverter: 12 kWh

wills106, 2022-11-27:
@blackwhitebear8 @dewafre @kai-seipp 
With 0.7.0b47 do your scaling look correct now?
Also are your timed charge / discharge looking more normal now?
You times for example should be 0-23 hours and 0-59 mins. Before they were probably all random looking?

blackwhitebear8, 2022-11-27:
The update broke more here xd
[image]
Theres no battery connected still

wills106, 2022-11-27:
Can you share the exact entity names for those two broken ones please.

wills106, 2022-11-27:
I haven't changed any battery entities so I don't know what's going on with the battery one. Unless you added something that wasn't a battery there?

blackwhitebear8, 2022-11-27:
I saw there were a few more broken ones
Meter active power
Battery power
And house load becouse it doesnt do anything anymore

wills106, 2022-11-27:
@blackwhitebear8 could I have a screen shot of how they look and a description of what they should be. I don't know if I have scaled too high or too low.

blackwhitebear8, 2022-11-27:
well battery should be 0
meter active power is idk what it needs to be becouse house load doesnt show anything anymore
[image]

blackwhitebear8, 2022-11-27:
i m also seeing this now:
Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:226
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 12:27:02 (12 occurrences)
Last logged: 12:27:08

Solis: read failed at 0x8199: house_load_total
Solis: read failed at 0x819b: house_load_today
Solis: read failed at 0x819c: house_load_yesterday
[image]

wills106, 2022-11-27:
"Battery Power" and "House Load" haven't been changed in a while.

Does 0.7.0b48 correct the rear fail of:
Solis: read failed at 0x8199: house_load_total
Solis: read failed at 0x819b: house_load_today
Solis: read failed at 0x819c: house_load_yesterday

blackwhitebear8, 2022-11-27:
Doesnt look like it i also see this now

Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.solax_modbus.sensor
Source: custom_components/solax_modbus/sensor.py:124
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 12:41:21 (1 occurrences)
Last logged: 12:41:21

input register already declared: 0x816a energy_storage_control_switch

wills106, 2022-11-27:
0.7.0b49 will correct "Energy Storage Control Switch"

blackwhitebear8, 2022-11-27:
everything seems back to normal now again only meter active power is still saying 300.000W and by sensors its still expecting KWH for some sensors

wills106, 2022-11-27:
> meter active power is still saying 300.000W

I am guessing it isn't expecting any round then, it should just be 300w

> still expecting KWH for some sensors

Could you let me know which ones and I can correct the native_unit_of_measurement for them.

blackwhitebear8, 2022-11-27:
solis meter active power is in kwh (maybe that explains why it has .000 when its displayed in W)
more sensors i havent added / in use atm that do this

kai-seipp, 2022-11-27:
> @blackwhitebear8 @dewafre @kai-seipp With 0.7.0b47 do your scalings look correct now? Also are your timed charge / discharge looking more normal now? You times for example should be 0-23 hours and 0-59 mins. Before they were probably all random looking?

The entity sensor.solis_meter_active_power must be in W not in kWh. 
Scaling of Grid Import / Export looks good now. Still don't Trust the numbers though... 

I did see you introduced a control for different programs of the inverter. Nice! 
My suggestion is to use the working of what people get in that little documentation Solis is sharing... 
[image]

So I assume "Spontaneous Mode" should be "Auto Mode"?

wills106, 2022-11-27:
Could I have some screen shots of these sensors from the Integration page an not a dashboard.

"Meter Active Power"
"Meter Active Power Total"

If any 3 Phase also:
 
"Meter Active Power L1"
"Meter Active Power L2"
"Meter Active Power L3"

It's hard to tell if the dashboard is doing stuff to the values or not, I need to see the raw numbers then I can correct them.
According to the docs they should all be kW

> So I assume "Spontaneous Mode" should be "Auto Mode"?

Yea I was just going of the docs, I thought it was a funny name!

blackwhitebear8, 2022-11-27:
Meter Active Power
[image]
Meter Active Power Total
[image]
Meter Active Power L1
[image]
Meter Active Power L2
[image]
Meter Active Power L3
[image]

wills106, 2022-11-27:
I forgot there are two Meter Active Power in the docs!
I was looking at the wrong one...

So on a 3Phase the following are correct?

"Meter Active Power Total"
"Meter Active Power L1"
"Meter Active Power L2"
"Meter Active Power L3"

wills106, 2022-11-27:
For
"Meter Active Power L1"
"Meter Active Power L2"
"Meter Active Power L3"
"Meter Active Power Total"
Could I have a screen shot of all 4 of them from the Integration page like:

[image]

That way I am see them in the same time instance.
I am assuming L1 + L2 + L3 should = Power Total?

blackwhitebear8, 2022-11-27:
[image]

wills106, 2022-11-27:
So when Active Power Total was -0,488kW was Active Power 488w?

blackwhitebear8, 2022-11-27:
yes and the - means its being imported - bought atm

wills106, 2022-11-27:
Right ok, I understand what's going on now. I'll make some corrections.

wills106, 2022-11-27:
0.7.0b50 should correct this now.

blackwhitebear8, 2022-11-27:
the extra 0 are gone now

wills106, 2022-11-27:
Do any other sensors seem to be either named incorrect or scaled incorrect?

blackwhitebear8, 2022-11-27:
is there a way to get all sensors in a dashboard at once so i can look at them?

wills106, 2022-11-27:
Best way it to take a screenshot of the Integration Page.

Or create a blank Dashboard page and add them all using the Add to Dashboard Button. They might be more compact that way.

blackwhitebear8, 2022-11-27:
what do you mean with intergration page?

wills106, 2022-11-27:
Settings / Devices & Services / 
[image]

You should see a big list.

blackwhitebear8, 2022-11-27:
you can change configs via there of the inverter????

wills106, 2022-11-27:
If you click that 1 device you will see all of the entities.

There is the option to set charge / discharge times etc. But untested.

blackwhitebear8, 2022-11-27:
i hope you can do something with this
[image]

wills106, 2022-11-27:
It's a bit low quality and some the units of measurement are chopped off.

But nothing obviously wrong is sticking out to me?

blackwhitebear8, 2022-11-27:
battery voltage shows 1.7V but there is no battery connected?
what is a bypass load?
meter current shows high A at all times i dont think theres 18amps flowing though it when theres only 1000 watts of usage
and is there a sensor for inverter errors?

kai-seipp, 2022-11-27:
You can check most of the values shown in HA by comparing them to the ones shown within the Inverters menu. Go there on Info, Meter and click through those. 

This integration can only show what the inverter is passing on.

wills106, 2022-11-27:
Battery voltage - It might just not be very accurate at low voltages, so it's a phantom rading.
Bypass load - no idea to be honest, it's just in the docs. Anything in your User Manual?
Meter Current - do you think it should be 1.8A?
Inverter Errors - Yes, but not yet implemented. I need to add reading individual bits of a register. It's on the Todo List V2. It's needed for SolaX as well.

blackwhitebear8, 2022-11-27:
meter current isnt right if it was 1.8 it would be too low idk how it calculates it

kai-seipp, 2022-11-27:
I checked meter current against my 3EM, yes the decimal must go one to the left. 13,0 should be 1,30

wills106, 2022-11-27:
My house load is about 350w ~ and the Inverter current is 1.6A to 1.7A

blackwhitebear8, 2022-11-27:
my load is about 775W and the current shows 22,8A

blackwhitebear8, 2022-11-27:
2.28A cant be right either?

kai-seipp, 2022-11-27:
> meter current isnt right if it was 1.8 it would be too low idk how it calculates it

Please check against the values your inverter or what your meter display shows.

wills106, 2022-11-27:
Yea that works out there about's 415v = 1.86A

blackwhitebear8, 2022-11-27:
meter shows around 5.1A atm

blackwhitebear8, 2022-11-27:
20.04 is displayed in HA

kai-seipp, 2022-11-27:
> 2.28A cant be right either?

Why not, seams quite plausible to me.

blackwhitebear8, 2022-11-27:
2.28A with 775W load is not correct 😂 should be around 3.3A then

kai-seipp, 2022-11-27:
Well I will call it a Sunday for today. Maybe we should all try to bundle our feedback and requests a little. I would hate to see Wills personal battery at low power.

wills106, 2022-11-27:
Well some of your PV will be powering things. Not just grid pull.

wills106, 2022-11-27:
You will never get things to add up 100%
PV, Battery (when you get one) Grid are all at different voltages. There will also be losses through conversions etc.

I think that 18A does need changing to 1.8A though. It will be close enough.

[11 more replies]

closed 2022-11-29 by wills106