wills106/homeassistant-solax-modbus #214: [BUG]

blackwhitebear8, 2022-11-25:
**Describe the bug**
I m getting: unrecognized Solis inverter type - serial number : unknown and Solis: reading serial number from address 0x80ec failed; other address may succeed

I m using a DSD TECH SH-U10 rs485 to usb adaptor it has a silicon labs cp2102 chip in it. its correctly wired up to our solis 10KW Hybird inverter

**Mandatory details**
1. Integration Version: 0.7.0.5
2. HA Version: Home Assistant 2022.11.4
3. Inverter Brand: solis
4. Plugin Used: solis
5. First 6 digits of Serial: 110CA2
6. Connection Method: Serial (DSD TECH SH-U10 rs485 )

**Detailed Error Log**
Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:84
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 14:50:20 (1 occurrences)
Last logged: 14:50:20

Solis: reading serial number from address 0x80ec failed; other address may succeed

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:86
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 14:50:20 (1 occurrences)
Last logged: 14:50:20

Solis: cannot find serial number, even not for other Inverter

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:97
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 14:50:20 (1 occurrences)
Last logged: 14:50:20

unrecognized Solis inverter type - serial number : unknown

If there are no log's showing for SolaX try the following depending on your language settings, you can find the full logs under:

Settings → System → Logs > at bottom of page press “LOAD FULL LOGS”

Now the full logs are loaded. If you scroll down, you will see them.
Once the full logs are shown, you can either use the search function in your browser to search for “solax” related entries or use the search entry field on top of the page.

Search for

solax

2022-11-25 15:00:04.261 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] Solis: reading serial number from address 0x80ec failed; other address may succeed
2022-11-25 15:00:04.264 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] Solis: cannot find serial number, even not for other Inverter
2022-11-25 15:00:04.265 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized Solis inverter type - serial number : unknown

**Additional context**
Add any other context about the problem here.

wills106, 2022-11-25:
Are you using the default baud rate of 9600?

Have you previously been able to talk to this Inverter either using the Modbus built into HA or a program like Gmodbus?

blackwhitebear8, 2022-11-25:
I m using the default 9600 yes and no i have not tried other programs yet

wills106, 2022-11-25:
Are you sure you have the data pins the correct way round?
With Modbus A goes to A and B goes to B. They don't need swapping over unlike a Null Modem Cable.

blackwhitebear8, 2022-11-25:
a goes to a and b goes to b

blackwhitebear8, 2022-11-25:
i used this picture for the pinout on the inverter itself https://community-assets.home-assistant.io/original/4X/7/5/0/750b95a33f06f15259bee9bd843ef19fc27e56b6.jpeg

Pin 3 goes to a+ and pin 4 to a-

wills106, 2022-11-25:
@kai-seipp are you able to confirm if this Diagram is correct as I believe you have the same Inverter.
Did you have to enable anything of the Inverter as well? As there doesn't appear to be any communications if it is connected correctly?

kai-seipp, 2022-11-25:
Hey Guys, 

The pins are correct 3 / 4 are Modbus A and B. So that is correct. What took me a while to figure out is that I had to use the "COM" Port (green) and not the MODBUS Port as one could expect... ;) 

And of cause you have to check the Address given within the inverters Menu so it matches the Modbus ID you specify within this script on setup. 

You don't need to use ground or if (to shield the wire) you don't connect it to your inverter but just to your Modbus communication module. 

Hope this helps!

blackwhitebear8, 2022-11-25:
"The pins are correct 3 / 4 are Modbus A and B. So that is correct. What took me a while to figure out is that I had to use the "COM" Port (green) and not the MODBUS Port as one could expect... ;)" what do you mean with com port green etc? i can see the device in side linux as /dev/ttyUSB0? and on the inverter we dont have any settings for modbus or com

blackwhitebear8, 2022-11-25:
also can this come becouse of the cable length between the inverter and the serial usb? (around 10m)

wills106, 2022-11-25:
RS485 can go upto 1200m if Terminated correctly.

I'm not sure at what exact distance you need to ground the cable and possibly introduce a terminating resistor at.

blackwhitebear8, 2022-11-25:
i dont have it grounded atm is that required? and if so how would i do that

blackwhitebear8, 2022-11-25:
this is what it looks on the pc/servers side btwhttps://cdn.discordapp.com/attachments/805950455359602709/1045735535987142806/IMG_20221125_171832.jpg

kai-seipp, 2022-11-25:
[image]

This image shows tie two ports on my Solis inverter. 

To set modbus address I have to go to "settings" and than to "Address". The number I define there is the modbus ID.

blackwhitebear8, 2022-11-25:
i put the cables in the COM/CAM port does it need to be in the rs one somehow?

kai-seipp, 2022-11-25:
[image]

If you scroll in you can see the pin IDs inside the socket.

blackwhitebear8, 2022-11-25:
looks correctly wired up to me then

kai-seipp, 2022-11-25:
> https://cdn.discordapp.com/attachments/805950455359602709/1045735535987142806/IMG_20221125_171832.jpg

Do you have an image where we can see the labels on the ports?

kai-seipp, 2022-11-25:
> i dont have it grounded atm is that required? and if so how would i do that

I don't think you have to ground it for 10m. But this depends on the interference on the way. But at a private home it should be ok. If you want to try: you would just have to find the ground connector on your usb dongle. And connect one of the extra wires to it. That's it. It would not be connected anywhere on the inverters side.

blackwhitebear8, 2022-11-25:
i have re attached all the 4 wires and pins so the usb gets also 5v and ground from the inverter and lets see tomorrow what it does when the sun is backup :p becouse ye rn the inverter is offline

wills106, 2022-11-25:
It looks like @blackwhitebear8 is using Unshielded CATx cable?
I'm not sure if grounding one of the other twisted pairs would offer much help? I would assume you would have to ground the outer shielding for it to make a difference?

Are you able to use a shorter cable just for test purposes? Could you use a laptop next to your Inverter?

I don't it's necessary to connect the 5v between devices.

Like @kai-seipp mentions are you able to show a picture of how you are connected at the Inverter?

blackwhitebear8, 2022-11-25:
https://cdn.discordapp.com/attachments/1026751450874904598/1045782707982041170/IMG_20221125_201936.jpg
https://cdn.discordapp.com/attachments/1026751450874904598/1045782708204359800/IMG_20221125_202058.jpg
connections ^^ and its a shielded cat6 cable the shielding is cut off where the plastic stops.

kai-seipp, 2022-11-25:
I would be careful using pins 1 and 2. I don't have it to hand but if I remember correctly one is feeding 5V and the other 24V for the Solis data stick. I did find a breakout sheet somewhere on the Solis help portal. But this is just out of my head. But anyway: there really is no need to use anything else but 3 and 4. I only used these and used totally unshielded single wires you would usually use to connect a door bell. But I only needed to bridge about 1m.

blackwhitebear8, 2022-11-25:
i had found this diagram and it said 5v and ground https://community-assets.home-assistant.io/original/4X/7/5/0/750b95a33f06f15259bee9bd843ef19fc27e56b6.jpeg

kai-seipp, 2022-11-25:
> i had found this diagram and it said 5v and ground https://community-assets.home-assistant.io/original/4X/7/5/0/750b95a33f06f15259bee9bd843ef19fc27e56b6.jpeg
> 
> 

Checked it with the Solis portal. Seams to be correct! 

https://ginlongsolis.freshdesk.com/support/solutions/articles/36000012141-rs-485-wiring-with-circular-connector

kai-seipp, 2022-11-25:
Solis: 

A typical wiring scheme calls for an IN and an OUT, that means that each run will have a positive, a negative, a common wire, and a ground. The common wire isn't needed here, but the + and - are. Think of A as positive and B as negative. Each in / out has both a positive (A) or a negative (B).

kai-seipp, 2022-11-25:
Did you check the address set within your inverter? Does it match the one set here at the integration?

blackwhitebear8, 2022-11-25:
where can i check this? i have never seen any options for serial or modbus connections in it?

kai-seipp, 2022-11-25:
Follow this here. https://user-images.githubusercontent.com/22188947/204049730-61f2cafc-ab85-484f-ac9d-2de318f1dbba.MOV

blackwhitebear8, 2022-11-25:
will try this tomorrow morning ty. Also do you have abatterij or do you live somewhere where its still day now?

kai-seipp, 2022-11-25:
I have a battery. This is why one of my two inverter is still running. 

The other one is dead now as well.

blackwhitebear8, 2022-11-25:
we have a battery aswell but its LV and ours only accepts HV.... this wasnt clearly documented on it or its papers it came with..

kai-seipp, 2022-11-25:
(I am a neighbor from Germany.) So also no sun here. 

I really? I didn't know Solis only accepts HW. I have a pylontech force H2.

blackwhitebear8, 2022-11-25:
maybe this is not the right place for this but do you maybe have discord? that I can contact you on?

wills106, 2022-11-25:
Unfortunately it gets confusing what is classed as HV and LV from a battery perspective.

I have seen the 48v Pylontech batteries rebranded as both Low Voltage and High Voltage.
As they get connected in parallel I would class them more as LV.

Unlike the SolaX which are about 105 - 110v each and get wired in series. I have 3 batteries connected to my Gen3 SolaX which is around 315v at the moment.

To confuse matters with Solis you seem to have two side by side Inverters:

RHI-nK-48ES-5G
RHI-nK-HVES-5G

(n = kW rating)

kai-seipp, 2022-11-25:
Solis is the worse when it comes to documentation! Everything I know about their deceives I don't know from them... 

Discord: a good idea, maybe we can setup a channel for this integration. I have it on my work Mac installed. Not using it much though as we use slack for work.

blackwhitebear8, 2022-11-26:
after setting the address it started doing more but now i m getting this:

Logger: homeassistant.util.async_
Source: util/async_.py:180
First occurred: 10:04:00 (560 occurrences)
Last logged: 10:08:39

Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 277: return self._client.read_input_registers(address, count, **kwargs)

wills106, 2022-11-26:
The blocking calls is a known issue with using RS485 - USB https://github.com/wills106/homeassistant-solax-modbus#known-issues

It's nothing to worry about. Are you getting any entities returning values now?

blackwhitebear8, 2022-11-26:
yes i m getting data

wills106, 2022-11-26:
You could possibly configure the logger to ignore the blocking calls?
https://www.home-assistant.io/integrations/logger/

blackwhitebear8, 2022-11-26:
is there a way to get thise live?
[image]
i put these in here but i want to confirm these are the right ones
[image]

wills106, 2022-11-26:
There is an example here how to have it live with an additional card. https://github.com/wills106/homeassistant-solax-modbus/wiki/Energy-Dashboard's

Top left should be Solis Grid Import Today.

blackwhitebear8, 2022-11-26:
the live card says the house is using as much as there is being generated? nothing imported or exported

wills106, 2022-11-26:
The Power Distribution card? or the screen shot you posted just?

blackwhitebear8, 2022-11-26:
The Power Distribution card
[image]

wills106, 2022-11-26:
Can you share what entities you used for it?

blackwhitebear8, 2022-11-26:
[image]

wills106, 2022-11-26:
solis_measured_power & solis_battery_power_charge don't exist.
We need to work out what the equivalent is for the Solis.

blackwhitebear8, 2022-11-26:
also i have found out why the usage is as high as generation the 3 phase meter seems to be sending out 0 for everything to the inverter

wills106, 2022-11-26:
For the battery try "Battery Power"

blackwhitebear8, 2022-11-26:
we have no working battery on it yet so that will stay 0/null xd

wills106, 2022-11-26:
If you select the correct entity it might show 0? Not sure. battery_power_charge is a SolaX name, just need to drop the _charge.

wills106, 2022-11-26:
For the Grid try "Meter Active Power" that should be the 3 Phases combined.

blackwhitebear8, 2022-11-26:
ye i had found that just now but it shows 0 just like on the inverter is receiving 0 from the meter for some reason

wills106, 2022-11-26:
What about the one without meter?
"Active Power"

wills106, 2022-11-26:
> the 3 phase meter seems to be sending out 0 for everything to the inverter

Is this a Modbus Power meter such as a Eastron or similar? Was it supplied with the Solar Installation?

blackwhitebear8, 2022-11-26:
i have corrected the meter settings inside solis and now its sending out the correct usage

wills106, 2022-11-26:
If some of your reading still seem out, there might be further work required to correct them. #210

kai-seipp, 2022-11-26:
> i have corrected the meter settings inside solis and now its sending out the correct usage

I have had the same issue: My installing company did configure an Eastron Meter, but delivered a Acrel Meter... That did not work at all...

blackwhitebear8, 2022-11-26:
anyone know how to configure a card like this with these meters? https://github.com/ulic75/power-flow-card

kai-seipp, 2022-11-26:
for your information (from an other thread): 

I don't trust my inverters import / exports readings at all.
I use a shelly 3EM to monitor these values. So I have not yet really checked them.

But yes they do need scaling from what I can see.
Here is what I could find:

Grid Import Yesterday Shelly: 7,66 kWh
Grid Import Yesterday Solis South Inverter: 84 kWh
Grid Import Yesterday Solis North Inverter: 60 kWh

Grid Export Yesterday Shelly: 3,09 kWh
Grid Export Yesterday Solis South Inverter: 12 kWh
Grid Export Yesterday Solis North Inverter: 12 kWh

[22 more replies]

closed 2022-11-29 by wills106