wills106/homeassistant-solax-modbus #215: Home Assistant shows incorrect Battery % in HA Device Page

jamespreedy, 2022-11-26:
Hiya, hope you're well. If you need any additional information on this one, please let me know.
Thanks in advance!

Go to ... Settings => Devices & Services => SolaX => 1 Device
In the top right of the screen, to the left of the SolaX Power, it shows a % and a battery symbol.

For all other battery powered devices I have in Home Assistant, this illustrates the remaining battery % for that device.
For this integration, it doesn't represent the remaining charge of my SolaX triple power batteries.

For example, currently, it shows 30%.

[image]

However, my current battery capacity is 12%.

[image]

**Mandatory details**
1. Integration Version: 0.6.4
2. HA Version: 2022.11.4
3. Inverter Brand: SolaX
4. Plugin Used: ?
5. First 6 digits of Serial: H34A10
6. Connection Method: Waveshare RS485 to RJ45 Ethernet DIN Rail Mount

wills106, 2022-11-26:
On the Gen4 Inverters we have incorrectly added "device_class=DEVICE_CLASS_BATTERY," to some extra entities that shouldn't have it, like your "Backup DIscharge Min SOC"

HA has probably decided that's your battery level through alphabetical order.

wills106, 2022-11-26:
The new release 0.7.0b45 will fix this.

As you are on 0.6.4 pay attention to the note about removing the Integration before upgrading.

jamespreedy, 2022-11-26:
You're a gent, thank you very much.

Stupid question of the day. I have more entities in HA than configuration options on the iOS SolaX app. The SolaX PDF manual and Installation PDFs don't mention many of the entities I have as from this add-on. Is there a document somewhere that describes what each of these entities / options do? Perhaps a Solax Modbus protocol spec doc?

Thanks again.

wills106, 2022-11-26:
For the Gen4 you have an [overview here](https://github.com/wills106/homeassistant-solax-modbus/wiki/SolaX-Gen4-Modes-of-Operation)

You are best looking around in the discussion part, or creating one if there isn't anything suitable.

I'll close this issue down now the battery % is fixed.
But like I say feel free to use the discussion section while we try and workout the different feature etc.

closed 2022-11-26 by wills106