wills106/homeassistant-solax-modbus #219: [BUG] Multiple Inverters may not work well

infradom, 2022-11-28:
**Describe the bug**
When declaring 2 inverters with different IP addresses, the second inverter seems to use the IP address of the first inverter.
Problably an error in the config_flow 

**Mandatory details**
1. Integration Version: 6.x, 7.x
2. HA Version: recent 
3. Inverter Brand: Solax , but other brands may suffer from the same problem
4. Plugin Used: plugin_solax
5. First 6 digits of Serial: H34T08
6. Connection Method: TCP

**Detailed Error Log**
No relevant infromation found in logs, I may need to add more debug output

**Additional context**
Only relevant to users with multiple inverters

wills106, 2023-01-05:
I'm thinking of closing this issue down as I am not aware of any Issues.

@matthewjporter you are not having any issues are you where the second Inverter mimic's the first?
I can't really think of anyone else with dual Inverters?

wills106, 2023-01-21:
Going to close down as had no reports of issues with Multiple Inverters.

closed 2023-01-21 by wills106