wills106/homeassistant-solax-modbus #223: [FEATURE REQUEST] Pgrid Bias setting

alibasta, 2022-11-29:
With the latest FW update G4 inverters received a Pgrid Bias setting with the following options:

[image]

Currently this value does not seem to be visible in the HA integration.

wills106, 2022-11-29:
Yes, I can add this in.

alibasta, 2022-11-29:
I'm able to try it out manually. Is it already pushed?

wills106, 2022-11-29:
Yes

alibasta, 2022-11-29:
I updated the plugin_solax.py containing the "Pgrid Bias" setting and restared the server. "Pgrid Bias" is _not_ visible as entity. Is there anything more I need to do?

wills106, 2022-11-29:
Is the select not visible or the sensor or both?
The sensor isn't visible by default but the select should be.

If it's not try removing the Integration from the Integration page and re-adding it.

wills106, 2022-11-29:
If you are on 0.6.4 you will need to move to 0.7.0bx I have just released 0.7.0b58 with the changes in case you are still on 0.6.4

alibasta, 2022-11-29:
Thanks!! It works as it is supposed to.

closed 2022-11-29 by alibasta