wills106/homeassistant-solax-modbus #225: [BUG]Only 4 entities appearing  Solax G4 X1 7.5

pocahozas, 2022-11-30:
Hello.
After installing and configuring  de integration there is only one device and four entities:
Battery Awaken
Unlock Inverter
Unlock Inverter - Advanced
Inverter Temperature

I tried connecting by TCP an serial, but its the same.

1. Integration Version: lastest
2. HA Version: latest
3. Inverter Brand: Solax G4 X1 7.5

Are those the only entities to show or is there anyway to have others

Thank you very much

wills106, 2022-12-01:
Do you get an error in your logs like "unrecognized inverter type - serial number : {your_serial_number_here}"

pocahozas, 2022-12-01:
Yes, I have received many errors included "unrecognized inverter type - serial number : {your_serial_number_here" . It's clear that I made something wrong. Can you please help with this? Thank you very much guys

Here are the first errors in the log file:

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:88
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 16:33:54 (1 occurrences)
Last logged: 16:33:54

SolaX: reading serial number from address 0x0 failed; other address may succeed

Logger: pymodbus.client.sync
Source: custom_components/solax_modbus/__init__.py:276
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 16:33:54 (2 occurrences)
Last logged: 16:33:54

[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/__init__.py:276
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 16:33:54 (2 occurrences)
Last logged: 16:33:54

SolaX: attempt to read serialnumber failed at 0x0
SolaX: attempt to read serialnumber failed at 0x300
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 75, in _read_serialnr
    inverter_data = hub.read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
  File "/config/custom_components/solax_modbus/__init__.py", line 276, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 108, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusSerialClient(rtu baud[19200])]

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:90
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 16:33:54 (1 occurrences)
Last logged: 16:33:54

SolaX: reading serial number from address 0x300 failed; other address may succeed

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:97
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 16:33:54 (1 occurrences)
Last logged: 16:33:54

SolaX: cannot find serial number, even not for MIC

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:138
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 16:33:54 (1 occurrences)
Last logged: 16:33:54

unrecognized inverter type - serial number : unknown

wills106, 2022-12-01:
I guessed the Gen4 X1 7.5kW would have a serial starting 'H475' but I guess not.
What is the first 4 digits of your serial? I can add it to the Integration.

wills106, 2022-12-01:
How exactly are you trying to connect to your Inverter?

If you have working communications you shouldn't be getting:
unrecognized inverter type - serial number : unknown

But:
"unrecognized inverter type - serial number : H475xxx}"

For example.

pocahozas, 2022-12-01:
The serial number starts with 'H475'. 
I'm trying to connect through the pocket wifi, so I tried first the TCP but I got the same result. So maybe that's the problem. Is it possible to connect by the pocket wifi?
Thank you

wills106, 2022-12-01:
> Is it possible to connect by the pocket wifi?

As a general rule I always say no. While there are some versions out there to do provide Modbus over TCP, I don't know which do and personally I don't recommend them anyway.

Checkout the Wiki for know working RS485 - Ethernet adaptors, pay special attention to the Non working section.

pocahozas, 2022-12-01:
Thank you very much

wills106, 2022-12-05:
Do you want me to leave this issue open while you find a compatible method to connect to your Inverter?
Or shall I close the issue down?

pocahozas, 2022-12-05:
You can close it.
Thanks a lot for your time

closed 2022-12-06 by wills106

msgre, 2023-04-13:
Hi @pocahozas, I experienced similar situation yesterday: https://github.com/wills106/homeassistant-solax-modbus/issues/311#issuecomment-1507111619