wills106/homeassistant-solax-modbus #226: [FEATURE REQUEST] Wake up inverter automatically when writing registers  when inverter is in sleep mode

infradom, 2022-12-02:
This feature request issue replaces partly issue #48 
@ezfrag2021 has suggested to use an automation for this, thanks a lot for the sample code
@MOLTE wonders if this behavior cannot be implemented by default in the integration

In this thread, I want to discuss the possible approaches and their advantages and risks.

The approach that comes to my mind is:

- Modify the write_register code so that it:

   - first checks whether the inverter is in sleep mode
   - if not in sleep mode: write directly (as current implementation)
   - if in sleep mode:
      - put the request in a queue (new)
      - send a wake_up command
 
- detect when state changes from sleep to normal mode
   - when such a state change is detected, the queue is read and all changes are applied with write_register
   -

wills106, 2022-12-05:
Some risks I can think of is:

If a user doesn't know the Integration is waiting for the Inverter to wake up, could they keep trying different values. Then the Inverter wakes up, does the Integration write all the different values to the Inver one after another. Potentially leaving the Inverter in an undesired state?

With an automation. Say the user wants to charge for 30mins during a cheaper period in the night. They have an automation to put the Inverter into charge mode at 02:00 but the Inverter doesn't doesn't start charging till 02:07 and charges for 30 mins, but the 7 min slot at 02:30 is very expensive. I don't know if people have dynamic 15 min prices but this would be even worse potentially.

infradom, 2022-12-05:
In my current proposal, the system only remembers the last written value for each entity. So it will only write each modified parameter only once. I expect the system to be awake after a couple of polling cycles, so seconds.
But you are right, this is not without risks.

wills106, 2022-12-05:
I think it's a good idea, I just want to make sure we have thought about any potential risks.

wills106, 2022-12-08:
@ezfrag2021 @M0LTE @mrpohoda

Do you want to try @infradom code change for waking, waiting for Inver to be awake, before writing to Inverter code?
Try 0.7.0b71 or later.

Just be warned that section of the code hasn't been tested by either of us yet.

M0LTE, 2022-12-08:
Thanks for the heads up, I'll roll forwards when I have a clear time after which I'll be able to roll back if needed :)

wills106, 2022-12-08:
Would you also be able to provide information on the response time for the Inverter waking up and responding?

M0LTE, 2022-12-08:
Of course.

M0LTE, 2022-12-12:
I have not yet switched to the beta but I do note that when I use the SolaX
Battery Awaken button when the inverter is in idle mode, my SolaX Run Mode
does NOT change from Idle to Checking (or anything else).

Should it?

infradom, 2022-12-13:
There was an error in the implementation of auto-wake. I have issued a new PR that should correct this. (not yet released)
The bad new is that the method uses the equivalent of the awake button method, so if that does not work, auto-awake will probably also not work...

infradom, 2022-12-13:
I deleted the 2022.12.5 release as it made things worse.
Will correct soon ... but probably not today

M0LTE, 2022-12-13:
Hi Jan

Would you mind explaining what the awake button actually does, so that I
can try to find out from SolaX why it doesn’t work on my inverter
specifically please?

Thanks
Tom

infradom, 2022-12-13:
Hello Tom,

For all known Solax hybrid models, the awake button is defined as: 
send value 1 to write register with address 0x56 hex or address 86 decimal
According to the Solax documents, this register has a description: Bat_Awaken and we should write a 1 to activate

I have no more information about this command.

What I try to do is to queue such a wakeup command whenever the user sends a write action while the inverter is asleep. When the sytem detects that use_mode becomes normal, it sends the queued command (only once)
A working/tested version of this addition has not yet been published

Jan

infradom, 2022-12-13:
@MOLTE : One important addition: the button currently does not work anymore due to a failed attempt to implement auto-awake. I have created a pull request to correct this, but it must be tested first before we release it. If you are familiar with Github, you can apply the change to __init__.py to your installation. The pull request is: https://github.com/wills106/homeassistant-solax-modbus/pull/242

M0LTE, 2022-12-13:
Thank you for this clear information.

M0LTE, 2022-12-17:
> The bad new is that the method uses the equivalent of the awake button method, so if that does not work, auto-awake will probably also not work...

Please see #250

wills106, 2022-12-17:
As per my comment in #250 writing to "Modbus Power Control" is planned but needs further work on write_registers and is on the Todo List V2 #200

M0LTE, 2022-12-17:
Can I sponsor this work?

wills106, 2022-12-17:
write_registers will be expanded over the next couple of days to allow full "Modbus Power Control"

infradom, 2022-12-20:
See wiki page https://github.com/wills106/homeassistant-solax-modbus/wiki/Modbus-Power-Control-(remotecontrol-power)
There is currently no link yet to the battery awake button. This is only the first step

wills106, 2022-12-20:
I have just released [2022.12.9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2022.12.9)

M0LTE, 2022-12-20:
Great, thanks.

Given “there is currently no link yet to the battery awake button” (ref
Jan) - would you value feedback for this release or shall I hold off
upgrading again?

infradom, 2022-12-20:
We can already test whether a remote_power button activation triggers a wakeup of the inverter. Once it is dark (in a couple of hours), my inverter is configured to go to sleep. So then I will check if remote_power can wake up the inverter.
Once this works, we can rebind the awake button to this function (and activate auto-awake)

wills106, 2022-12-20:
I could be possible we don't even need the awake button depending on how the Modbus Power Control Works.
It might automatically override the Inverter?

infradom, 2022-12-20:
Yes indeed, if there is an automation that uses Modbus Power Control, the inverter might wakeup automatically
I will test this evening

wills106, 2022-12-20:
My Inverter never goes to sleep so I don't know. We would need someone with a Gen3 to test.

How it's currently setup it will only discharge anyway. I would have to alter the scale it include positive numbers. It's only set to negative for export.

M0LTE, 2022-12-20:
FWIW all I want is for the existing controls (particularly around self-use
min SoC) to be changeable after the inverter has gone into Idle mode.

infradom, 2022-12-20:
I tested on Gen4, and modbus power control does not wake up the inverter. It does allow us (together with autorepeat) to keep the inverter awake, but this will consume more energy.
We keep searching. With the cloud app, I could only awake the inverter by switching it off and back on, and then quickly perform an action that avoids that the inverter will reenter idle mode (e.g. lower discharge SOC level)

infradom, 2022-12-20:
Maybe we need to implement write register SystemOnOFF 0x1C, switching off and back on will probably briefly awaken the inverter. (But that are actions that are writen to EEPROM apparently. Maybe writing just an ON will awake the inverter. I have not tested this yet ..

infradom, 2022-12-21:
Even from the cloud app, In my Gen4 I could not wake up the inverter without performing a system_off - system_on cycle.
After the system_on, I only have a few seconds to modify the selfuse SOC level.
We implemented system_on and system_off buttons (should be released soon after some more testing).
These buttons work if the inverter is in normal mode, or switched off
I did not succeed in waking up an inverter in idle mode with these buttons.
I will re-read the threads to find out how people managed to wake up a Gen4 inverter through an automation

M0LTE, 2022-12-21:
I *can* change settings (commonly, self-use min SoC) on my Gen 4 when it’s
in Idle from the SolaX Cloud app. When I change that setting, it changes
from Idle to Checking then to Normal, and starts running the house off the
batteries.

infradom, 2022-12-21:
Thanks for the info, 
In my case, the reason for being in idle mode was still present (battery soc lower that min), maybe for that reason it did not work. I will retry again this evening ...

M0LTE, 2022-12-21:
To be clear I have not attempted this using the modbus integration

infradom, 2022-12-21:
I tried to understand the behavior in the cloud app first before experimenting again with the modbus integration. But I was surprised that I could not wake it up in the cloud app

infradom, 2022-12-21:
Still no success in waking up the inverter, whatever I try ...

M0LTE, 2022-12-27:
I asked SolaX

> How, specifically, do we use 0x7E/0x7F to bring the gen 4 out of Idle mode and into Normal mode?

Their response:

> The parameters need be set for power control:
> 0x007C : select mode1,
> 0x007D : choose the type of target set,
> 0x007E: set the value of power target,
> 0x0082: set the power target hold time.

Is this useful?

infradom, 2022-12-27:
Thanks Tom, this is exactly what we implemented with our Modbus Power Control.
It is not yet linked to our auto-wakeup code since I could not wake up my inverter with this Modbus Power Control.
I would appreciate if you could test this also (manually). See Wiki page https://github.com/wills106/homeassistant-solax-modbus/wiki/Modbus-Power-Control-(remotecontrol-power) for a short description. Dont forget to click the trigger link after setting the parameters.

M0LTE, 2022-12-27:
Will do, though I’m not able to at present.

If this doesn’t work then I suggest we work with the SolaX engineer who is
responding to me.

Shall I point them here?

M0LTE, 2023-01-03:
Hi @infradom I've just got to reading that doc and getting ready to test.

If I'm understanding correctly, it seems that to wake the gen 4 inverter from idle, we almost need to take over control of power flow on an ongoing basis? Rather than merely waking the inverter, allowing settings to be adjusted through the existing means?

I need to sit and really study that document to understand how it might be applicable to my situation- seems to be a totally different way of running the inverter.

My use is as follows. During a 4 hour period each night I take a grid charge. During the day I'm in self-use mode. I'd like to reserve the last usable 10% of the battery for EPS mode and switch from self use mode to backup mode if grid power is lost. That should just be a simple automation but I lose the ability to switch mode after the inverter drops to Idle run mode.

Any tips or thoughts?

Thanks for your work so far.

infradom, 2023-01-04:
Hello Tom, 
I used to work in the mode you describe in the past. 
Since we implemented the (prototype) remotecontrol Power, I tend to run in one of these modes almost the entire day and night. Yes, an automation changes the settings to make sure batteries are charged at night (when prices  and PV forecast are low).  In fact, currently, my automations switch between 3 modes:

- self use mode (or power_control with target = 0 and import_limit = 0) . Typically used when hourly prices are high
- power_control with target = 0, import_limit= non-zero. Typically used during periods with itermediate prices when it is not worth charging/discharging betteries for a couple of cents
- grid_control (or power_control with a positive target): to force charge the battery when prices are really low

I am assuming here that you also have dynamic hourly (day-ahead) prices. For static fixed prices, simpler rules can be used.
By using this method, I keep the battery level much higher than in the past (dependent on the solar forecast), so it is very unlikely that the inverter goes to sleep. The self-power consumption in the second mode is relatively low as the inverter just passes power from the grid to the house load.

Regarding wake up: I am not sure whether the remotepower commands can wake up the inverter. My first tests were negative, but I did not put much effort in testing and debugging yet.

radoslavius, 2023-01-25:
Hi,

if I use local connect in the mobile app or remote settings on the solaxcloud website and set option system on/off to ON and click to save, the inverter will wake up. I can not monitor the mobile app, but POST data in the browser can - see attached screenshot

[image]

I'm not sure if this will help you, or if you may already know, but I'm looking for a way to wake up the inverter at night. It is an X3-Hybrid-G4 inverter

Thank you

wills106, 2023-01-26:
That should be the same as Button "System On" although that doesn't map to 47 (0x2F) "System On" is 0x1C
The registers around 0x2F are to do with Calibration / Adjustment registers.

markxroberts, 2023-01-30:
I seem to be able to change system mode when in Idle mode by interacting via SolaxCloud, entering my unlock pin and then returning to the main page.  The inverter then enters checking mode.  The data submitted is to register 0.  Pin concealed in example below.  Is this possible via Modbus?
````
Data: [{"reg":0,"val":"XXXX"}]
````

wills106, 2023-01-30:
The button "Unlock Inverter" / "Unlock Inverter - Advanced" uses 2014 / 6868 already.
Just so you know the values in the Cloud don't map 100% to the Modbus registers.

M0LTE, 2023-01-30:
Can the PIN those buttons use be configured? This may be why they have
never worked for me.

M0LTE, 2023-01-30:
> Are you not using the default pin codes?
> You could manually change 2014 to what you want in the code.
> Just search for "Unlock Inverter" in plugin_solax

No, not using the default PIN codes. Wonder if I can get it to read from an
environment variable. Thanks for the heads up. This could be useful because
the power control stuff is frankly way too complex for my needs.

wills106, 2023-01-30:
> Wonder if I can get it to read from an environment variable.

Should be able to use the relatively new Text entitie.
But it's not something I have looked into, so I don't know how to use it yet.

It would be worth manually changing the value locally and seeing if it works first.

M0LTE, 2023-01-30:
Will do

infradom, 2023-02-01:
> the power control stuff is frankly way too complex for my needs.

I edited the documentation of modbus power control, it should be less frightening now, and it is by far the simplest model for controlling the Gen4 inverter (in my opinion). I would suggest you give it a try, without automation to start with ...

wills106, 2023-02-07:
@markxroberts @M0LTE 
Did either of you try and manually change the pin that "Unlock Inverter" uses?
Did it actually wake up the Inverter?

M0LTE, 2023-02-07:
I’ve changed my PIN back to default but not had the opportunity to check
idle due to usage patterns since. Will report back.

markxroberts, 2023-02-08:
Disappointingly it doesn’t work for me.

wranglatang, 2023-03-24:
Hi all, Weighing in here, I have a Gen 4 hybrid and would be happy to test. My inverter has the default pin (at least I have never changed it). I have only ever seen my inverter in the state of "Normal", "Idle" & "Waiting", and i have never been able to get it out of an Idle state using the buttons are set up with the integration.

Any thing that I can do to test and help out?

wills106, 2023-09-03:
I'm closing this down as I don't know what extra command is being sent from the cloud to wake up the Inverter if it's sleeping when you try to change the Work Mode for example.

Also my Inverter never goes to sleep, so this isn't something I can investigate.

closed 2023-09-03 by wills106

ninoz, 2023-11-19:
As a bit of a hack I have found sending a rest command to the inverter will wake it up.

In my configuration.yaml I added:

```
rest_command:

  solax_local_charge_battery_from_grid:
    url: !secret solax_local_ip
    method: post
    payload: !secret solax_local_set_battery_level_payload
```

In secrets.yaml I added:
```
solax_local_ip: http://192.168.1.x/ # IP of solax module
solax_local_set_battery_level_payload: 'optType=setReg&pwd=$$$REG_NO$$$&data={"num":1,"Data":[{"reg":31,"val":"{{ 100}}"}]}'
```
You can obtain the the registration number (REG_NO) from [Solax Cloud ](https://www.solaxcloud.com/).

Then in my automations I check if the inverter is idle, if it is I then send the rest command, wait for a few seconds and check the state

```
  - alias: Check if Inverter is idle
    if:
      - condition: state
        entity_id: sensor.solaxmodbus_run_mode
        state: Idle Mode
    then:
      - repeat:
          sequence:
            - service: rest_command.solax_local_charge_battery_from_grid
              data: {}
            - delay:
                hours: 0
                minutes: 0
                seconds: 3
                milliseconds: 0
          until:
            - condition: state
              entity_id: sensor.solaxmodbus_run_mode
              state: Waiting
```

Excentyl, 2024-01-18:
Thanks @ninoz, I have tried the above and it works great!

Personally, I have set up an automation which only wakes the inverter when I hit the remotecontrol trigger (because this is how I charge the battery from grid)

`alias: "Roof Solar: Wake on Remotecontrol"
description: ""
trigger:
  - platform: state
    entity_id:
      - button.roof_solar_remotecontrol_trigger
condition: []
action:
  - if:
      - condition: state
        entity_id: sensor.roof_solar_run_mode
        state: Idle Mode
    then:
      - service: rest_command.solax_local_charge_battery_from_grid
        metadata: {}
        data: {}
mode: single`

@wills106 - I hope you don't mind, I am tagging you in this in case you didn't see it. I am not sure if something like this could be built into the custom component, but if not it may be worth adding details on the rest command in the documentation?