wills106/homeassistant-solax-modbus #232: [BUG] In the top right of the screen, to the left of the SolaX Power, it shows unavailable % and a battery symbol

pchaves1, 2022-12-06:
Hi, hope you're well.

Go to ... Settings => Devices & Services => SolaX => 1 Device
In the top right of the screen, to the left of the SolaX Power, it shows unavailable % and a battery symbol.
I have a Solax Gen4 inverter.

[image]

I'm using the latest version from today.
Any advice?

Thanks in advance!

wills106, 2022-12-07:
When you say latest do you mean latest stable 0.6.4 or the latest Beta? As I released two Beta's yesterday.

wills106, 2022-12-07:
When you went from 0.6.4 to 0.7.0bx did you make sure you removed it from the Integration page and restarted HA before adding it back?

pchaves1, 2022-12-07:
Sorry, your are correct!
I did not
Now it's OK
[image]

wills106, 2022-12-08:
I'll close this as the bug is fixed in 0.7.0b

closed 2022-12-08 by wills106