wills106/homeassistant-solax-modbus #233: [BUG] Disconnection of the modbus connection

hlavina, 2022-12-07:
Hello, I have a problem with the constant disconnection of the modbus connection (about 1x a day at different times). It will only help to set up the wifi dongle anew and restart HA.

Can you advise where the problem with the constant disconnection could be?

**Mandatory details**
1. Integration Version: 4da271b
2. HA Version: Home Assistant 2022.11.5, Supervisor 2022.11.2, Frontend 20221108.0 - latest
3. Inverter Brand: Střídač Solax X3 g4 Hybrid 10kW, wifi dongle

**Log**
`Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:282 
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues)) 
First occurred: 13:29:00 (1 occurrences) 
Last logged: 13:29:00

Modbus Error: [Connection] ModbusTcpClient(192.168.2.12:502): Connection unexpectedly closed 0.000041 seconds into unbounded read without response from unit before it closed connection: SolaX cannot read input registers at device 1 position 0x8
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 336, in read_modbus_block
    if typ == 'input': realtime_data = self.read_input_registers(unit=self._modbus_addr, address=block.start, count=block.end - block.start)
  File "/config/custom_components/solax_modbus/__init__.py", line 282, in read_input_registers
    return self._client.read_input_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/common.py", line 125, in read_input_registers
    return self.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 109, in execute
    return self.transaction.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 174, in execute
    response, last_exception = self._transact(
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 294, in _transact
    result = self._recv(response_length, full)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 362, in _recv
    result = self.client.framer.recvPacket(expected_response_length)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/__init__.py", line 49, in recvPacket
    return self.client.recv(size)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 89, in recv
    return self._recv(size)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 297, in _recv
    return self._handle_abrupt_socket_close(
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 342, in _handle_abrupt_socket_close
    raise ConnectionException(msg)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(192.168.2.12:502): Connection unexpectedly closed 0.000041 seconds into unbounded read without response from unit before it closed connection`

wills106, 2022-12-07:
The pocket WiFi isn't officially supported.

This is mentioned in the Readme https://github.com/wills106/homeassistant-solax-modbus#070-beta-x-adds-in-preliminary-support-for-ginlong-solis--sofar-solar

wills106, 2022-12-08:
Closing this issue as probably related to Pocket WiFi as no other reports of this issue to date.
If you try one of the supported RS485 adaptors and still have issues please re-raise the issue.

closed 2022-12-08 by wills106