wills106/homeassistant-solax-modbus #235: [BUG] - Fail to execute config flow in latest version of HA

hallidaydaniel, 2022-12-09:
Following an update to the latest Modbus integration and HA, 
**Mandatory details**
1. Integration Version:
2. HA Version: 2022.12.1
3. Inverter Brand: Solax 
4. Plugin Used: 
5. First 6 digits of Serial: XAC362
6. Connection Method: MODBUS TCP

**Detailed Error Log**
Logger: aiohttp.server
Source: helpers/schema_config_entry_flow.py:165
First occurred: 12:40:24 (2 occurrences)
Last logged: 12:40:33

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 81, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 136, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 180, in post
    return await super().post(request, flow_id)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 73, in wrapper
    result = await method(view, request, data, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 281, in async_configure
    result = await self._async_handle_step(
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 368, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
  File "/usr/src/homeassistant/homeassistant/helpers/schema_config_entry_flow.py", line 305, in _async_step
    result = await self._common_handler.async_step(step_id, user_input)
  File "/usr/src/homeassistant/homeassistant/helpers/schema_config_entry_flow.py", line 129, in async_step
    return await self._async_form_step(step_id, user_input)
  File "/usr/src/homeassistant/homeassistant/helpers/schema_config_entry_flow.py", line 165, in _async_form_step
    user_input = await form_step.validate_user_input(self, user_input)
TypeError: _validate_base() takes 1 positional argument but 2 were given

**Additional context**
When setting up the integration, you can't proceed past the first step: 
[image]

I have tried updating HA, restarting HA, and removing and re-adding the integration in HACS.

wills106, 2022-12-09:
The issue of not being able to Install the Integration on 2022.12.x has been resolved thanks to @infradom in [0.7.0b74](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/0.7.0b74)

I'll close this and the linked issue down now.

wills106, 2022-12-09:
I have decided to reopen this for a couple of days incase anyone else upgrading to HA 2022.12.x doesn't realise you need to update this custom_component Integration.

wills106, 2022-12-17:
Closing down as no further reports and 2022.12.x has been out for a while now.

closed 2022-12-17 by wills106